lwfinger/rtl8188eu #169: Driver stops working on Fedora 23

carldong, 2016-04-09:
First, this driver worked for a long time before, but today the device cannot be turned on.

    $ sudo iwconfig wlan0 power on
    Error for wireless request "Set Power Management" (8B2C):
        SET failed on device wlan0 ; Operation not permitted

I have tried added `rtw_power_mgmt=0` parameter to the module, and recompiled a few times without success. I also tried to enable the interface using `ifconfig wlan0 up`, no success. It is a WN-725N which did work for a long time.

My current kernel is 4.4.6. I tried to revert to previous kernels (unfortunately I destroyed their modules by mistake), but still no success. `iwconfig` always give `Power Management: off`

unclechu, 2016-04-09:
I can up interface by ifconfig: `ifname wlp0s20u1u3 up` and I see this interface in `ifconfig` but NetworkManager doesn't see it even after restart NetworkManager.

kuntan2237, 2016-04-09:
I got the same problem with Fedora 23 too, it seems this issue was introduced by some latest Fedora 23 updates.
Someone else was asking the same question at Fedora forum -> http://forums.fedoraforum.org/showthread.php?t=309485 <http://forums.fedoraforum.org/showthread.php?t=309485>

I have to roll back my system to 21 and the dongle working fine.

-Kun

lucianogreiner, 2016-04-09:
I have the same problem with wlp0s20u2 :-/

carldong, 2016-04-11:
Looks like it is not limited to rtl8188eu. I got another dongle with RTL8812au and it has the same problem. I think the problem is very recent, not related to the kernel(since old kernels fail too)

carldong, 2016-04-11:
Here I attach the two suspicious updates I performed which may caused the wireless to break:

```
Begin time     : Fri Mar 18 19:19:53 2016
Begin rpmdb    : 5365:b34a6e8ff781672bcca945b45df69c681006f1eb
End time       :            19:24:47 2016 (294 seconds)
End rpmdb      : 5385:4eed38c11173eee49b0c2d04cea54c7723cbfcad
User           : -------
Return-Code    : Success
Command Line   : update
Transaction performed with:
    Upgraded      dnf-1.1.6-2.fc23.noarch         @updates
    Upgraded      rpm-4.13.0-0.rc1.11.fc23.x86_64 @updates
Packages Altered:
    Upgraded   MUMPS-5.0.1-10.fc23.x86_64                                       @updates
    Upgrade          5.0.1-12.fc23.x86_64                                       @updates
    Upgraded   MUMPS-common-5.0.1-10.fc23.noarch                                @updates
    Upgrade                 5.0.1-12.fc23.noarch                                @updates
    Upgraded   MUMPS-devel-5.0.1-10.fc23.x86_64                                 @updates
    Upgrade                5.0.1-12.fc23.x86_64                                 @updates
    Upgraded   MUMPS-mpich-5.0.1-10.fc23.x86_64                                 @updates
    Upgrade                5.0.1-12.fc23.x86_64                                 @updates
    Upgraded   MUMPS-mpich-devel-5.0.1-10.fc23.x86_64                           @updates
    Upgrade                      5.0.1-12.fc23.x86_64                           @updates
    Upgraded   MUMPS-openmpi-5.0.1-10.fc23.x86_64                               @updates
    Upgrade                  5.0.1-12.fc23.x86_64                               @updates
    Upgraded   MUMPS-openmpi-devel-5.0.1-10.fc23.x86_64                         @updates
    Upgrade                        5.0.1-12.fc23.x86_64                         @updates
    Upgraded   abrt-2.8.0-2.fc23.x86_64                                         @updates
    Upgrade         2.8.0-3.fc23.x86_64                                         @updates
    Upgraded   abrt-addon-ccpp-2.8.0-2.fc23.x86_64                              @updates
    Upgrade                    2.8.0-3.fc23.x86_64                              @updates
    Upgraded   abrt-addon-coredump-helper-2.8.0-2.fc23.x86_64                   @updates
    Upgrade                               2.8.0-3.fc23.x86_64                   @updates
    Upgraded   abrt-addon-kerneloops-2.8.0-2.fc23.x86_64                        @updates
    Upgrade                          2.8.0-3.fc23.x86_64                        @updates
    Upgraded   abrt-addon-pstoreoops-2.8.0-2.fc23.x86_64                        @updates
    Upgrade                          2.8.0-3.fc23.x86_64                        @updates
    Upgraded   abrt-addon-python3-2.8.0-2.fc23.x86_64                           @updates
    Upgrade                       2.8.0-3.fc23.x86_64                           @updates
    Upgraded   abrt-addon-vmcore-2.8.0-2.fc23.x86_64                            @updates
    Upgrade                      2.8.0-3.fc23.x86_64                            @updates
    Upgraded   abrt-addon-xorg-2.8.0-2.fc23.x86_64                              @updates
    Upgrade                    2.8.0-3.fc23.x86_64                              @updates
    Upgraded   abrt-cli-2.8.0-2.fc23.x86_64                                     @updates
    Upgrade             2.8.0-3.fc23.x86_64                                     @updates
    Upgraded   abrt-dbus-2.8.0-2.fc23.x86_64                                    @updates
    Upgrade              2.8.0-3.fc23.x86_64                                    @updates
    Upgraded   abrt-desktop-2.8.0-2.fc23.x86_64                                 @updates
    Upgrade                 2.8.0-3.fc23.x86_64                                 @updates
    Upgraded   abrt-gui-2.8.0-2.fc23.x86_64                                     @updates
    Upgrade             2.8.0-3.fc23.x86_64                                     @updates
    Upgraded   abrt-gui-libs-2.8.0-2.fc23.x86_64                                @updates
    Upgrade                  2.8.0-3.fc23.x86_64                                @updates
    Upgraded   abrt-libs-2.8.0-2.fc23.x86_64                                    @updates
    Upgrade              2.8.0-3.fc23.x86_64                                    @updates
    Upgraded   abrt-plugin-bodhi-2.8.0-2.fc23.x86_64                            @updates
    Upgrade                      2.8.0-3.fc23.x86_64                            @updates
    Upgraded   abrt-python3-2.8.0-2.fc23.x86_64                                 @updates
    Upgrade                 2.8.0-3.fc23.x86_64                                 @updates
    Upgraded   abrt-retrace-client-2.8.0-2.fc23.x86_64                          @updates
    Upgrade                        2.8.0-3.fc23.x86_64                          @updates
    Upgraded   abrt-tui-2.8.0-2.fc23.x86_64                                     @updates
    Upgrade             2.8.0-3.fc23.x86_64                                     @updates
    Upgraded   appstream-data-23-7.fc23.noarch                                  @updates
    Upgrade                   23-8.fc23.noarch                                  @updates
    Upgraded   ardour4-4.6.0-1.fc23.x86_64                                      @updates
    Upgrade            4.7.0-2.fc23.x86_64                                      @updates
    Upgraded   ardour4-audiobackend-alsa-4.6.0-1.fc23.x86_64                    @updates
[... cut ...]
    Upgraded   kdelibs-ktexteditor-6:4.14.17-1.fc23.x86_64                      @updates
    Upgrade                        6:4.14.18-2.fc23.x86_64                      @updates
    Obsoleting kdelibs-nepomuk-6:4.14.18-2.fc23.x86_64                          @updates
    Install    kdelibs-nepomuk-6:4.14.18-2.fc23.x86_64                          @updates
    Upgraded   kdelibs-webkit-6:4.14.17-1.fc23.x86_64                           @updates
    Upgrade                   6:4.14.18-2.fc23.x86_64                           @updates
```

tangowh, 2016-04-12:
I managed to get the wifi working by downgrading NetworkManager
`# sudo dnf downgrade NetworkManager --allowerasing`

However the iwconfig still fails. Not sure it is a problem though.
`# sudo iwconfig wlp0s6f1u1 power on
Error for wireless request "Set Power Management" (8B2C) :
    SET failed on device wlp0s6f1u1 ; Operation not permitted.
#`

tangowh, 2016-04-12:
There is a NetworkManager bug open https://bugzilla.gnome.org/show_bug.cgi?id=764877
Here is also the Fedora bug https://bugzilla.redhat.com/show_bug.cgi?id=1325631

lwfinger, 2016-04-12:
The iwconfig "problem" is that that ioctl is not implemented. Ignore it.

unclechu, 2016-04-12:
@tangowh thanks a lot for this solution `sudo dnf downgrade NetworkManager --allowerasing`, it works for me!

carldong, 2016-04-12:
Confirmed temporary solution of downgrading networkmanager works