lwfinger/rtl8188eu #170: archlinux can't make

ruizhipeng, 2016-04-11:
`roy@Dipper ~/Downloads/rtl8188eu (git)-[master] % uname -a
Linux Dipper 4.4.5-1-ARCH #1 SMP PREEMPT Thu Mar 10 07:38:19 CET 2016 x86_64 GNU/Linux`

I'm a fresher who from China ,maybe my english will hard to understand. Sorry for that. 
When i make the driver ,it print an error.

`roy@Dipper ~/Downloads/rtl8188eu (git)-[master] % make all
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.4.5-1-ARCH/build M=/home/roy/Downloads/rtl8188eu  modules
make[1]: Entering directory '/usr/lib/modules/4.4.5-1-ARCH/build'
make[1]: *** No rule to make target 'modules'。 停止。
make[1]: Leaving directory '/usr/lib/modules/4.4.5-1-ARCH/build'
Makefile:147: recipe for target 'modules' failed
make: *** [modules] Error 2`

and if you can help me,I'll be very pleased.

lwfinger, 2016-04-11:
You have not installed the kernel headers. You may have to inquire at an ARCH forum on the exact name of the package. I do not use ARCH, and the command I use would not help you.

stuartpb, 2016-04-13:
The name of the package with Linux headers on Arch Linux is `linux-headers`:

```sh
sudo pacman -Syu
sudo pacman -S linux-headers
```

KillerDAN, 2016-04-13:
I have the same issue on an amd64 machine.

I have two kernel trees installed with the same packages installed for each
one.

3.2.0-4 tree builds fine.

3.16.0-4 does not.

lwfinger, 2016-04-13:
Without any build errors, how am I supposed to know what is going wrong? My crystal ball is broken.

KillerDAN, 2016-04-13:
You don't need a cristal ball... @lwfinger the build error is the exact "same issue" one opening this thread.

lwfinger, 2016-04-13:
Then same answer. You always need kernel headers that match the running kernel.

ruizhipeng, 2016-04-14:
maybe when the kernel fresh this issue will appear again so if i have any other questions please help me again as usual. Best wishes and enjoy the nice day ^ ^

KillerDAN, 2016-04-14:
And again you are not reading this trough.
Kernel headers are installed for both kernel trees with the same packages
for each tree.

I don't this is an issue with rtl8188eu code by itself but something
changes after a certain version in the way the packages distribute the
files and the kernel is compiled.

ruizhipeng, 2016-04-14:
maybe you should check the modules?just naive guess

allaudet, 2016-04-14:
The `make[1]: *** No rule to make target 'modules'` error means that you are not pointing to the correct Kernel Headers.
Check if your Makefile's **KSRC** option is pointing to the correct ones.
It is common that KSRC gets its value depending on a `KVER=$(shell uname -r)` variable, maybe there is where the problem resides.

In any case, there is nothing related with this Driver at this point.

KillerDAN, 2016-04-14:
I did check where it points, that is how I know it compiles against one
kernel tree and not the other because I change to which tree I want it
build against.
I have been compiling this driver for some 2 years now and this is only
happening with this latest kernel tree.
The driver, or the hardware/ firmware do have some other issue but those
are off topic. Every 3/ 4 weeks it stops working and I need to remove and
load the module again.

lwfinger, 2016-04-14:
The make file is designed to build against the kernel that is currently running. As stated above, if you want to build against some other kernel, you will need to muck with the symbol KVER.

If you get stability for 3 or 4 weeks, that is excellent. Some people only get minutes.

By the way, I do read thoroughly and I did diagnose your problem as one with the headers. I have been doing this for a long time, and that error message is absolutely diagnostic.

micenote, 2018-02-07:
My crystal ball is broken. 
wtf?? ^^

lwfinger, 2018-02-07:
WTF indeed.

closed 2018-02-07 by lwfinger