lwfinger/rtl8188eu #171: Not playing nicely with NetworkManager 1.0.12-1

stuartpb, 2016-04-13:
Users have been reporting issues with NetworkManager not recognizing this family of wireless adapters after upgrading to NetworkManager 1.0.12-1:

https://bbs.archlinux.org/viewtopic.php?id=210868

Arch Linux Forum user V1del [hypothesizes](https://bbs.archlinux.org/viewtopic.php?pid=1617699#p1617699) that this issue is being caused by the drivers not putting the devices into the correct management state, which conflicts with [a patch introduced to the NetworkManager source tree in February](https://cgit.freedesktop.org/NetworkManager/NetworkManager/commit/?h=1.0.12&id=a3c0166a2064c41408d1e08defee3a7e4a0ce3f7).

gstvrmrlpz, 2016-04-13:
any workaround?

lwfinger, 2016-04-13:
Have you tried the v4.1.8_9499 branch? It implements all of the cfg80211 calls and probably works better with NetworkManager. I do not have any problems with the version of NM that I am using.

stuartpb, 2016-04-13:
Actually, my own chipset is using the https://github.com/gnab/rtl8812au driver, which appears to be having a similar problem. I'll file a corresponding bug over there.

GitatHub, 2016-04-16:
to workaround:
in fedora 23
sudo dnf downgrade NetworkManager --allowerasing

this will downgrade NetworkManager, and this "fixes" the problem

GitatHub, 2016-04-16:
the v4.1.8_9499 branch does not work with NM1.0.12-1 either, and downgrading NM fixed the issue.

joneslee85, 2016-06-30:
I've tested on NM 1.2.2-1 and it works fine. I think it is safe to upgrade and close this ticket.