lwfinger/rtl8188eu #172: RPi2 + OSMC + TL-WN725N - WiFi disconnecting and connecting

apsikus, 2016-04-13:
Fighting with this setup for over 2 months.

RPi2 with newest OSMC installed, adapter TL-WN725N using driver module R8188EU and my WiFi connection is disconnecting and connecting each 30seconds.
No chance to use any add-on in OSMC.

Please help!

Should I install different driver? 
Compile own for OSMC?

Link to OSMC logs: http://paste.osmc.io/rulomupara

lwfinger, 2016-04-13:
Try loading the driver module using "rtw_power_mgnt=0". That will disable power-saving mode on the link. That process might be confusing your router.

apsikus, 2016-04-13:
I have set it doing this:
create file using command: `sudo nano /etc/modprobe.d/r8188eu.conf`
add single line to it: `options r8188eu rtw_power_mgnt=0 rtw_enusbss=0`

and it didn't change anything.

As you can check in log I have also set `max_usb_current` as active.

They should do the same, but I hope that are not in conflict.

lwfinger, 2016-04-13:
Wifi power management is not the same as the power used by the USB device. It controls whether the radio goes into power-save mode while inactive.

I am only guessing at which parameter to use. Please add "rtw_smart_ps=0" to the line in r8188eu.conf.

lwfinger, 2016-04-13:
Please ignore the suggestion above and use "rtw_ips_mode=0". That is the correct one to disable power-save.

apsikus, 2016-04-15:
options r8188eu rtw_power_mgnt=0 rtw_ips_mode=0 rtw_enusbss=0

http://paste.osmc.io/ijusomefum

Looks like WiFi connection is stable :)

PS. any idea how else I can check it?

apsikus, 2016-04-19:
Any other suggestions?

For which system your drivers are compiled?
Do you have any tutorial to compile it for other systems?

lwfinger, 2016-04-19:
Due to Linux module requirements, it is difficult for anyone other than the distro to provide a pre-compiled binary. In other words, this driver is not pre-built for any system.

There are lots of Web articles on how to build external modules. The required components are make, gcc, the headers that match your running kernel, and the kernel development package. It is also best to install git to handle the source for the driver. Once you have all those things installed, the commands "make" and "sudo make install" are all it takes.

apsikus, 2016-04-20:
If it is not pre-built would it work in OSMC, if I will replace standard bin with your file: rtl8188eufw.bin?

ANy link which you see as useful? I'm not Linux specialist, but will try :)

allaudet, 2016-04-20:
If you are looking for pre-builded drivers, and the odds are with you, try with the Raspbian ones:
[https://www.raspberrypi.org/forums/viewtopic.php?t=62371](url)
Otherwise you will have to "specialize" a little bit more on Linux and compile the driver by yourself (a worth "know how" IMO).

lwfinger, 2016-04-20:
Those .bin files are called firmware, which is specialized code that runs on the CPU that is embedded in the device. Those files are pre-built as that code is generally not open source. In addition, they do not depend on the operating system. For example, the same information is used for Windows that is needed for Linux. The only difference is that Windows incorporates that information as data statements in the code and pushes the firmware without needing to read an external file. That method could be used on Linux, but is generally discouraged in favor of reading the firmware from an external file.

What the individual Linux user needs to build is the driver, which runs on the main, general-purpose CPU. That is unique to each kernel version, and also depends on the version of the compiler used to build the main kernel and the external driver. As stated above, learning to build an external driver is a useful skill.

apsikus, 2016-04-20:
> If you are looking for pre-builded drivers, and the odds are with you, try with the Raspbian ones:
https://www.raspberrypi.org/forums/viewtopic.php?t=62371

If I will decide to use it how to install?

It is enough to use this commands:`wget https://dl.dropboxusercontent.com/u/80256631/8188eu-v7-2015yyzz.tar.gz
tar xzf 8188eu-v7-2015yyzz.tar.gz
./install.sh`

lwfinger, 2016-04-20:
I have no idea if those commands will work. That source is not provided by me and I have no idea as to its quality or usage.

allaudet, 2016-04-21:
> It is enough to use this commands:
> `wget https://dl.dropboxusercontent.com/u/80256631/8188eu-v7-2015yyzz.tar.gz`
> `tar xzf 8188eu-v7-2015yyzz.tar.gz`
> `./install.sh`

That's half of the work.
Realize that https://dl.dropboxusercontent.com/u/80256631/8188eu-v7-2015yyzz.tar.gz does not exist, because the letters **yyzz** have a meaning.

Once you discover what code fits your kernel (`uname -a`), then yes.

apsikus, 2016-04-21:
But OSMC has different kernel then Raspbian, and that is the reason why I have asked if for example using link for newest kernel driver will work in other system.

allaudet, 2016-04-21:
Fast answer: Compile yourself the goddamn driver xD

google(_osmc kernel headers_)[2]: This is a how-to compile tutorial for another chipset driver, but the essence is the same:
https://discourse.osmc.tv/t/tutorial-compiling-a-working-kernel-driver-for-rtl8192eu/5252

Just point out the most important part of whole the tutorial:
> Just for my convenience, I'm assuming you are running Raspberry Pi 1 with "3.18.13-1-osmc" kernel.
This is MOST LIKELY not your case, so make sure to "search and replace" my instructions as follows:
3.18.13-1-osmc: **Change to the output of 'uname -r' command** on your Pi.
rbp1: **Change to rbp2 if you are on Raspberry Pi 2**.
bcm2708: **Change to bcm2709 if you are on Raspberry Pi 2**.

apsikus, 2016-04-21:
I'm planning to do it, really.
But I see too many differences, that's why I'm not sure if I will find all prerequisites as I'm not so Linux experienced.
I have found this link: https://sites.google.com/site/easylinuxtipsproject/reserve-7#TOC-Realtek-RTL8188EU-chipset-0bda:8179-