lwfinger/rtl8188eu #175: alfa awus036nhv

expandables2, 2016-04-24:
I have openwrt how can i install this driver for it? 
Wikidev has rtl8188eus

lwfinger, 2016-04-25:
What kernel does your openWRT use? If it is 3.12 or newer, there is a driver built in. Otherwise you need to cross-compile the driver in this repo using the same build system as was used to build your firmware.

expandables2, 2016-04-25:
I get OpenWrt Chaos Calmer 15.05.1 with kernel 3.18.23.
You said the driver is built in? I have a tp link wr1043nd v2 and a tp link wr1043nd v1 and when I put the alfa awus036nhv in the tp link wr1043nd v2 nothing shows up in Wireless Overview only build in radio(0) but my tp link wr1043nd v1 shows radio(1) but when i click scan nothing comes up.
I install the kmod-net-rtl8188eu for kernel 3.18.23-1.
What am I doing wrong? My alfa R36 works well with alfa awus036nhv but my Alfa R36 is dying please can you help me get the alfa awus036nhv to work with my tp-link router. Thanks!

expandables2, 2016-04-25:
I am a noob please help thanks I would really appreciate it.