lwfinger/rtl8188eu #179: TL-WN725N driver Ubuntu disconnect problem

pteroctopus, 2016-06-23:
I can't get my TL-WN725N WiFi USB adapter to work on Ubuntu. I tried every solution I found on internet and they didn't work. Description of the problem and my system is bellow.
[Also asked on askedubuntu (http://askubuntu.com/questions/790525/tl-wn725n-driver-ubuntu-disconnect-problem)]

* Problem: WiFi connection works great when the machine starts for a minute or less, then WiFi connection disconnects and I am not able to manually start the interface with:
"ifup wlxec086b0d080d"
* Log "/var/log/syslog" doesn't show any errors.

* Question: Is this a driver problem or some configuration problem. How to fix this?

System and configuration info

* Kernel version: Linux ubuntu 4.4.0-24-generic #43-Ubuntu SMP Wed Jun 8 19:27:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

* Kernel version: 4.4.0-24-generic

* Linux installed on virtual machine: Virtualbox Version 5.0.22 r108108 (Windows 7 Host)

* Ubuntu 16.04 LTS; Release:	16.04; Codename:	xenial

* Wifi Nano USB adapter: TL-WN725N

* Driver installed according to this guide: http://brilliantlyeasy.com/ubuntu-linux-tl-wn725n-tp-link-version-2-wifi-driver-install/
* Driver installed correctly
* Added "blacklist r8188eu" to "/etc/modprobe.d/blacklist.conf" to disable loading of old driver.

* Network interfaces configured in "/etc/network/interfaces", configuration bellow:

```

auto lo

iface lo inet loopback

auto enp0s3

iface enp0s3 inet dhcp

auto wlxec086b0d080d

iface wlxec086b0d080d inet dhcp

	wpa-ssid AndroidAP

	wpa-psk ***censored***

```

* $ lsusb

Bus 001 Device 003: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter

* $ ifconfig

wlxec086b0d080d Link encap:Ethernet  HWaddr **:**:6b:0d:08:**

          inet addr:192.168.43.206  Bcast:192.168.43.255  Mask:255.255.255.0

          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1

          RX packets:865 errors:0 dropped:18 overruns:0 frame:0

          TX packets:640 errors:0 dropped:1 overruns:0 carrier:0

          collisions:0 txqueuelen:1000 

          RX bytes:1075999 (1.0 MB)  TX bytes:83546 (83.5 KB)

* $ iwconfig

wlxec086b0d080d  IEEE 802.11bgn  ESSID:"AndroidAP"  Nickname:"<WIFI@REALTEK>"

          Mode:Managed  Frequency:2.437 GHz  Access Point: **:11:**:2F:DF:**   

          Bit Rate:72.2 Mb/s   Sensitivity:0/0  

          Retry:off   RTS thr:off   Fragment thr:off

          Power Management:off

          Link Quality=99/100  Signal level=-40 dBm  Noise level=0 dBm

          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0

          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

* $ sudo lshw -C network

  *-network

       description: Wireless interface

       physical id: 1

       bus info: usb@1:2

       logical name: wlxec086b0d080d

       serial: **:**:6b:0d:08:**

       capabilities: ethernet physical wireless

       configuration: broadcast=yes driver=r8188eu ip=192.168.43.206 
multicast=yes wireless=IEEE 802.11bgn

Thank you.

joneslee85, 2016-06-30:
It's hard to tell, I am on Kernel 4.2.5 (on Arch though) and the dongle works perfectly. I think you should try to upgrade to 4.2.5 to see if it is the driver module or not.

pteroctopus, 2016-06-30:
Hi, You mean downgrade to 4.2.5? Because I'm on 4.4.0-24.

uname -a
Linux ubuntu 4.4.0-24-generic #43-Ubuntu SMP Wed Jun 8 19:27:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

joneslee85, 2016-06-30:
@pteroctopus sorry, my typo, it is 4.5.2

pteroctopus, 2016-07-01:
Installed the lastest kernel version. USB adapter didn't work at all:
uname -a
Linux ubuntu 4.6.3-040603-generic #201606241434 SMP Fri Jun 24 18:36:33 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

Then I installed lwfingers driver and it worked the same as before: less than half a minute and then the connection dropped.

Br,

joneslee85, 2016-07-01:
I think what you could do is to install a different distro, like Arch and test it out. If the symptom still persist, then it is hardware fault.

tnvrsingh, 2016-07-24:
I have the same problem on both 14.04 and 16.04. I got it to work on 14.04 but not on 16.04.

Followed the above mentioned guide and many others on the Ubuntu forums.

Its not a hardware fault because it works perfectly on Windows 10 and Windows 8.1 ( although the company provides these drivers).

ssnikiforov, 2016-11-10:
I have the same problem. Ubuntu 16.04, kernel 4.4.0-47-generic.

BruceBuckland, 2016-12-16:
I had a problem where I insert the TL-WN725N and Ubuntu finds it, and I connect to the network, and I get slow upload and download. (1mbps up, 4ish mbps down).

Then I reboot and I get 25mbps down and 11mbps up.

Other machines on the network get 90 mbps down and 11 mbps up (but I think the fact that this adapter does not do 802.11ac is why it is slower on the download).

I have NOT had the problem of the network adapter slowing... Machine has been up for 15 minutes... still reasonably performing.  After it has been up for a day I will try unplug replug to see if i get the problem again.

echipachenko, 2018-02-14:
The same problem
Linux ***** 4.13.0-35-generic #39-Ubuntu SMP Mon Feb 12 11:06:39 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

rrjanbiah, 2021-08-04:
Getting occasional disconnects and also speed is almost half (10mbps vs 20mbps)

4.15.0-142-generic #146~16.04.1-Ubuntu SMP Tue Apr 13 09:27:15 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux