lwfinger/rtl8188eu #181: can not work on centos7

onsix9, 2016-08-25:
i've installed the driver,but ,when i run command :iw dev . it shows 'nl80211 not found', so ,what can i do ,how can i configure and connect to WIFI ?thanks .

mes:
1. $ lsusb
Bus 001 Device 003: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
2.$ ip addr
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN 
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host 
       valid_lft forever preferred_lft forever
2: enp3s0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP qlen 1000
    link/ether fc:aa:14:d9:81:f0 brd ff:ff:ff:ff:ff:ff
    inet 192.168.2.123/24 brd 192.168.2.255 scope global dynamic enp3s0
       valid_lft 3516sec preferred_lft 3516sec
    inet6 fe80::feaa:14ff:fed9:81f0/64 scope link 
       valid_lft forever preferred_lft forever
3: wlp0s26u1u2: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 48:8a:d2:1d:82:59 brd ff:ff:ff:ff:ff:ff

3.$ iwconfig
lo        no wireless extensions.

virbr0-nic  no wireless extensions.

virbr0    no wireless extensions.

enp3s0    no wireless extensions.

wlp0s26u1u2  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lwfinger, 2016-08-25:
Your kernel configuration is faulty. You have not configured some of the parts needed for wifi.

onsix9, 2016-08-28:
what should i do ,after i did this :modprobe cfg80211 . then 'iw dev' or 'iw list' etc ,there are no outputs .
what is wrong with ,,which ? kernel or my wifi adapter ? how can i fix it ?thank you .

lwfinger, 2016-08-28:
Module cfg80211 is NOT the wireless driver. Have you loaded r8188eu? It seems that the answer is no.

onsix9, 2016-08-29:
i run 'lsmod |grep 8188' , output :8188eu .i don't know whether it is the wireless driver which i have compiled and installed .if it is ,it has been loaded on, or something wrong with the name ,r8188eu or 8188eu? thank you ; and 'modprobe r8188eu' shows that module not found

lwfinger, 2016-08-29:
The one in the kernel is r8188eu. The GitHub version is 8188eu.

I have no idea what is your problem, but it looks as if the ARCH user-space routines are at fault. Are you running NetworkManager, or some other code?

onsix9, 2016-08-30:
so ,it's that ive installed github's wireless driver and it is loaded on as
a name of '8188eu'; i do the same steps on github on CentOS6 and it works
,by the way ,it is managed by NetworkManager on centos6 ;
i runned NetworkManager on centos7 ,but i can't see the item of WIFI
connection ;i do nothing else code . shall i stop NetworkManager? and
except it what's other ways to configure WIFI ,it will be better if it is
graphic tools ,if not ,command line also ok,, , have some tutorials ,thanks

lwfinger, 2016-08-30:
It seems that Centos7 is the problem. I cannot debug that. You need to contact them about the problem.

onsix9, 2016-08-31:
i installed NetworkManager-wifi ,but NetworkManager can not run ; and
command:iw dev or iw list ,it doesn't show 'nl80211 not found' any more
,but still no outputs .

command result about NM below:
$ service NetworkManager status
Redirecting to /bin/systemctl status  NetworkManager.service
● NetworkManager.service - Network Manager
   Loaded: loaded (/usr/lib/systemd/system/NetworkManager.service; enabled;
vendor preset: enabled)
   Active: failed (Result: start-limit) since Wed 2016-08-31 08:46:03 HKT;
25min ago
  Process: 3536 ExecStart=/usr/sbin/NetworkManager --no-daemon
(code=dumped, signal=SEGV)
 Main PID: 3536 (code=dumped, signal=SEGV)

if NM can run ,i can configure WIFI on Gnome Desktop with "Network
Connections" ,but now ,there is nothing in it ,maybe because NM is not
running;

it also shows this:
# iw wlp0s29u1u3 scan | grep SSID
command failed: No such device (-19)

i'm going crazy ;

lwfinger, 2016-08-31:
Don't ask me. I do not use ARCH, and I never will!!!! That is where the problem is.

You need to look at the dmesg output to see why loading the driver is failing. If there are no errors, then ARCH user-space is completely broken.

onsix9, 2016-09-09:
no ,i'm not using ARCH ,I'm using CentOS 7 OS

lwfinger, 2016-09-09:
The same is true for Centos.

zppro, 2016-09-14:
my centos7 is  the same  problem,@onsix9 ,do you solve ?

dennisit, 2017-02-03:
I have the same problem！ + 1@

samlabs821, 2018-05-02:
I think centos7 not supported....

zcooler, 2018-05-03:
@samlabs821 try ti install kernel 4.x
On kernel 4.15 I`ve managed to run it