lwfinger/rtl8188eu #186: tl-wn725N not working on kernel linux v 3.17

smibcn, 2016-10-05:
Hello, 
I'm trying to set up my tl-wn725n v2 using the driver rtl8188eu in a linux kernel (3.17 version) compiled using yocto dizzy. The architecture is an arm am335x which is similar to Beagle Bone Black.
The steps I've followed are the next: 
1-Generate linux configuration using menuconfig:

         CONFIG_WIRELESS=y
         CONFIG_WIRELESS_EXT=y
         CONFIG_WEXT_CORE=y
         CONFIG_WEXT_PROC=y
         CONFIG_WEXT_SPY=y
         CONFIG_WEXT_PRIV=y
         CONFIG_CFG80211=y
         CONFIG_CFG80211_WEXT=y
         CONFIG_LIB80211=y
         CONFIG_LIB80211_CRYPT_WEP=y
         CONFIG_LIB80211_CRYPT_CCMP=y
         CONFIG_LIB80211_CRYPT_TKIP=y
         CONFIG_MAC80211=y
         CONFIG_MAC80211_HAS_RC=y
         CONFIG_MAC80211_RC_MINSTREL=y
         CONFIG_MAC80211_RC_MINSTREL_HT=y
         CONFIG_MAC80211_RC_DEFAULT_MINSTREL=y
         CONFIG_MAC80211_RC_DEFAULT="minstrel_ht"
         
         CONFIG_RFKILL=y
         CONFIG_RFKILL_LEDS=y
         CONFIG_RFKILL_REGULATOR=y
         
         CONFIG_WLAN=y
         CONFIG_HOSTAP=m
         CONFIG_HOSTAP_FIRMWARE=y
         CONFIG_RTL_CARDS=m
         
          CONFIG_STAGING=y
         CONFIG_R8188EU=m
         CONFIG_88EU_AP_MODE=y
         CONFIG_WIRELESS=y
         CONFIG_WIRELESS_EXT=y
         CONFIG_WEXT_CORE=y
         CONFIG_WEXT_PROC=y
         CONFIG_WEXT_SPY=y
         CONFIG_WEXT_PRIV=y
         CONFIG_CFG80211=y
         CONFIG_CFG80211_WEXT=y
         CONFIG_LIB80211=y
         CONFIG_LIB80211_CRYPT_WEP=y
         CONFIG_LIB80211_CRYPT_CCMP=y
         CONFIG_LIB80211_CRYPT_TKIP=y
         CONFIG_MAC80211=y
         CONFIG_MAC80211_HAS_RC=y
         CONFIG_MAC80211_RC_MINSTREL=y
         CONFIG_MAC80211_RC_MINSTREL_HT=y
         CONFIG_MAC80211_RC_DEFAULT_MINSTREL=y
         CONFIG_MAC80211_RC_DEFAULT="minstrel_ht" 
         CONFIG_RFKILL=y
         CONFIG_RFKILL_LEDS=y
         CONFIG_RFKILL_REGULATOR=y
         
         CONFIG_WLAN=y
         CONFIG_HOSTAP=m
         CONFIG_HOSTAP_FIRMWARE=y
         CONFIG_RTL_CARDS=m
         
          CONFIG_STAGING=y
         CONFIG_R8188EU=m
         CONFIG_88EU_AP_MODE=y
         
After that, I've added the next packages at my image (.bb recipe):
         iw wpa-supplicant hostapd 

Once, the embedded system is built and is booting, the systems shows the next messages:

         [    2.742724] usb 2-1: new high-speed USB device number 2 using musb-hdrc
         [    2.904674] usb 2-1: device v0bda p8179 is not supported
         [    2.910263] usb 2-1: New USB device found, idVendor=0bda, idProduct=8179
         [    2.917440] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
         [    2.924960] usb 2-1: Product: 802.11n NIC
         [    2.929164] usb 2-1: Manufacturer: Realtek
         [    2.933491] usb 2-1: SerialNumber: 00E04C0001
         [    7.313640] r8188eu: module is from the staging directory, the quality is unknown, you have been warned.
         [    7.509115] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
         [    7.779753] usbcore: registered new interface driver r8188eu

when I try to up the wlan0 appears the next message (ifconfig wlan0 up):

         [  102.413361] R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1

         [  102.452272] 

         [  102.453873] =================================

         [  102.458441] [ INFO: inconsistent lock state ]

         [  102.463018] 3.17.0-dirty #1 Tainted: G         C    

         [  102.468221] ---------------------------------

         [  102.472791] inconsistent {HARDIRQ-ON-W} -> {IN-HARDIRQ-W} usage.

         [  102.479083] ifconfig/455 [HC1[1]:SC0[0]:HE0:SE1] takes:

         [  102.484553]  (&(&(pqueue->lock))->rlock){?.....}, at: [<bf027ae8>] rtw_free_xmitbuf+0x54/0xd8 [r8188eu]

         ...

         MAC Address = xx:xx:xx:xx:xx:xx

iwconfig output:

          wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   

          Sensitivity:0/0  

          Retry:off   RTS thr:off   Fragment thr:off

          Encryption key:off

          Power Management:off

          Link Quality:0  Signal level:0  Noise level:0

          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0

          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

My _etc/network/interfaces_ configuration is: 

          iface wlan0 inet dhcp
                    wireless_mode managed
                    wireless_essid My_Wifi
                    wpa-driver wext
          wpa-conf /etc/wpa_supplicant.conf

and the wpa_supplicant.conf is: 

          ctrl_interface=/var/run/wpa_supplicant
          ctrl_interface_group=0
          update_config=1

          network={
                    ssid="SP_ID"
                    sscan_id=1
                    proto=RSN
                    key_mgmt=WPA-PSK
                    pairwise=CCMP TKIP
                    #psk="xxxxxxx"
                    psk=long_number_obtained from wpa_pass_phrase
          }

The best thing is that i'm possible to scan all the avalaibable wifis using the _iwlist wlan0 scan_, but not i'm able to connect it.

Can Anyone help me? If it is necessary more information, I'm glad to post it.

Thanks for your replays!

CGarces, 2016-10-30:
Have you tried to compile an older version.?
Looks like fcc9034b0d2170b4942133a43a9521aab2e0beac was working fine on legacy kernels.

smibcn, 2016-11-15:
Hi! 
currently, It is working fine and I didn't have any problem yet. However, it worries me the message about it can enter in a DEADLOCK state. 
I'm aware that is a warning, but I don't know how to handle it and if it can be a problem in the future.
rtlu8188eu firmware is what you can find in the linux kernel version 3.17. As you can see above:

R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1

Thanks for your replay.