lwfinger/rtl8188eu #187: How about v4.3.0.8_13968.20150417 codeline?

macmpi, 2016-10-13:
One can find a more advanced version in source code on [TP-Link website](http://www.tp-link.com/il/download/TL-WN725N.html#Driver): it very much seems a legit source.
Wouldn't it make sense to have it here?

lwfinger, 2016-10-13:
Sure. Why don't you submit a patch?

macmpi, 2016-10-13:
sorry but dunno how to do it :(

lwfinger, 2016-10-14:
OK. I have no time or interest. This driver will soon be replaced by the mainline driver named rtl8xxxu.

closed 2016-10-14 by lwfinger

macmpi, 2016-10-14:
Great news, thanks: any estimated timeframe?
Does it mean rtl8xxxu will now be hostapd-friendly on 4.4.x?

Indeed, v4.3.0.8_13968 works with base hostapd on 4.4.x (nl80211 driver), so WPA2 authentication is fine, without need of a custom hostapd version anymore.
I hope rtl8xxxu will bring similar benefit.

Thanks for any thought.

lwfinger, 2016-10-14:
Driver rtl8xxxu has full mac80211, cfg80211, and nl80211 capabilities. It works with any hostapd that uses dbus.

macmpi, 2016-10-14:
Great, thanks for all the hard work!
Can't wait the release

lwfinger, 2016-10-14:
I did not do it - Jes Soensen from RedHat is the author. Kernel 4.8 is already released.

macmpi, 2016-10-23:
Any chance I can compile rtl8xxxu on 4.4.y and get support for 8188eu on top of currently available Raspian ?
Or it does much depend on critical kernel updates between 4.4 and 4.8, and not worth it?
I fear kernel 4.8 may take some time to come to Raspian (is 4.8 the earliers one?). 
Thanks.