lwfinger/rtl8188eu #189: Driver does not support authentication/assocation or connect commands

clightcap, 2016-11-30:
Good afternoon @lwfinger ,

This is more of a question than an issue. I'm trying to use the RTL8188EU driver with my AM3351 board running Linux kernel 4.4.12 from ti-processor-sdk 3.00.00.04 but it seems the driver is not working properly. First, is the RTL8188EU driver found in the kernel config drivers/staging section identical to this driver?

The module loads when I plug in my device, and I'm able to see my wlan0 device. The lsusb command also properly shows my device. I'm also able to scan for WiFi networks and see my local AP, so the driver must be working to some extent. However, I'm not able to connect to my access point with WPA authentication.

The error message indicates that the driver does not support authentication/association or connect commands. Is there a kernel config option that I'm overlooking? Please let me know if you have any suggestions.

root@am335x-evm:~# iwconfig
wlan0 unassociated Nickname:"<WIFI@REALTEK>"
Mode:Auto Frequency=2.412 GHz Access Point: Not-Associated 
Sensitivity:0/0 
Retry:off RTS thr:off Fragment thr:off
Encryption key:off
Power Management:off
Link Quality:0 Signal level:0 Noise level:0
Rx invalid nwid:0 Rx invalid crypt:0 Rx invalid frag:0
Tx excessive retries:0 Invalid misc:0 Missed beacon:0

sit0 no wireless extensions.

lo no wireless extensions.

root@am335x-evm:~# ifconfig
lo Link encap:Local Loopback 
inet addr:127.0.0.1 Mask:255.0.0.0
inet6 addr: ::1%132976/128 Scope:Host
UP LOOPBACK RUNNING MTU:65536 Metric:1
RX packets:160 errors:0 dropped:0 overruns:0 frame:0
TX packets:160 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1 
RX bytes:12160 (11.8 KiB) TX bytes:12160 (11.8 KiB)

wlan0 Link encap:Ethernet HWaddr 40:A5:EF:0E:F1:C7 
UP BROADCAST MULTICAST MTU:1500 Metric:1
RX packets:0 errors:0 dropped:0 overruns:0 frame:0
TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
collisions:0 txqueuelen:1000 
RX bytes:0 (0.0 B) TX bytes:0 (0.0 B)

root@am335x-evm:~# wpa_supplicant -D nl80211,wext -i wlan0 -c /etc/wpa_zc.conf -dd 
wpa_supplicant v2.5-devel
random: Trying to read entropy from /dev/random
Successfully initialized wpa_supplicant
Initializing interface 'wlan0' conf '/etc/wpa_zc.conf' driver 'nl80211,wext' ctrl_interface 'N/A' bridge 'N/A'
Configuration file '/etc/wpa_zc.conf' -> '/etc/wpa_zc.conf'
Reading configuration file '/etc/wpa_zc.conf'
Line: 1 - start of a new network block
ssid - hexdump_ascii(len=10):
41 44 45 58 20 47 55 45 53 54 GUEST 
PSK - hexdump(len=32): [REMOVED]
Priority group 0
id=0 ssid='GUEST'
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 3 iftype 2 (STATION)
nl80211: Failed to set interface 3 to mode 2: -19 (No such device)
wlan0: Failed to initialize driver interface - try next driver wrapper
wlan0: Unsupported driver 'wext'
Failed to add interface wlan0
wlan0: Cancelling scan request
wlan0: Cancelling authentication timeout

Thank you for your support!

Best regards,
Chris

lwfinger, 2016-11-30:
Depending on which branch of this repo you are using, the code may be essentially that of the kernel, or it may be very different. The master branch matches the kernel - neither support any nl80211 commands. Branch v4.1.8_9499 does support those commands.

Both branches do support connection with NetworkManager controlling the interface. I think there is something wrong with your user-space routines.

clightcap, 2016-11-30:
I'm using the master branch. Should I be using the wext driver instead of nl80211?

lwfinger, 2016-11-30:
The nl80211 driver will certainly not work with the master branch.

clightcap, 2016-12-01:
Hi @lwfinger ,
Thank you for your feedback. I realize this is more a debugging question than an issue, and I appreciate your support.

In that case, would you recommend I use wpa_supplicant with wext driver option or switch to using networkmanager or nmcli? I'm seeing this output message from wpa_supplicant

wlan0: Unsupported driver 'wext'

Does this mean r8188eu driver does not support wext or wpa_supplicant? It seems that wext is not supported by my wpa_supplicant, using wpa_supplicant -h
                                                                                                                                                                                           
usage:                                                                                                                                                                                     
  wpa_supplicant [-BddhKLqqtvW] [-P<pid file>] [-g<global ctrl>] \ 
...
drivers:                                                                                                                                                                                   
  nl80211 = Linux nl80211/cfg80211                                                                                                                                                         
options:                                                                                                                                                                                   
  -b = optional bridge interface name                                                                                                                                                      
  -B = run daemon in the background   
...

Do you have any recommendations on using wpa_supplicant or nmcli with r8188eu module? I'm able to list nearby APs using iwlist, so I believe fundamentally the module is working properly, but I'm not able to establish a connection.

Thanks again,
Chris

clightcap, 2016-12-01:
In addition to the options of using wpa_supplicant with wext driver or nmcli, I could also switch to branch v4.1.8_9499 and using wpa_supplicant with nl80211 driver. Which of these three options do you recommend?

lwfinger, 2016-12-01:
To use wext with wpa_supplicant, you will need to dig up an old wpa_supplicant that supports that driver. Who knows what other problems there would be? The other two options are about equally easy to implement. You take your choice.

closed 2016-12-01 by lwfinger

clightcap, 2016-12-01:
Thanks for the feedback!

lucumt, 2018-03-18:
@clightcap  have you solved it?I face the same issue!

hxxful, 2018-04-09:
Allwinner H5, Linux4.14.0, hostapd2.6 rtl8188eu, also face the same issue
how to solve this?

jostidetosti, 2018-09-30:
I'm also looking for a solution to this.

lwfinger, 2018-09-30:
What driver are you using - the one in the kernel, or the one from this repo? If the latter, which branch? Keep in mind that, as I clearly stated earlier, that the master branch will NOT support any nl80211 commands. Only branch v4.1.8_9499 has nl80211/cfg80211 support built in.