lwfinger/rtl8188eu #190: Can't connect to hidden SSID with WPA2

MikeVRT, 2016-12-06:
Can't connect to hidden SSID with WPA2. If I unhide that SSID - connection works.
If connect to unhidden SSID, remember it connection, than hide SSID - connection not works too.
Complied last stable version for now -  Ubuntu 16.04 LTE.

smibcn, 2017-01-10:
Hello, 

I don't know I've you have already solve your problem. 

I had a similar issue. I was using the driver version added on linux kernel 3.17 when i realised that it last more than 10 minutes to connect with a hidden wifi (initially I tought that the problem was the same as you). 
I tried a lot of different options, however, what worked for me was to change the driver version. I get the driver from linux kernel 4.4.39 and I used with linux kernel 3.17.
I don't why but it worked for me.
I hope you'll solve your problem as soon as possible.

MrEngman, 2017-01-10:
You need to set up wpa_supplicant.conf correctly. Something like 
```
country=GB
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
ap_scan=1

network={
        ssid="network-name"
        psk="password"
        scan_ssid=1
        bssid=AP-MAC-Address
}
```

smibcn, 2017-01-10:
MrEngman,
Your are right!!! I had the problem even using wpa_supplicant. That's my answer was focus in other issue.
Obviously, The first step to test the connection to a hidden wifi is to use wpa_supplicant.conf.

MrEngman, 2017-01-10:
The important bits are ap_scan, scan_ssid and bssid. These should then get the wifi to look for your particular AP when it doesn't transmit it's ssid.