lwfinger/rtl8188eu #193: system reboots when connecting usb device after installing rtl8188eu

Pintglass, 2017-01-16:
I have compiled your driver for my alfa r36 which is running openwrt 15.05 kernel version 3.18.44
After insmod rtl8188eu i connect my awus036nhv but system immediately reboots, i have tried to capture any dmesg entries when the crash happens but nothing is logged.
The r36 is mips based, could this be my issue?
Platform: Ralink RT3050FTarget: ramips
Instruction Set: MIPS32
Sub Instruction Set: MIPS32 24K/E series
I have tried the driver that is in the kernel, which doesn't crash my machine but the only message i get from the logs is  
usb 1-1: new high-speed USB device number 2 using dwc2
 usb 1-1: no of_node; not parsing pinctrl DT
after that i get nothing?
Just one more question when i compile your driver i get a file size of about 1041KB does that seem about right?

lwfinger, 2017-01-17:
The problem could be something to do with the MIPS architecture. Is that MIPS hardware little- or big-endian?

I have no idea about MIPS object size. On x86_64, the .ko file is 667 KB, thus the size of your driver is about right.

Pintglass, 2017-01-17:
It is little endian.
Is the driver that is included with openwrt the same as yours as the file size of that is only 500KB, the reason i ask is i'm just wondering if i'm not compiling correctly?
I do seem to of made some progress as im now getting the following when i connect the usb device
usb 1-1: new high-speed USB device number 3 using dwc2
usb 1-1: no of_node; not parsing pinctrl DT
r8188eu 1-1:1.0: no of_node; not parsing pinctrl DT
Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer (0)

I do have the frimware in /lib/firmware/rtlwifi

Pintglass, 2017-01-20:
I have connected to the serial port of my r36 to see if i could get some more information about what happens  when i connect my awus036nhv with your driver installed, i can post the logs if you wouldnt mind taking a look at them, i dont want to just post masses of log entries if they are of no use to you.

lwfinger, 2017-01-21:
Put the logs on pastebin.com or some similar site and post the link here.

Pintglass, 2017-01-21:
I've posted them on pastebin link is http://pastebin.com/MtWa1sG7

lwfinger, 2017-01-21:
That dump indicates that the NULL pointer dereference happened in _rtw_malloc(), but that routine never accesses any pointers. I conclude that something has overwritten that section of code, and that the real problem has already been triggered; however, I have no idea what might be causing it.

Pintglass, 2017-01-21:
This is probably a stupid question, but should i be using the hostapd that is included in your source?

lwfinger, 2017-01-21:
The standard one likely uses the nl80211 interface that does not work with the that calling sequences.

Pintglass, 2017-01-21:
Could that be why i'm having issues, or is the issue i have due to me using your driver on a mips architecture?

lwfinger, 2017-01-21:
I think that the MIPS architecture is more likely to be the problem. You have not really gotten far enough for hostapd to be a problem.

Pintglass, 2017-01-22:
Is there anything i can write into you code to try and see how far it gets?

Pintglass, 2017-01-23:
Have you or snbody else ever got a rtl8188eu to work with openwrt? as I can't seen to find any evidence of anybody who's posted on your site or  anywhere else on the net being successful even on a X86 setup.

lwfinger, 2017-01-23:
The printk() command is what you use, but I'm not sure where to put such commands.

I do not know whether it has worked with openWRT, but I think it has.

Which version of openWRT are you using? I will try to download that version for x86 and try it in a VirtualBox VM.

Pintglass, 2017-01-23:
I'm using openwrt version 15.05 with a kennel version off 3.18.45. I've seen reports of it working on the rpi but never with openwrt.

Pintglass, 2017-01-29:
Did you get chance to test support for your driver in openwrt on a x86 setup