lwfinger/rtl8188eu #194: Oops near `rtw_txframes_sta_ac_pending`

nisargjhaveri, 2017-01-20:
```
Jan 20 23:53:23 raspberrypi kernel: [96233.752818] Unable to handle kernel NULL pointer dereference at virtual address 00000064
Jan 20 23:53:23 raspberrypi kernel: [96233.765338] pgd = c0004000
Jan 20 23:53:23 raspberrypi kernel: [96233.770290] [00000064] *pgd=00000000
Jan 20 23:53:23 raspberrypi kernel: [96233.775945] Internal error: Oops: 17 [#1] ARM
Jan 20 23:53:23 raspberrypi kernel: [96233.782353] Modules linked in: bridge stp llc ip6table_filter ip6_tables ipt_MASQUERADE nf_nat_masquerade_ipv4 iptable_nat
 nf_conntrack_ipv4 nf_defrag_ipv4 8188eu(O) asix nf_nat_ipv4 nf_nat nf_conntrack libphy iptable_filter ip_tables x_tables snd_bcm2835 snd_pcm snd_timer snd bcm28
35_wdt bcm2835_gpiomem uio_pdrv_genirq uio ipv6
Jan 20 23:53:23 raspberrypi kernel: [96233.823788] CPU: 0 PID: 425 Comm: RTW_CMD_THREAD Tainted: G           O    4.4.38+ #938
Jan 20 23:53:23 raspberrypi kernel: [96233.837033] Hardware name: BCM2708
Jan 20 23:53:23 raspberrypi kernel: [96233.843134] task: d8888000 ti: d88ec000 task.ti: d88ec000
Jan 20 23:53:23 raspberrypi kernel: [96233.852218] PC is at rtw_txframes_sta_ac_pending+0x64/0x6c [8188eu]
Jan 20 23:53:23 raspberrypi kernel: [96233.862181] LR is at rtl8188eu_hal_xmit+0x40/0x1a4 [8188eu]
Jan 20 23:53:23 raspberrypi kernel: [96233.870745] pc : [<bf188960>]    lr : [<bf1ae0a8>]    psr: 60000013
Jan 20 23:53:23 raspberrypi kernel: [96233.870745] sp : d88edeb0  ip : d88edec0  fp : d88edebc
Jan 20 23:53:23 raspberrypi kernel: [96233.888115] r10: 00000000  r9 : 00000011  r8 : dcf7a0c8
Jan 20 23:53:23 raspberrypi kernel: [96233.896299] r7 : dcf62d04  r6 : dcf3d468  r5 : dcf62cfc  r4 : dcf3c000
Jan 20 23:53:23 raspberrypi kernel: [96233.905845] r3 : 00000054  r2 : 00000000  r1 : dcf62d04  r0 : dcf3c000
Jan 20 23:53:23 raspberrypi kernel: [96233.915290] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
Jan 20 23:53:23 raspberrypi kernel: [96233.925526] Control: 00c5387d  Table: 1a650008  DAC: 00000055
Jan 20 23:53:23 raspberrypi kernel: [96233.934250] Process RTW_CMD_THREAD (pid: 425, stack limit = 0xd88ec188)
Jan 20 23:53:23 raspberrypi kernel: [96233.943938] Stack: (0xd88edeb0 to 0xd88ee000)
Jan 20 23:53:23 raspberrypi kernel: [96233.951370] dea0:                                     d88ededc d88edec0 bf1ae0a8 bf188908
Jan 20 23:53:23 raspberrypi kernel: [96233.965546] dec0: dcf62cfc dcf6220c dcf7a004 00000001 d88edeec d88edee0 bf197ff0 bf1ae074
Jan 20 23:53:23 raspberrypi kernel: [96233.979657] dee0: d88edf24 d88edef0 bf174e78 bf197fd4 60000013 dcf3c000 dcf3d3a0 da518140
Jan 20 23:53:23 raspberrypi kernel: [96233.993792] df00: dcf3c000 dcf3d3a0 bf1c8be8 bf1eef78 d88c9200 bf1cdc70 d88edf64 d88edf28
Jan 20 23:53:23 raspberrypi kernel: [96234.008103] df20: bf154b20 bf174d7c d8ac99a0 dcf3c000 bf15495c dcf3d3ac 00000000 00000000
Jan 20 23:53:23 raspberrypi kernel: [96234.022683] df40: d8ac99a0 dcf3c000 bf15495c 00000000 00000000 00000000 d88edfac d88edf68
Jan 20 23:53:23 raspberrypi kernel: [96234.037462] df60: c003d2e8 bf154968 d88ec000 00000000 d8ac99a0 dcf3c000 00000000 d88edf7c
Jan 20 23:53:23 raspberrypi kernel: [96234.052497] df80: d88edf7c 00000000 d88edf88 d88edf88 d8ac99a0 c003d20c 00000000 00000000
Jan 20 23:53:23 raspberrypi kernel: [96234.067854] dfa0: 00000000 d88edfb0 c000f8a8 c003d218 00000000 00000000 00000000 00000000
Jan 20 23:53:23 raspberrypi kernel: [96234.083369] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Jan 20 23:53:23 raspberrypi kernel: [96234.099006] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000 00000000 00000000
Jan 20 23:53:23 raspberrypi kernel: [96234.116198] [<bf188960>] (rtw_txframes_sta_ac_pending [8188eu]) from [<bf1ae0a8>] (rtl8188eu_hal_xmit+0x40/0x1a4 [8188eu])
Jan 20 23:53:23 raspberrypi kernel: [96234.136464] [<bf1ae0a8>] (rtl8188eu_hal_xmit [8188eu]) from [<bf197ff0>] (rtw_hal_xmit+0x28/0x34 [8188eu])
Jan 20 23:53:23 raspberrypi kernel: [96234.155282] [<bf197ff0>] (rtw_hal_xmit [8188eu]) from [<bf174e78>] (tx_beacon_hdl+0x108/0x170 [8188eu])
Jan 20 23:53:23 raspberrypi kernel: [96234.173802] [<bf174e78>] (tx_beacon_hdl [8188eu]) from [<bf154b20>] (rtw_cmd_thread+0x1c4/0x30c [8188eu])
Jan 20 23:53:23 raspberrypi kernel: [96234.191948] [<bf154b20>] (rtw_cmd_thread [8188eu]) from [<c003d2e8>] (kthread+0xdc/0xf8)
Jan 20 23:53:23 raspberrypi kernel: [96234.207874] [<c003d2e8>] (kthread) from [<c000f8a8>] (ret_from_fork+0x14/0x2c)
Jan 20 23:53:23 raspberrypi kernel: [96234.219078] Code: ea000002 e2823054 ea000000 e2823018 (e5930010) 
Jan 20 23:53:23 raspberrypi kernel: [96234.306971] ---[ end trace e4938cf0047493b6 ]---
```

On master branch.
```
$ uname -a
Linux raspberrypi 4.4.38+ #938 Thu Dec 15 15:17:54 GMT 2016 armv6l GNU/Linux
```

lwfinger, 2017-01-21:
If it is not installed, please install gdb. Then do the following:

gdb 8188eu.ko
l *rtw_txframes_sta_ac_pending+0x64
quit

The second line starts with el space star. Post the output from gdb. That will tell me which line is failing.

I also found a place that might be the problem. Please pull and try the new version.

nisargjhaveri, 2017-01-21:
```
$ gdb 8188eu.ko
...
Reading symbols from 8188eu.ko...(no debugging symbols found)...done.
(gdb) l *rtw_txframes_sta_ac_pending+0x64
No symbol table is loaded.  Use the "file" command.
```

nisargjhaveri, 2017-01-21:
I have it installed using dkms, if that has anything to do with `no debugging symbols found`.

lwfinger, 2017-01-21:
Using dkms is not the problem. Your system is compiled with debugging disables, which is common for systems with limited RAM.

Please see if it happens again with my recent changes.

nisargjhaveri, 2017-01-21:
Sure. I'll try the new version and see if anything happens again.

nisargjhaveri, 2017-03-30:
It happened again. Here is the stack trace.
```
Mar 30 15:48:37 raspberrypi kernel: [38904.745345] Unable to handle kernel NULL pointer dereference at virtual address 00000064
Mar 30 15:48:37 raspberrypi kernel: [38904.757847] pgd = c0004000
Mar 30 15:48:37 raspberrypi kernel: [38904.762745] [00000064] *pgd=00000000
Mar 30 15:48:37 raspberrypi kernel: [38904.768456] Internal error: Oops: 17 [#1] ARM
Mar 30 15:48:37 raspberrypi kernel: [38904.774863] Modules linked in: 8188eu(O) bridge stp llc ip6table_filter ip6_tables ipt_MASQUERADE nf_nat_masquerade_ipv4 i
ptable_nat asix nf_conntrack_ipv4 nf_defrag_ipv4 nf_nat_ipv4 nf_nat nf_conntrack iptable_filter libphy ip_tables x_tables snd_bcm2835 snd_pcm snd_timer snd bcm28
35_gpiomem bcm2835_wdt uio_pdrv_genirq uio ipv6 [last unloaded: 8188eu]
Mar 30 15:48:37 raspberrypi kernel: [38904.818408] CPU: 0 PID: 1754 Comm: RTW_CMD_THREAD Tainted: G           O    4.4.38+ #938
Mar 30 15:48:37 raspberrypi kernel: [38904.831712] Hardware name: BCM2708
Mar 30 15:48:37 raspberrypi kernel: [38904.837795] task: d88c5f60 ti: d89f4000 task.ti: d89f4000
Mar 30 15:48:37 raspberrypi kernel: [38904.846855] PC is at rtw_txframes_sta_ac_pending+0x68/0x74 [8188eu]
Mar 30 15:48:37 raspberrypi kernel: [38904.856807] LR is at rtl8188eu_hal_xmit+0x40/0x1a4 [8188eu]
Mar 30 15:48:37 raspberrypi kernel: [38904.865344] pc : [<bf2b4964>]    lr : [<bf2da0b0>]    psr: 20000013
Mar 30 15:48:37 raspberrypi kernel: [38904.865344] sp : d89f5eb0  ip : d89f5ec0  fp : d89f5ebc
Mar 30 15:48:37 raspberrypi kernel: [38904.882662] r10: 00000000  r9 : 00000011  r8 : dd26e0c8
Mar 30 15:48:37 raspberrypi kernel: [38904.890823] r7 : dd25c254  r6 : dd231468  r5 : dd25c24c  r4 : dd230000
Mar 30 15:48:37 raspberrypi kernel: [38904.900344] r3 : 00000054  r2 : 00000000  r1 : dd25c254  r0 : dd230000
Mar 30 15:48:37 raspberrypi kernel: [38904.909761] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
Mar 30 15:48:37 raspberrypi kernel: [38904.919964] Control: 00c5387d  Table: 18908008  DAC: 00000055
Mar 30 15:48:37 raspberrypi kernel: [38904.928654] Process RTW_CMD_THREAD (pid: 1754, stack limit = 0xd89f4188)
Mar 30 15:48:37 raspberrypi kernel: [38904.938399] Stack: (0xd89f5eb0 to 0xd89f6000)
Mar 30 15:48:37 raspberrypi kernel: [38904.945809] 5ea0:                                     d89f5edc d89f5ec0 bf2da0b0 bf2b4908
Mar 30 15:48:37 raspberrypi kernel: [38904.959928] 5ec0: dd25c24c dd25c364 dd26e004 00000001 d89f5eec d89f5ee0 bf2c3ff8 bf2da07c
Mar 30 15:48:37 raspberrypi kernel: [38904.974004] 5ee0: d89f5f24 d89f5ef0 bf2a0e78 bf2c3fdc 60000013 dd230000 dd2313a0 da639f40
Mar 30 15:48:37 raspberrypi kernel: [38904.988109] 5f00: dd230000 dd2313a0 bf2f4bf0 bf31af78 da747a00 bf2f9c78 d89f5f64 d89f5f28
Mar 30 15:48:37 raspberrypi kernel: [38905.002399] 5f20: bf280b20 bf2a0d7c d8b0c380 dd230000 bf28095c dd2313ac 00000000 00000000
Mar 30 15:48:37 raspberrypi kernel: [38905.016965] 5f40: d8b0c380 dd230000 bf28095c 00000000 00000000 00000000 d89f5fac d89f5f68
Mar 30 15:48:37 raspberrypi kernel: [38905.031737] 5f60: c003d2e8 bf280968 d89f4000 00000000 d8b0c380 dd230000 00000000 d89f5f7c
Mar 30 15:48:37 raspberrypi kernel: [38905.046762] 5f80: d89f5f7c 00000000 d89f5f88 d89f5f88 d8b0c380 c003d20c 00000000 00000000
Mar 30 15:48:37 raspberrypi kernel: [38905.062115] 5fa0: 00000000 d89f5fb0 c000f8a8 c003d218 00000000 00000000 00000000 00000000
Mar 30 15:48:37 raspberrypi kernel: [38905.077623] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mar 30 15:48:37 raspberrypi kernel: [38905.093255] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000 e3a03c01 e51b2040
Mar 30 15:48:37 raspberrypi kernel: [38905.110435] [<bf2b4964>] (rtw_txframes_sta_ac_pending [8188eu]) from [<bf2da0b0>] (rtl8188eu_hal_xmit+0x40/0x1a4 [8188eu])
Mar 30 15:48:37 raspberrypi kernel: [38905.130676] [<bf2da0b0>] (rtl8188eu_hal_xmit [8188eu]) from [<bf2c3ff8>] (rtw_hal_xmit+0x28/0x34 [8188eu])
Mar 30 15:48:37 raspberrypi kernel: [38905.149486] [<bf2c3ff8>] (rtw_hal_xmit [8188eu]) from [<bf2a0e78>] (tx_beacon_hdl+0x108/0x170 [8188eu])
Mar 30 15:48:37 raspberrypi kernel: [38905.167997] [<bf2a0e78>] (tx_beacon_hdl [8188eu]) from [<bf280b20>] (rtw_cmd_thread+0x1c4/0x30c [8188eu])
Mar 30 15:48:37 raspberrypi kernel: [38905.186140] [<bf280b20>] (rtw_cmd_thread [8188eu]) from [<c003d2e8>] (kthread+0xdc/0xf8)
Mar 30 15:48:37 raspberrypi kernel: [38905.202077] [<c003d2e8>] (kthread) from [<c000f8a8>] (ret_from_fork+0x14/0x2c)
Mar 30 15:48:37 raspberrypi kernel: [38905.213281] Code: e2823054 ea000000 e2823018 e3530000 (15930010) 
Mar 30 15:48:37 raspberrypi kernel: [38905.524490] ---[ end trace b43ab6e79b64ed0f ]---
```