lwfinger/rtl8188eu #195: Ping is not working

aneesalappatt, 2017-01-31:
I have used the master brach also tried with the ver 4.1.8 of my rtl8188ee device [My custom imx6 sabresd having the pcie to usb]. So the interface of the wifi device is usb. I installed the driver and i can configure the AP of my lab and can't able to get the ip address using udhcpc. So i can't able to ping the wlan0. My kernel version is 4.1.8.

lwfinger, 2017-01-31:
I do not understand how you are getting a USB interface with an RTL8188EE card. Of course, if you are not making a connection, ping will fail.

On some pastebin site, please post the output of the dmesg command, the output of "lspci -nn", and lsusb. With those, I may be able to help.

aneesalappatt, 2017-02-01:
Hello Sir,

I can successfully port the new rtl8188eu driver and I got the ping and the
ip using dhcp. I want to port the same driver into android 6.0 version.
Please letme how can I get this informations.

Anees

aneesalappatt, 2017-02-01:
Hello Sir,

Please see the dmesg output file.

Anees

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Initializing cgroup subsys cpu
[    0.000000] Initializing cgroup subsys cpuacct
[    0.000000] Linux version 4.1.15-226048-g082e62a-dirty (anees@anees-H61H2-M4) (gcc version 4.9.x-google 20140827 (prerelease) (GCC) ) #65 SMP PREEMPT Wed Feb 1 10:15:59 IST 2017
[    0.000000] CPU: ARMv7 Processor [412fc09a] revision 10 (ARMv7), cr=10c53c7d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: Freescale i.MX6 DualLite SABRE Smart Device Board
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 261888
[    0.000000] free_area_init_node: node 0, pgdat c11e1680, node_mem_map eeff7000
[    0.000000]   Normal zone: 1520 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 194560 pages, LIFO batch:31
[    0.000000]   HighMem zone: 67328 pages, LIFO batch:15
[    0.000000] PERCPU: Embedded 13 pages/cpu @eefc0000 s21952 r8192 d23104 u53248
[    0.000000] pcpu-alloc: s21952 r8192 d23104 u53248 alloc=13*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 260368
[    0.000000] Kernel command line: console=ttymxc3,115200 root=/dev/mmcblk1p2
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 1020320K/1047552K available (11558K kernel code, 943K rwdata, 3568K rodata, 1024K init, 628K bss, 27232K reserved, 0K cma-reserved, 269312K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xf0000000 - 0xff000000   ( 240 MB)
                   lowmem  : 0xc0000000 - 0xef800000   ( 760 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0xc0008000 - 0xc0fc59ac   (16119 kB)
                     .init : 0xc1000000 - 0xc1100000   (1024 kB)
                     .data : 0xc1100000 - 0xc11ebd98   ( 944 kB)
                      .bss : 0xc11ebd98 - 0xc1288e0c   ( 629 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] Preemptible hierarchical RCU implementation.
[    0.000000] 	Additional per-CPU info printed with stalls.
[    0.000000] 	RCU restricting CPUs from NR_CPUS=4 to nr_cpu_ids=2.
[    0.000000] RCU: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=2
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] L2C-310 erratum 769419 enabled
[    0.000000] L2C-310 enabling early BRESP for Cortex-A9
[    0.000000] L2C-310 full line of zeros enabled for Cortex-A9
[    0.000000] L2C-310 ID prefetch enabled, offset 16 lines
[    0.000000] L2C-310 dynamic clock gating enabled, standby mode enabled
[    0.000000] L2C-310 cache controller enabled, 16 ways, 512 kB
[    0.000000] L2C-310: CACHE_ID 0x410000c8, AUX_CTRL 0x76050001
[    0.000000] mxc_clocksource_init 3000000
[    0.000000] Switching to timer-based delay loop, resolution 333ns
[    0.000006] sched_clock: 32 bits at 3000kHz, resolution 333ns, wraps every 715827882841ns
[    0.000022] clocksource mxc_timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 637086815595 ns
[    0.000918] Console: colour dummy device 80x30
[    0.000933] Calibrating delay loop (skipped), value calculated using timer frequency.. 6.00 BogoMIPS (lpj=30000)
[    0.000946] pid_max: default: 32768 minimum: 301
[    0.001021] Security Framework initialized
[    0.001030] SELinux:  Initializing.
[    0.001059] SELinux:  Starting in permissive mode
[    0.001067] AppArmor: AppArmor disabled by boot time parameter
[    0.001101] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001111] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.001793] Initializing cgroup subsys freezer
[    0.001810] Initializing cgroup subsys debug
[    0.001836] CPU: Testing write buffer coherency: ok
[    0.002168] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[... 466 lines cut ...]
[   22.790587] usbcore: registered new interface driver r8188eu
[   22.983327] R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1
[   23.384668] MAC Address = d0:53:49:b3:c8:7d
[   23.392735] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   24.709299] R8188EU: INFO indicate disassoc
[   24.824205] R8188EU: INFO assoc success
[   24.849548] R8188EU: INFO indicate disassoc
[   24.926465] R8188EU: INFO assoc success
[   24.945125] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   24.964297] R8188EU: INFO indicate disassoc
[   25.028949] R8188EU: INFO assoc success
[   25.434064] cfg80211: Timeout while waiting for CRDA to reply, restoring regulatory settings
[   25.434086] cfg80211: Restoring regulatory settings
[   25.434104] cfg80211: Kicking the queue
[   28.584061] cfg80211: Timeout while waiting for CRDA to reply, restoring regulatory settings
[   28.584083] cfg80211: Restoring regulatory settings
[   28.584102] cfg80211: Kicking the queue
[   31.734041] cfg80211: Timeout while waiting for CRDA to reply, restoring regulatory settings
[   31.734063] cfg80211: Restoring regulatory settings
[   31.734081] cfg80211: Kicking the queue
[   34.442583] random: nonblocking pool is initialized
[   34.884078] cfg80211: Timeout while waiting for CRDA to reply, restoring regulatory settings
[   34.884102] cfg80211: Restoring regulatory settings
[   34.884121] cfg80211: Kicking the queue
[   94.884077] cfg80211: Verifying active interfaces after reg change

lwfinger, 2017-02-01:
Do you have CRDA installed? It does not appear to be available.

I still want to see the output of "lspci -nn" and "lsusb".

The kernel in Android is Linux. Once you open a command-line terminal, then I think installing the driver is the same as in Linux. If it is not the same, I do not know how to do it.

aneesalappatt, 2017-02-01:
Hello SIr,

My rtl8188eu driver is not stable. Its showing the following error and
can't able to connect.

root@imx6qdlsolo:/lib/modules# insmod 8188eu_working.ko
Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
EEPROM ID = 0x8129
usbcore: registered new interface driver r8188eu
root@imx6qdlsolo:/lib/modules# R8188EU: Firmware Version 11, SubVersion 1,
Signature 0x88e1
MAC Address = d0:53:49:b3:c8:7d
IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
random: nonblocking pool is initialized
R8188EU: INFO indicate disassoc
R8188EU: INFO indicate disassoc
R8188EU: INFO indicate disassoc
R8188EU: INFO indicate disassoc
R8188EU: INFO indicate disassoc

Please see the following command out put.

root@imx6qdlsolo:/lib/modules# lsusb
Bus 003 Device 001: ID 1d6b:0001
Bus 001 Device 001: ID 1d6b:0002
Bus 002 Device 002: ID 0bda:8179
Bus 002 Device 001: ID 1d6b:0002

root@imx6qdlsolo:/lib/modules# lspci -nn

lwfinger:-Do you have CRDA installed? It does not appear to be available?
Anees:- How to installed CRDA?

Anees

aneesalappatt, 2017-02-02:
Hello Sir,

I have ported the rtl8188eu in to android version 6.0 and I have the
following error. The same driver code was working fine with the Linux. My
android having the problem.

1|root@sabresd_6dq:/ # insmod /system/lib/modules/8188eu.ko
Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
EEPROM ID = 0x8129
usbcore: registered new interface driver r8188eu

root@sabresd_6dq:/system/lib/modules # ifconfig wlan0
up
r8188eu 2-1:1.0: Direct firmware load for rtlwifi/rtl8188eufw.bin failed
with error -2
r8188eu 2-1:1.0: Falling back to user helper
Request firmware failed with error 0xfffffff5
ifconfig: ioctl 8914: Operation not permitted

root@sabresd_6dq:/ # ifconfig
wlan0
wlan0     Link encap:Ethernet  HWaddr D0:53:49:B3:C8:7D
          BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 TX bytes:0

please let me know the comments.

lwfinger, 2017-02-02:
You do not need to quote my comments in a GitHub issue. Thet are always available.

I think you have not installed the firmware.

aneesalappatt, 2017-02-02:
Hi Sir,

I have copied the same rtl8188eufw.bin which is worked in linux system in
to the android /system/lib/firmware/rtlwifi/rtl8188eufw.bin. So I have
installed the firmware in to android. Please let me know any specified
folder of android to copy the firmware other than linux.

Anees

lwfinger, 2017-02-02:
Stop the quoting!!!!!!!!!!!!!

I know nothing of Android firmware loading. You will have to research that yourself.

aneesalappatt, 2017-02-08:
Hello Sir,

I have installed the rtl8188eu driver in to android 6.0 version and driver
is working. But when I connect the wpa_supplicant, the call not success. I
want to connect the rtl8188eu device to the AP of router and get the
ipaddress using dhcpcd. Pleasesee the log below.

130|root@sabresd_6dq:/system/lib/modules # logcat | grep
wpa
01-01 00:01:34.492  1424  1424 D wpa_supplicant: wpa_supplicant
v2.5-devel-6.0.1_rtw_r17190.20160415
01-01 00:01:34.492  1424  1424 D wpa_supplicant: random: Trying to read
entropy from /dev/random
01-01 00:01:34.492  1424  1424 I wpa_supplicant: Successfully initialized
wpa_supplicant
01-01 00:01:34.492  1424  1424 D wpa_supplicant: Initializing interface
'wlan0' conf '/data/misc/wifi/wpa_supplicant.conf' driver 'nl80211' ctrl
_interface 'N/A' bridge 'N/A'
01-01 00:01:34.492  1424  1424 D wpa_supplicant: Configuration file
'/data/misc/wifi/wpa_supplicant.conf' -> '/data/misc/wifi/wpa_supplicant.con
f'
01-01 00:01:34.493  1424  1424 D wpa_supplicant: Reading configuration file
'/data/misc/wifi/wpa_supplicant.conf'
01-01 00:01:34.502  1424  1424 D wpa_supplicant: ctrl_interface='wlan0'
01-01 00:01:34.502  1424  1424 D wpa_supplicant: update_config=1
01-01 00:01:34.502  1424  1424 D wpa_supplicant: device_name='sabresd_6dq'
01-01 00:01:34.502  1424  1424 D wpa_supplicant: manufacturer='Freescale'
01-01 00:01:34.502  1424  1424 D wpa_supplicant: model_name='SABRESD-MX6DQ'
01-01 00:01:34.502  1424  1424 D wpa_supplicant:
model_number='SABRESD-MX6DQ'
01-01 00:01:34.502  1424  1424 D wpa_supplicant: serial_number=''
01-01 00:01:34.502  1424  1424 D wpa_supplicant:
config_methods='physical_display virtual_push_button'
01-01 00:01:34.502  1424  1424 D wpa_supplicant: p2p_listen_reg_class=81
01-01 00:01:34.502  1424  1424 D wpa_supplicant: p2p_listen_channel=11
01-01 00:01:34.502  1424  1424 D wpa_supplicant: persistent_reconnect=1
01-01 00:01:34.502  1424  1424 D wpa_supplicant: Line: 25 - start of a new
network block
01-01 00:01:34.502  1424  1424 D wpa_supplicant: ssid - hexdump(len=12): 41
4c 43 4f 44 45 58 20 42 53 4e 4c
01-01 00:01:34.502  1424  1424 D wpa_supplicant: PSK (ASCII passphrase) -
hexdump(len=9): [REMOVED]
01-01 00:01:34.502  1424  1424 D wpa_supplicant: key_mgmt: 0x2
01-01 00:01:34.502  1424  1424 D wpa_supplicant: proto: 0x3
01-01 00:01:34.502  1424  1424 D wpa_supplicant: pairwise: 0x18
01-01 00:01:34.502  1424  1424 D wpa_supplicant: group: 0x18
01-01 00:01:34.539  1424  1424 D wpa_supplicant: PSK (from passphrase) -
hexdump(len=32): [REMOVED]
01-01 00:01:34.539  1424  1424 D wpa_supplicant: Priority group 0
01-01 00:01:34.539  1424  1424 D wpa_supplicant:    id=0 ssid='ALCODEX BSNL'
01-01 00:01:34.540  1424  1424 I wpa_supplicant: rfkill: Cannot open RFKILL
control device
01-01 00:01:34.540  1424  1424 D wpa_supplicant: nl80211: RFKILL status not
available
01-01 00:01:34.541  1424  1424 D wpa_supplicant: nl80211: Supported cipher
00-0f-ac:1
01-01 00:01:34.541  1424  1424 D wpa_supplicant: nl80211: Supported cipher
00-0f-ac:5
01-01 00:01:34.541  1424  1424 D wpa_supplicant: nl80211: Supported cipher
00-0f-ac:2
01-01 00:01:34.541  1424  1424 D wpa_supplicant: nl80211: Supported cipher
00-0f-ac:4
01-01 00:01:34.541  1424  1424 D wpa_supplicant: nl80211: Supported vendor
command: vendor_id=0x1018 subcmd=1
01-01 00:01:34.541  1424  1424 D wpa_supplicant: nl80211: Supported vendor
command: vendor_id=0x1a11 subcmd=4106
01-01 00:01:34.541  1424  1424 D wpa_supplicant: nl80211: Supported vendor
command: vendor_id=0x1a11 subcmd=4107
01-01 00:01:34.541  1424  1424 D wpa_supplicant: nl80211: Supported vendor
event: vendor_id=0x1018 subcmd=0
[... 146 lines cut ...]
notification - portEnabled=0
01-01 00:01:35.397  1424  1424 D wpa_supplicant: EAPOL: External
notification - portValid=0
01-01 00:01:35.456  1424  1424 D wpa_supplicant: wlan0: WPA: Clear old PMK
and PTK
01-01 00:01:35.456  1424  1424 D wpa_supplicant: wlan0: Cancelling scan
request
01-01 00:01:35.456  1424  1424 D wpa_supplicant: wlan0: Cancelling
authentication timeout
01-01 00:01:35.457  1424  1424 D wpa_supplicant: Remove interface wlan0
from radio phy1
01-01 00:01:35.457  1424  1424 D wpa_supplicant: Remove radio phy1
01-01 00:01:35.457  1424  1424 I wpa_supplicant: nl80211: deinit
ifname=wlan0 disabled_11b_rates=0
01-01 00:01:35.486  1424  1424 D wpa_supplicant: nl80211: Remove monitor
interface: refcount=0
01-01 00:01:35.486  1424  1424 D wpa_supplicant: netlink: Operstate:
ifindex=6 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
01-01 00:01:35.519  1424  1424 D wpa_supplicant: nl80211: Set mode ifindex
6 iftype 2 (STATION)
01-01 00:01:35.520  1424  1424 D wpa_supplicant: nl80211: Unsubscribe mgmt
frames handle 0x3e3c3849 (mode change)

Regards
Anees