lwfinger/rtl8188eu #198: Crash when leaving sleep mode.

RavRabbit, 2017-02-08:
On Linux 3.4 (sunxi), after the boot, the board is put to sleep with  `echo standby > /sys/power/state` or `echo mem > /sys/power/state`
After waking up, the 8188eu driver crashes with the following message :

`[  283.332888] WARNING: at drivers/base/firmware_class.c:597 request_firmware+0x88/0xbc()
[  283.341422] Modules linked in: 8188eu 
[  283.351506] [<c0014138>] (unwind_backtrace+0x0/0xe8) from [<c002e740>] (warn_slowpath_common+0x54/0x64)
[  283.371207] [<c002e740>] (warn_slowpath_common+0x54/0x64) from [<c002e7ec>] (warn_slowpath_null+0x1c/0x24)
[  283.387879] [<c002e7ec>] (warn_slowpath_null+0x1c/0x24) from [<c0291774>] (request_firmware+0x88/0xbc)
[  283.399968] [<c0291774>] (request_firmware+0x88/0xbc) from [<bf074768>] (rtl8188e_FirmwareDownload+0x190/0x660 [8188eu])
[  283.415495] [<bf074768>] (rtl8188e_FirmwareDownload+0x190/0x660 [8188eu]) from [<bf07d73c>] (rtl8188eu_hal_init+0x200/0xd10 [8188eu])
[  283.431735] [<bf07d73c>] (rtl8188eu_hal_init+0x200/0xd10 [8188eu]) from [<bf065f9c>] (rtw_hal_init+0x1c/0xd8 [8188eu])
[  283.446778] [<bf065f9c>] (rtw_hal_init+0x1c/0xd8 [8188eu]) from [<bf091444>] (_netdev_open+0xf0/0x460 [8188eu])
[  283.461369] [<bf091444>] (_netdev_open+0xf0/0x460 [8188eu]) from [<bf0917d4>] (netdev_open+0x20/0x34 [8188eu])
[  283.475854] [<bf0917d4>] (netdev_open+0x20/0x34 [8188eu]) from [<bf0919cc>] (pm_netdev_open+0x24/0x28 [8188eu])
[  283.490662] [<bf0919cc>] (pm_netdev_open+0x24/0x28 [8188eu]) from [<bf093e00>] (rtw_resume_process+0x78/0x19c [8188eu])
[  283.505986] [<bf093e00>] (rtw_resume_process+0x78/0x19c [8188eu]) from [<bf093f38>] (rtw_resume+0x14/0x18 [8188eu])
[  283.528815] [<bf093f38>] (rtw_resume+0x14/0x18 [8188eu]) from [<c02bdff0>] (usb_resume_interface.isra.1+0x7c/0x11c)
[  283.539877] [<c02bdff0>] (usb_resume_interface.isra.1+0x7c/0x11c) from [<c02bea3c>] (usb_resume+0x74/0x150)
[  283.559929] [<c02bea3c>] (usb_resume+0x74/0x150) from [<c02b2d14>] (usb_dev_resume+0xc/0x10)
[  283.569912] [<c02b2d14>] (usb_dev_resume+0xc/0x10) from [<c028be40>] (dpm_run_callback.isra.5+0x2c/0x74)
[  283.590801] [<c028be40>] (dpm_run_callback.isra.5+0x2c/0x74) from [<c028c750>] (device_resume+0xe0/0x1bc)
[  283.601906] [<c028c750>] (device_resume+0xe0/0x1bc) from [<c028c848>] (async_resume+0x1c/0x44)
[  283.611180] [<c028c848>] (async_resume+0x1c/0x44) from [<c004fab0>] (async_run_entry_fn+0xa0/0x1e0)
[  283.620879] [<c004fab0>] (async_run_entry_fn+0xa0/0x1e0) from [<c004344c>] (process_one_work+0x1c8/0x424)
[  283.636084] [<c004344c>] (process_one_work+0x1c8/0x424) from [<c00451fc>] (worker_thread+0x16c/0x414)
[  283.645727] [<c00451fc>] (worker_thread+0x16c/0x414) from [<c00496e8>] (kthread+0x90/0x94)
[  283.654563] [<c00496e8>] (kthread+0x90/0x94) from [<c000f05c>] (kernel_thread_exit+0x0/0x8)
[  283.663401] ---[ end trace 156e4ec7f82bd696 ]---
[  283.668491] r8188eu 2-1:1.0: firmware: rtlwifi/rtl8188eufw.bin will not be loaded
[  283.676361] Request firmware failed with error 0xfffffff5
[  283.682275] r8188eu 2-1:1.0: reset_resume error -1`

The driver tries to load the firmware, but there is a bug in the kernel which make the function request_firmware to crash after wakup.

The error doesn't occurs if the wifi has been enabled before, because enabling wifi loads the firmware.

So, either the driver should load the firmware when it is loaded (modprobe 8188eu.ko), or, when resuming from sleep mode, it shouldn't load it if it hasn't been loaded before.

lwfinger, 2017-02-08:
I'm sorry, but I provide this Realtek driver as a service. I try to fix any crashes, but I make no promises. Besides, this is a WARNING not a BUG or OOPS.

If you figure out a patch, I will be happy to incorporate it, but I will not be debugging this myself.

RavRabbit, 2017-02-09:
It is a bug, because the wifi doesn't work after enabling sleep mode.
I understand that you are not the developer of the driver, so, until I can figure out a good patch, the current workaround is to enable and disable wifi in the boot process.