lwfinger/rtl8188eu #199: doesn't work on Manjaro Stable(Linux 4.4)

turtleDev, 2017-02-12:
rtl8188eu (either via make install or dkms) just won't work.

The driver output is
```
[ 2086.538678] usb 2-2: new high-speed USB device number 7 using xhci_hcd
[ 2086.714705] bFWReady == _FALSE call reset 8051...
[ 2086.734850] RTL871X: rtw_ndev_init(wlan0)
[ 2087.263419] rtl8188eu 2-2:1.0 enp0s20u2: renamed from wlan0
[ 2087.289821] IPv6: ADDRCONF(NETDEV_UP): enp0s20u2: link is not ready
[ 2087.538749] ==> rtl8188e_iol_efuse_patch 
[ 2087.766775] IPv6: ADDRCONF(NETDEV_UP): enp0s20u2: link is not ready
[ 2087.774604] RTL871X: indicate disassoc
[ 2087.876013] IPv6: ADDRCONF(NETDEV_UP): enp0s20u2: link is not ready
[ 2087.878242] RTL871X: set bssid:00:00:00:00:00:00
[ 2087.878254] RTL871X: set ssid [\xffffffef\xfffffff7] fw_state=0x00000008
[ 2087.927628] IPv6: ADDRCONF(NETDEV_UP): enp0s20u2: link is not ready
[ 2087.930414] RTL871X: indicate disassoc
[ 2088.032131] IPv6: ADDRCONF(NETDEV_UP): enp0s20u2: link is not ready
[ 2088.048812] RTL871X: indicate disassoc
[ 2088.150491] IPv6: ADDRCONF(NETDEV_UP): enp0s20u2: link is not ready
[ 2088.168935] RTL871X: indicate disassoc
[ 2088.270528] IPv6: ADDRCONF(NETDEV_UP): enp0s20u2: link is not ready
[ 2088.288862] RTL871X: indicate disassoc
[ 2088.390504] IPv6: ADDRCONF(NETDEV_UP): enp0s20u2: link is not ready
... repeat till infinity
```

any idea on what could cause this? I also have Ubuntu 14.04 (linux 4.4) and it works perfect there.

turtleDev, 2017-02-12:
I tried using `wpa_supplicant` in order to get more information about the problem;

```
Successfully initialized wpa_supplicant
Initializing interface 'enp0s20u2' conf '/tmp/wpa.conf' driver 'default' ctrl_interface 'N/A' bridge 'N/A'
Configuration file '/tmp/wpa.conf' -> '/tmp/wpa.conf'
Reading configuration file '/tmp/wpa.conf'
Priority group 0
   id=0 ssid='TP-LINK_920DA0'
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:10
nl80211: Supported cipher 00-0f-ac:8
nl80211: Supported cipher 00-0f-ac:9
nl80211: Using driver-based off-channel TX
nl80211: Driver-advertised extended capabilities (default) - hexdump(len=8): 00 00 00 00 00 00 00 40
nl80211: Driver-advertised extended capabilities mask (default) - hexdump(len=8): 00 00 00 00 00 00 00 40
nl80211: interface enp0s20u2 in phy phy0
nl80211: Set mode ifindex 8 iftype 2 (STATION)
nl80211: Failed to set interface 8 to mode 2: -19 (No such device)
nl80211: Could not configure driver mode
nl80211: deinit ifname=enp0s20u2 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=8 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 8 iftype 2 (STATION)
nl80211: Failed to set interface 8 to mode 2: -19 (No such device)
enp0s20u2: Failed to initialize driver interface
Failed to add interface enp0s20u2
enp0s20u2: Cancelling scan request
enp0s20u2: Cancelling authentication timeout
Off-channel: Clear pending Action frame TX (pending_action_tx=(nil)
```

dje4321, 2017-02-12:
There is a issue with newer versions of wpa_supplicant. what version of wpa_supplicant are you running? mine is 2.5-5 on fedora 25

turtleDev, 2017-02-12:
@dje4321 I'm running v2.6

dje4321, 2017-02-12:
try downgrading wpa_supplicant down to a maximum of 2.5

turtleDev, 2017-02-12:
no dice. downgraded to 2.5, to no avail.

dje4321, 2017-02-12:
might try 2.4. not sure at this point

turtleDev, 2017-02-13:
I doubt this is an issue with wpa_supplicant, but I'll give it a try later.

turtleDev, 2017-02-13:
so taking your cue, I checked what version of `wpa_supplicant` ubuntu was running, and it turned out to be v2.1, so I downgraded to v2.1 on from Arch Linux Archieve, but it still did not fix the issue.

turtleDev, 2017-03-19:
bump. Still doesn't work :(

What other options do I have?
Alternatively, can someone help me debug the problem? I'd be happy to write patch.

ashwinravianandan, 2017-03-20:
Try running wpa_supplicant with -DWEXT

netllama, 2017-04-16:
Pretty sure that you're running into this issue:
http://www.thelinuxrain.com/articles/getting-realtek-8188eu-wireless-adapters-to-work-in-linux-and-possibly-other-wireless-realtek-chipsets

This bug is likely a duplicate of https://github.com/lwfinger/rtl8188eu/issues/197

turtleDev, 2017-04-17:
I've already read that article, and tried the solutions listed; they don't work, or at least, didn't for me :)

dje4321, 2017-04-17:
Have you tried the hidden networks menu? what i have to use now since the wireless adapter wont turn on automaticly

tnvrsingh, 2017-04-25:
Connecting via the hidden networks menu is the way to go for now. That's what I did on my arch install with Gnome. 

Searching for networks doesn't work

guysoft, 2018-01-04:
@ dje4321 2.4 does not work for me here, connects to open networks but not wpa

lwfinger, 2018-01-04:
I just checked with openSUSE Tumbleweed with NetworkManager controlling the device. My wpa_supplicant version is 2.6. It works with WPA2.

I do not know what the problem is, but I am quite sure it is in your user-space stuff, not the driver.

guysoft, 2018-01-05:
I upgraded to wpa_supplicant version 2.6, rebooted, and it still does not work on debian sid.

lwfinger, 2018-01-06:
And how are you controlling the network?

guysoft, 2018-01-06:
Network manager,  I can try wickd

lwfinger, 2018-01-06:
As I said earlier, the driver works fine with NetworkManager on openSUSE Tumbleweed using KDE as the desktop. The problem is with your distro;s user code.

guysoft, 2018-01-06:
where can I debug this?
If open networks do work. And it fails both on my Kububtu and Debian. And
only for this driver (archos on pci works fine).

 Where should I look? What log should I check?

lwfinger, 2018-01-06:
You should start wpa_supplicant with maximum debugging (-ddd switch) and look in its logs, as well as in dmesg.