lwfinger/rtl8188eu #201: How to make it to useful for MacOS

Autines, 2017-03-08:
Hello,developer. I wonder if you can make it useful for MacOS.

lwfinger, 2017-03-08:
The short answer is "No.". I have no knowledge of the internals of MaxOS. The original drivers from Realtek have options to build the drivers for Linux, Windows, and FreeBSD. I would expect the version for MacOS to be similar to the latter, but no guarantee.

Does the manufacturer of your device have a MacOS driver on their website?

Autines, 2017-03-08:
Yes, but it is a lower version which always lead MacOS down. So I am looking for a new version for new system. The files below is driver for my usb ethernet.
[Installer.pkg.zip](https://github.com/lwfinger/rtl8188eu/files/828619/Installer.pkg.zip)
[RTL8188EU.kext.zip](https://github.com/lwfinger/rtl8188eu/files/828625/RTL8188EU.kext.zip)

closed 2018-11-10 by lwfinger