lwfinger/rtl8188eu #203: cannot set channel parameter when using ad hoc mode with iwconfig or iw

shantanoo-desai, 2017-03-27:
I have __Ralink WL1045N__ based `r8188eu` chipsets dongles which work on ad-hoc network. I can ping them but cannot set the channel parameters 

    sudo iwconfig wlan1 mode ad-hoc channel 11 essid test-adhoc

does not return an error message upon execution. However the channel value upon `iwconfig wlan1` is still `2.412 GHz` which is Channel 1.

I tried working on `iw dev wlan1 --debug set channel 11` which gives me the following message:
```

 Debug: Sent Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 44
    .type = 27 <0x1b>
    .flags = 5 <REQUEST,ACK>
    .seq = 1490631097
    .port = 750784569
  [GENERIC NETLINK HEADER] 4 octets
    .cmd = 2
    .version = 0
    .unused = 0
  [PAYLOAD] 24 octets
    08 00 03 00 04 00 00 00 08 00 26 00 9e 09 00 00 ..........&.....
    08 00 27 00 00 00 00 00                         ..'.....
---------------------------  END NETLINK MESSAGE   ---------------------------
-- Debug: Received Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [NETLINK HEADER] 16 octets
    .nlmsg_len = 64
    .type = 2 <ERROR>
    .flags = 0 <>
    .seq = 1490631097
    .port = 750784569
  [ERRORMSG] 20 octets
    .error = -95 "Operation not supported"
  [ORIGINAL MESSAGE] 16 octets
    .nlmsg_len = 16
    .type = 27 <0x1b>
    .flags = 5 <REQUEST,ACK>
    .seq = 1490631097
    .port = 750784569
---------------------------  END NETLINK MESSAGE   ---------------------------
command failed: Operation not supported (-95)

```

I have the compiled and install version on my Laptop. What is the possible problem?

lwfinger, 2017-03-27:
That old driver probably does not implement the complete cfg80211 set of operations, or there is a bug. If what you wish to do cannot be accomplished with iwconfig commands, then I suggest you switch to a newer kernel and drive your device with rtl8xxxu.

My involvement with this driver was to make it available for use and to fix any problems in building it or those that lead to kernel panics. I cannot fix cfg80211 command problems.

closed 2017-03-27 by lwfinger

shantanoo-desai, 2017-03-27:
Hi, I understand. the `rt2x00lib` are very handy and easy to work around with. The `8188eu` have always been a menace.