lwfinger/rtl8188eu #207: Can't `make all`

cryptid11, 2017-04-21:
```
$ make all
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/3.16.0-4-amd64/build M=/home/rodomonte/Documents/rtl8188eu  modules
make[1]: Entering directory '/usr/src/linux-headers-3.16.0-4-amd64'
make[1]: Entering directory `/usr/src/linux-headers-3.16.0-4-amd64'
  CC [M]  /home/rodomonte/Documents/rtl8188eu/os_dep/usb_intf.o
/home/rodomonte/Documents/rtl8188eu/os_dep/usb_intf.c:60:3: error: expected expression before ‘{’ token
 + {USB_DEVICE(0x2357, 0x010c)}, /* TP-Link TL-WN722N v2 */
   ^
/usr/src/linux-headers-3.16.0-4-common/scripts/Makefile.build:262: recipe for target '/home/rodomonte/Documents/rtl8188eu/os_dep/usb_intf.o' failed
make[4]: *** [/home/rodomonte/Documents/rtl8188eu/os_dep/usb_intf.o] Error 1
/usr/src/linux-headers-3.16.0-4-common/Makefile:1353: recipe for target '_module_/home/rodomonte/Documents/rtl8188eu' failed
make[3]: *** [_module_/home/rodomonte/Documents/rtl8188eu] Error 2
Makefile:181: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-3.16.0-4-amd64'
Makefile:151: recipe for target 'modules' failed
make: *** [modules] Error 2
```

$ uname -a
Linux rod 3.16.0-4-amd64 #1 SMP Debian 3.16.39-1+deb8u2 (2017-03-07) x86_64 GNU/Linux         

with same kernel 32bit version it works, so I suppose this doesn't support 64bit right?

lwfinger, 2017-04-21:
Sorry - my typo. It is now fixed.

No, this driver works for both 32- and 64-bits. Some vendor drivers might not work for 64 bits, but anything that I redistribute works for both.

cryptid11, 2017-04-21:
thanks now it works!

lwfinger, 2017-04-21:
Good.

closed 2017-04-21 by lwfinger