lwfinger/rtl8188eu #209: Recognition

quemexordium, 2017-05-07:
Hello. I'm running Fedora 25 (4.10.13-200.fc25.x86_64). I bought TP LINK TL-WN722N v2 usb wireless adapter with rtl8188eus chipset. Before I haven't compiled these files, when I plugged my usb wireless adapter, there were no lights and any sign of it working etc. and my device was not seen in `lsusb`. After I compiled all these files successfully, I saw the green light on my device but it is still not seen in `lsusb`. Here's the output:
```

Bus 002 Device 003: ID 8087:07da Intel Corp. 
Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 004: ID 0c45:644a Microdia 
Bus 001 Device 003: ID 0bda:0129 Realtek Semiconductor Corp. RTS5129 Card Reader Controller
Bus 001 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 003: ID 2357:010c  
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```
I'm using it on my laptop and it has a built-in wireless card and I can't deactivate/blacklist it. Whenever I turn my built-in wireless card off, my usb wireless adapter gets turned off too. And when they're both turned on, my computer uses its built-in wireless card instead of usb adapter. 

iwconfig:

```
virbr0    no wireless extensions.

enp7s0    no wireless extensions.

lo        no wireless extensions.

wlp8s0    IEEE 802.11  ESSID:"Tenda_3281B8"  
          Mode:Managed  Frequency:2.462 GHz  Access Point: C8:3A:35:32:81:B8   
          Bit Rate=150 Mb/s   Tx-Power=16 dBm   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=59/70  Signal level=-51 dBm  
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:100   Missed beacon:0

virbr0-nic  no wireless extensions.

wlp0s20u1  unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

```
ifconfig: 

```
enp7s0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether e0:db:55:d0:13:7b  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 584  bytes 52112 (50.8 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 584  bytes 52112 (50.8 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

virbr0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.122.1  netmask 255.255.255.0  broadcast 192.168.122.255
        ether 52:54:00:a9:28:84  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
[... 45 lines cut ...]
May 07 04:54:10 localhost.localdomain wpa_supplicant[1257]: ioctl[SIOCSIWAP]: Operation not permitted
May 07 04:54:10 localhost.localdomain NetworkManager[1052]: <info>  [1494122050.8372] sup-iface[0x556ca8ed9ec0,wlp0s20u1]: supports 1 scan SSIDs
May 07 04:54:10 localhost.localdomain NetworkManager[1052]: <info>  [1494122050.8403] device (wlp0s20u1): supplicant interface state: starting -> ready
May 07 04:54:10 localhost.localdomain NetworkManager[1052]: <info>  [1494122050.8404] device (wlp0s20u1): state change: unavailable -> disconnected (reason 'supplicant-available') [20 30 42]
May 07 04:54:10 localhost.localdomain kernel: IPv6: ADDRCONF(NETDEV_UP): wlp0s20u1: link is not ready
May 07 04:54:10 localhost.localdomain kernel: R8188EU: INFO indicate disassoc
May 07 04:54:10 localhost.localdomain NetworkManager[1052]: <warn>  [1494122050.9471] device (wlp0s20u1): set-hw-addr: new MAC address 9E:CE:08:1D:82:7D not successfully set (scanning)
May 07 04:54:10 localhost.localdomain kernel: IPv6: ADDRCONF(NETDEV_UP): wlp0s20u1: link is not ready
May 07 04:54:10 localhost.localdomain NetworkManager[1052]: <info>  [1494122050.9498] device (wlp0s20u1): supplicant interface state: ready -> disabled
May 07 04:54:10 localhost.localdomain NetworkManager[1052]: <info>  [1494122050.9756] device (wlp0s20u1): supplicant interface state: disabled -> inactive
May 07 04:54:12 localhost.localdomain kernel: R8188EU: INFO indicate disassoc
May 07 04:54:12 localhost.localdomain kernel: R8188EU: INFO indicate disassoc
May 07 04:54:12 localhost.localdomain NetworkManager[1052]: <warn>  [1494122052.2043] device (wlp0s20u1): set-hw-addr: new MAC address 9E:CE:08:1D:82:7D not successfully set (scanning)
May 07 04:54:12 localhost.localdomain kernel: IPv6: ADDRCONF(NETDEV_UP): wlp0s20u1: link is not ready
May 07 04:54:12 localhost.localdomain NetworkManager[1052]: <info>  [1494122052.2075] device (wlp0s20u1): supplicant interface state: inactive -> disabled
May 07 04:54:12 localhost.localdomain NetworkManager[1052]: <info>  [1494122052.2249] device (wlp0s20u1): supplicant interface state: disabled -> inactive
May 07 04:54:12 localhost.localdomain kernel: R8188EU: INFO indicate disassoc
May 07 04:54:12 localhost.localdomain NetworkManager[1052]: <warn>  [1494122052.3324] device (wlp0s20u1): set-hw-addr: new MAC address 9E:CE:08:1D:82:7D not successfully set (scanning)
May 07 04:54:12 localhost.localdomain kernel: IPv6: ADDRCONF(NETDEV_UP): wlp0s20u1: link is not ready
May 07 04:54:12 localhost.localdomain NetworkManager[1052]: <info>  [1494122052.3351] device (wlp0s20u1): supplicant interface state: inactive -> disabled
May 07 04:54:12 localhost.localdomain NetworkManager[1052]: <info>  [1494122052.3543] device (wlp0s20u1): supplicant interface state: disabled -> inactive
^C
```

I hope you guys can help me with that.

quemexordium, 2017-05-07:
I actually managed to blacklist my built-in wireless card with creating "blacklist.conf" in /etc/modprobe.d/ and adding the line: "blacklist iwlwifi". But when I rebooted my machine, my usb wireless adapter didn't work either. There were no lights on it etc. And I didn't see USB Wi-Fi option in my wireless settings. I think they're somehow interconnected. When both my built-in and usb wireless adapter is turned on, my usb wireless adapter doesn't find any networks. It says "No Networks" which is a bit odd because although my built-in wireless is connected, my usb wireless adapter should be able to scan networks as well.

quemexordium, 2017-05-09:
Here are some other things you might find useful:
lshw output of my usb wireless adapter:
```
*-network:2 DISABLED
       description: Wireless interface
       physical id: 4
       bus info: usb@3:1
       logical name: wlp0s20u1
       serial: 18:d6:c7:1b:be:23
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=r8188eu multicast=yes wireless=unassociated
```
lsmod | grep 8188:

`8188eu                733184  0
`
sudo iwlist scan:
`wlp0s20u1  No scan results
`

lwfinger, 2017-05-09:
What does "rfkill list all" show?

quemexordium, 2017-05-09:
```
[root@localhost nashjr]# rfkill list all
0: hci0: Bluetooth
	Soft blocked: no
	Hard blocked: no
1: phy0: Wireless LAN
	Soft blocked: no
	Hard blocked: no

```

lwfinger, 2017-05-09:
How about "iw dev"?

quemexordium, 2017-05-09:
```
phy#0
	Interface wlp8s0
		ifindex 4
		wdev 0x1
		addr 84:a6:c8:e2:3f:c4
		type managed
		channel 11 (2462 MHz), width: 40 MHz, center1: 2452 MHz
		txpower 16.00 dBm

```

lwfinger, 2017-05-09:
That is the "disabled" iwlwifi device. Your user-space system (NetworkManager, etc.) is not recognizing the USB device. That is a problem with that driver that I do not know how to fix.

Your solution will be to go to a new kernel (4.10) and use the rtl8xxxu in it.

quemexordium, 2017-05-09:
I enabled it again otherwise I wouldn't be able to reach you. 
What do you mean by that? My kernel version is 4.10.13 already. Do you mean that I should re-install a fresh kernel?

PrSdNt, 2017-05-28:
Hi @quemexordium,

Got pretty much the same setup as you. I've got the Dell XPS 15 9560 and while it's a great laptop, the built-in wifi is mediocre at best (it's an antenna issue).  So i bought me a TP-LINK WN722N v2.0 as a backup device. 

Distro: Fedora 25
Kernel: 4.10.16-200

This way i got it to work:

1. Install drivers (obviously)
1. Insert usb wifi device
2. Disable built in PCI-device: `$ sudo ifdown wlp2s0` (find your devicename with ifconfig)
3. Open Wifi Settings through NetworkManager dropdown.
4. Click "Connect to hidden network"
5. Choose Realtek adapter and create a dummy entry. SSID dummy, password dummy etc. Then click connect.
6. It won't connect, but the available wifi-networks will show up in the Select Network window.
7. Choose your network and connect. 

That's it! Hope it helps.

plasticassius, 2018-11-21:
I ran into a similar issue in which my built in wifi failed, and I can't seem to use user mode configuration tools for my TP LINK TL-WN722N v3. In my case, I was using wicd on Debian 9 LXDE.

```
$uname -a
Linux Rebuild 4.9.0-8-amd64 #1 SMP Debian 4.9.130-2 (2018-10-27) x86_64 GNU/Linux
$cat /etc/debian_version 
9.6
```
A fairly simple fix took care of the problem. I put the following into my /etc/network/interfaces file:
```
auto wlxMAC
allow-hotplug wlxMAC
iface wlxMAC inet dhcp 
    wpa-ssid SSID
    wpa-psk  PSK
```
My interface name turned out to be wlxMAC where MAC is the MAC address of the card. It may not be neccessary, but I also blacklisted my previous driver, and removed the wicd package which wasn't working.