lwfinger/rtl8188eu #210: Xubuntu freeze + caps lock indicator blinking when connecting to network.

Efelow, 2017-05-16:
Hello,
After compiling and installing the driver on Ubuntu 16.04.1 LTS (Xubuntu version, kernel 4.4.0-77-generic), I plugged my USB key (TL-WN722N). Then, I am able to see the available network. But once I enter the passphrase, after few second the system hang and the caps lock indicator is blinking.

Does anyone experience this problem ? If you need additionnal information, please tell me.

Thanks for your help.

Flo

lwfinger, 2017-05-16:
That is a kernel panic. We need to know what is logged when the crash happens.

Efelow, 2017-05-16:
I flush the syslog file before connecting the key and I reconnect the key and try to connect to the network in order filter a little bit all the logs.

> May 16 20:01:02 efelow-N250P kernel: [ 1062.640106] usb 1-7: new high-speed USB device number 2 using ehci-pci
May 16 20:01:02 efelow-N250P kernel: [ 1062.772994] usb 1-7: New USB device found, idVendor=2357, idProduct=010c
May 16 20:01:02 efelow-N250P kernel: [ 1062.773009] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
May 16 20:01:02 efelow-N250P kernel: [ 1062.773017] usb 1-7: Product: 802.11n NIC
May 16 20:01:02 efelow-N250P kernel: [ 1062.773024] usb 1-7: Manufacturer: Realtek
May 16 20:01:02 efelow-N250P kernel: [ 1062.773030] usb 1-7: SerialNumber: 00E04C0001
May 16 20:01:02 efelow-N250P mtp-probe: checking bus 1, device 2: "/sys/devices/pci0000:00/0000:00:13.2/usb1/1-7"
May 16 20:01:02 efelow-N250P mtp-probe: bus: 1, device: 2 was not an MTP device
May 16 20:01:03 efelow-N250P kernel: [ 1063.895705] 8188eu: module verification failed: signature and/or required key missing - tainting kernel
May 16 20:01:03 efelow-N250P kernel: [ 1063.901055] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
May 16 20:01:03 efelow-N250P kernel: [ 1063.934533] EEPROM ID = 0x8129
May 16 20:01:03 efelow-N250P kernel: [ 1063.937448] usbcore: registered new interface driver r8188eu
May 16 20:01:03 efelow-N250P kernel: [ 1063.965573] r8188eu 1-7:1.0 wlx18d6c71d0a83: renamed from wlan0
May 16 20:01:03 efelow-N250P NetworkManager[825]: <error> [1494957663.8459] (wlan0): couldn't get driver range information (19).
May 16 20:01:03 efelow-N250P NetworkManager[825]: <info>  [1494957663.8459] (wlan0): driver WEXT range request failed
May 16 20:01:03 efelow-N250P NetworkManager[825]: <warn>  [1494957663.8460] (wlan0) failed to initialize Wi-Fi driver for ifindex 3
May 16 20:01:03 efelow-N250P NetworkManager[825]: <warn>  [1494957663.8460] manager: wlan0: factory failed to create device: Device factory NMWifiFactory failed to create device wlan0
May 16 20:01:03 efelow-N250P NetworkManager[825]: <info>  [1494957663.8470] devices added (path: /sys/devices/pci0000:00/0000:00:13.2/usb1/1-7/1-7:1.0/net/wlx18d6c71d0a83, iface: wlx18d6c71d0a83)
May 16 20:01:03 efelow-N250P NetworkManager[825]: <info>  [1494957663.8470] device added (path: /sys/devices/pci0000:00/0000:00:13.2/usb1/1-7/1-7:1.0/net/wlx18d6c71d0a83, iface: wlx18d6c71d0a83): no ifupdown configuration found.
May 16 20:01:03 efelow-N250P kernel: [ 1064.086543] cfg80211: World regulatory domain updated:
May 16 20:01:03 efelow-N250P kernel: [ 1064.086553] cfg80211:  DFS Master region: unset
May 16 20:01:03 efelow-N250P kernel: [ 1064.086555] cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
May 16 20:01:03 efelow-N250P kernel: [ 1064.086560] cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
May 16 20:01:03 efelow-N250P kernel: [ 1064.086563] cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
May 16 20:01:03 efelow-N250P kernel: [ 1064.086566] cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
May 16 20:01:03 efelow-N250P kernel: [ 1064.086570] cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
May 16 20:01:03 efelow-N250P kernel: [ 1064.086574] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
May 16 20:01:03 efelow-N250P kernel: [ 1064.086577] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
May 16 20:01:03 efelow-N250P kernel: [ 1064.086581] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
May 16 20:01:03 efelow-N250P kernel: [ 1064.086584] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
May 16 20:01:05 efelow-N250P ModemManager[848]: <info>  Couldn't find support for device at '/sys/devices/pci0000:00/0000:00:13.2/usb1/1-7': not supported by any plugin
[NUL][NUL][NUL][NUL][NUL][NUL][NUL][NUL][NUL]
May 16 20:03:26 efelow-N250P rsyslogd: [origin software="rsyslogd" swVersion="8.16.0" x-pid="855" x-info="http://www.rsyslog.com"] start

In the kern.log file:

> May 16 20:01:02 efelow-N250P kernel: [ 1062.640106] usb 1-7: new high-speed USB device number 2 using ehci-pci
May 16 20:01:02 efelow-N250P kernel: [ 1062.772994] usb 1-7: New USB device found, idVendor=2357, idProduct=010c
May 16 20:01:02 efelow-N250P kernel: [ 1062.773009] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
May 16 20:01:02 efelow-N250P kernel: [ 1062.773017] usb 1-7: Product: 802.11n NIC
May 16 20:01:02 efelow-N250P kernel: [ 1062.773024] usb 1-7: Manufacturer: Realtek
May 16 20:01:02 efelow-N250P kernel: [ 1062.773030] usb 1-7: SerialNumber: 00E04C0001
May 16 20:01:03 efelow-N250P kernel: [ 1063.895705] 8188eu: module verification failed: signature and/or required key missing - tainting kernel
May 16 20:01:03 efelow-N250P kernel: [ 1063.901055] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
May 16 20:01:03 efelow-N250P kernel: [ 1063.934533] EEPROM ID = 0x8129
May 16 20:01:03 efelow-N250P kernel: [ 1063.937448] usbcore: registered new interface driver r8188eu
May 16 20:01:03 efelow-N250P kernel: [ 1063.965573] r8188eu 1-7:1.0 wlx18d6c71d0a83: renamed from wlan0
May 16 20:01:03 efelow-N250P NetworkManager[825]: <error> [1494957663.8459] (wlan0): couldn't get driver range information (19).
[... cut ...]
May 16 20:01:03 efelow-N250P kernel: [ 1064.086574] cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
May 16 20:01:03 efelow-N250P kernel: [ 1064.086577] cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
May 16 20:01:03 efelow-N250P kernel: [ 1064.086581] cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
May 16 20:01:03 efelow-N250P kernel: [ 1064.086584] cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
NUL NUL NUL NUL 

Tell me if you need other logs.
Thanks.

lwfinger, 2017-05-17:
These logs show nothing. Does Xubuntu use systemd logging - I expect that it does. After you reboot, run the command "sudo journalctl -e". Page backwards through the output to see if you see the panic dump.

Efelow, 2017-05-17:
I don't see any dump or any panic message. 
I make the journald persistant in order to get what was going on last boot. (with https://askubuntu.com/questions/763638/no-more-boot-logging-since-16-04/765307#765307) 
Then i use the command journalctl -b -1. 
Here is the log starting when I plugged the key into usb port:

> mai 17 12:44:57 efelow-N250P kernel: usb 1-7: new high-speed USB device number 2 using ehci-pci
mai 17 12:44:57 efelow-N250P kernel: usb 1-7: New USB device found, idVendor=2357, idProduct=010c
mai 17 12:44:57 efelow-N250P kernel: usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
mai 17 12:44:57 efelow-N250P kernel: usb 1-7: Product: 802.11n NIC
mai 17 12:44:57 efelow-N250P kernel: usb 1-7: Manufacturer: Realtek
mai 17 12:44:57 efelow-N250P kernel: usb 1-7: SerialNumber: 00E04C0001
mai 17 12:44:57 efelow-N250P mtp-probe[1943]: checking bus 1, device 2: "/sys/devices/pci0000:00/0000:00:13.2/usb1/1-7"
mai 17 12:44:57 efelow-N250P mtp-probe[1943]: bus: 1, device: 2 was not an MTP device
mai 17 12:44:59 efelow-N250P kernel: 8188eu: module verification failed: signature and/or required key missing - tainting kernel
mai 17 12:44:59 efelow-N250P kernel: Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
mai 17 12:44:59 efelow-N250P kernel: EEPROM ID = 0x8129
mai 17 12:44:59 efelow-N250P kernel: usbcore: registered new interface driver r8188eu
mai 17 12:44:59 efelow-N250P kernel: r8188eu 1-7:1.0 wlx18d6c71d0a83: renamed from wlan0
mai 17 12:44:59 efelow-N250P NetworkManager[694]: <error> [1495017899.2268] (wlan0): couldn't get driver range information (19).
mai 17 12:44:59 efelow-N250P NetworkManager[694]: <info>  [1495017899.2270] (wlan0): driver WEXT range request failed
mai 17 12:44:59 efelow-N250P NetworkManager[694]: <warn>  [1495017899.2271] (wlan0) failed to initialize Wi-Fi driver for ifindex 3
mai 17 12:44:59 efelow-N250P NetworkManager[694]: <warn>  [1495017899.2272] manager: wlan0: factory failed to create device: Device factory NMWifiFactory failed to create device wl
mai 17 12:44:59 efelow-N250P kernel: cfg80211: World regulatory domain updated:
mai 17 12:44:59 efelow-N250P kernel: cfg80211:  DFS Master region: unset
mai 17 12:44:59 efelow-N250P kernel: cfg80211:   (start_freq - end_freq @ bandwidth), (max_antenna_gain, max_eirp), (dfs_cac_time)
mai 17 12:44:59 efelow-N250P kernel: cfg80211:   (2402000 KHz - 2472000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
mai 17 12:44:59 efelow-N250P kernel: cfg80211:   (2457000 KHz - 2482000 KHz @ 40000 KHz), (N/A, 2000 mBm), (N/A)
mai 17 12:44:59 efelow-N250P kernel: cfg80211:   (2474000 KHz - 2494000 KHz @ 20000 KHz), (N/A, 2000 mBm), (N/A)
mai 17 12:44:59 efelow-N250P kernel: cfg80211:   (5170000 KHz - 5250000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (N/A)
mai 17 12:44:59 efelow-N250P kernel: cfg80211:   (5250000 KHz - 5330000 KHz @ 80000 KHz, 160000 KHz AUTO), (N/A, 2000 mBm), (0 s)
mai 17 12:44:59 efelow-N250P kernel: cfg80211:   (5490000 KHz - 5730000 KHz @ 160000 KHz), (N/A, 2000 mBm), (0 s)
mai 17 12:44:59 efelow-N250P kernel: cfg80211:   (5735000 KHz - 5835000 KHz @ 80000 KHz), (N/A, 2000 mBm), (N/A)
mai 17 12:44:59 efelow-N250P kernel: cfg80211:   (57240000 KHz - 63720000 KHz @ 2160000 KHz), (N/A, 0 mBm), (N/A)
mai 17 12:44:59 efelow-N250P NetworkManager[694]: <info>  [1495017899.2546] devices added (path: /sys/devices/pci0000:00/0000:00:13.2/usb1/1-7/1-7:1.0/net/wlx18d6c71d0a83, iface: w
mai 17 12:44:59 efelow-N250P NetworkManager[694]: <info>  [1495017899.2547] device added (path: /sys/devices/pci0000:00/0000:00:13.2/usb1/1-7/1-7:1.0/net/wlx18d6c71d0a83, iface: wl
mai 17 12:45:01 efelow-N250P ModemManager[708]: <info>  Couldn't find support for device at '/sys/devices/pci0000:00/0000:00:13.2/usb1/1-7': not supported by any plugin

In attached the journalctl-e after restart

[journalctl-e_after_restart.txt](https://github.com/lwfinger/rtl8188eu/files/1007617/journalctl-e_after_restart.txt)

lwfinger, 2017-05-18:
Still no panic message. These will appear on an CTRL-ALT-F10 console on my system. I have no idea where Ubuntu puts them.

Efelow, 2017-05-18:
Here is a photo of my screen what I get on CTRL-ALT-F10.

[image]

lwfinger, 2017-05-18:
That helps a lot. Unfortunately I still do not understand why it is crashing. The failure starts with a call to "dev_free_skb_any()". Normally, I would say that this call involves a NULL pointer, but the code explicitly checks for that contingency before it does the call. It appears that something is wrong with the skb in question even though it is not NULL. I'll be back when I have some ideas. Ping me in two days if I have not responded.