lwfinger/rtl8188eu #213: How does this compare to what TP-Link provides on their website?

nikita2206, 2017-06-13:
Hey, sorry for creating an Issue for that question.

I just wanted to know how is this related to what you could download from http://www.tp-link.com/us/download/TL-WN722N.html#Driver

It looks like TP-Link's version was last updated quite long ago (the archive says `20150417` in its name) but the version of the driver is 4.3.0.8 which is the most recent one I could find anywhere (maybe except this repo).

I tried your driver at first and it was working flaky - sometimes connections would just hang although `ip addr` shows that I'm still connected and have an IP address and my traffic is still routed through the WiFi dongle.

I ended up trying out TP-Link's driver but it was not ready for 4.* kernel so I had to patch it a bit. It _seems_ like it's working better now. I'm not sure if I should submit this patch anywhere.

JosephRedfern, 2017-06-15:
Hey @nikita2206. I'd be interested to see what you changed to make the TP-Link driver work under 4.*, so if you could provide a patch regardless of it's usefulness to this specific project then that'd be great. In particular, I'm struggling with references to struct members  like `STATION_INFO_TX_PACKETS` which I believe have been replaced in 4.x...

nikita2206, 2017-06-15:
@JosephRedfern sure, yeah some of these weren't that easy to find.
Here you go:

```
diff --git a/hal/hal_com_phycfg.c b/hal/hal_com_phycfg.c
index 8bb2747..16a85bf 100644
--- a/hal/hal_com_phycfg.c
+++ b/hal/hal_com_phycfg.c
@@ -2575,7 +2575,7 @@ Hal_ChannelPlanToRegulation(
 #ifdef CONFIG_LOAD_PHY_PARA_FROM_FILE
 
 extern char *rtw_phy_file_path;
-char	file_path[PATH_LENGTH_MAX];
+char	rtw_file_path[PATH_LENGTH_MAX];
 
 #define GetLineFromBuffer(buffer)	 strsep(&buffer, "\n")
 
@@ -2597,11 +2597,11 @@ phy_ConfigMACWithParaFile(
 
 	if ((pHalData->mac_reg_len == 0) && (pHalData->mac_reg == NULL))
 	{
-		rtw_merge_string(file_path, PATH_LENGTH_MAX, rtw_phy_file_path, pFileName);
+		rtw_merge_string(rtw_file_path, PATH_LENGTH_MAX, rtw_phy_file_path, pFileName);
 	
-		if (rtw_is_file_readable(file_path) == _TRUE)
+		if (rtw_is_file_readable(rtw_file_path) == _TRUE)
 		{
-			rlen = rtw_retrive_from_file(file_path, pHalData->para_file_buf, MAX_PARA_FILE_BUF_LEN);
+			rlen = rtw_retrive_from_file(rtw_file_path, pHalData->para_file_buf, MAX_PARA_FILE_BUF_LEN);
 			if (rlen > 0)
 			{
 				rtStatus = _SUCCESS;
@@ -2696,11 +2696,11 @@ phy_ConfigBBWithParaFile(
 
 	if ((pBufLen != NULL) && (*pBufLen == 0) && (pBuf == NULL))
 	{
-		rtw_merge_string(file_path, PATH_LENGTH_MAX, rtw_phy_file_path, pFileName);
+		rtw_merge_string(rtw_file_path, PATH_LENGTH_MAX, rtw_phy_file_path, pFileName);
 	
-		if (rtw_is_file_readable(file_path) == _TRUE)
+		if (rtw_is_file_readable(rtw_file_path) == _TRUE)
 		{
-			rlen = rtw_retrive_from_file(file_path, pHalData->para_file_buf, MAX_PARA_FILE_BUF_LEN);
+			rlen = rtw_retrive_from_file(rtw_file_path, pHalData->para_file_buf, MAX_PARA_FILE_BUF_LEN);
 			if (rlen > 0)
 			{
 				rtStatus = _SUCCESS;
@@ -3149,11 +3149,11 @@ phy_ConfigBBWithPgParaFile(
 
 	if ((pHalData->bb_phy_reg_pg_len == 0) && (pHalData->bb_phy_reg_pg == NULL))
 	{
-		rtw_merge_string(file_path, PATH_LENGTH_MAX, rtw_phy_file_path, pFileName);
+		rtw_merge_string(rtw_file_path, PATH_LENGTH_MAX, rtw_phy_file_path, pFileName);
 	
-		if (rtw_is_file_readable(file_path) == _TRUE)
+		if (rtw_is_file_readable(rtw_file_path) == _TRUE)
 		{
-			rlen = rtw_retrive_from_file(file_path, pHalData->para_file_buf, MAX_PARA_FILE_BUF_LEN);
+			rlen = rtw_retrive_from_file(rtw_file_path, pHalData->para_file_buf, MAX_PARA_FILE_BUF_LEN);
 			if (rlen > 0)
 			{
 				rtStatus = _SUCCESS;
@@ -3212,11 +3212,11 @@ phy_ConfigBBWithMpParaFile(
 
 	if ((pHalData->bb_phy_reg_mp_len == 0) && (pHalData->bb_phy_reg_mp == NULL))
 	{
-		rtw_merge_string(file_path, PATH_LENGTH_MAX, rtw_phy_file_path, pFileName);
+		rtw_merge_string(rtw_file_path, PATH_LENGTH_MAX, rtw_phy_file_path, pFileName);
 	
-		if (rtw_is_file_readable(file_path) == _TRUE)
[... 558 lines cut ...]
 		if (ch) {
@@ -387,7 +387,7 @@ static void _rtw_reg_apply_flags(struct wiphy *wiphy)
 	u16 channel;
 	u32 freq;
 
-	for (i = 0; i < IEEE80211_NUM_BANDS; i++) {
+	for (i = 0; i < NUM_NL80211_BANDS; i++) {
 		sband = wiphy->bands[i];
 
 		if (sband)
@@ -404,11 +404,11 @@ static void _rtw_reg_apply_flags(struct wiphy *wiphy)
 		if (channel <= 14)
 			freq =
 			    rtw_ieee80211_channel_to_frequency(channel,
-							       IEEE80211_BAND_2GHZ);
+							       NL80211_BAND_2GHZ);
 		else
 			freq =
 			    rtw_ieee80211_channel_to_frequency(channel,
-							       IEEE80211_BAND_5GHZ);
+							       NL80211_BAND_5GHZ);
 
 		ch = ieee80211_get_channel(wiphy, freq);
 		if (ch) {
```

JosephRedfern, 2017-06-16:
@nikita2206 Great, thanks for that. I also found this repo: https://github.com/mfruba/kernel which has some similar fixes applied -- although your code may be a little neater!