lwfinger/rtl8188eu #217: Monitor mode not supported

nikhilskynet, 2017-07-09:
monitor mod is not working.

lwfinger, 2017-07-09:
This has been discussed many times in these threads. If you want monitor mode, you will have to add it to the driver. It was not put in by Realtek. No, I do not know why.

closed 2017-07-09 by lwfinger

ksabarianandhan, 2017-10-23:
Can you please Explain me how to install those drivers. Or You direct me to that link. as i'm a beginner i cant install those.Please help me

nikhilskynet, 2017-10-23:
enter these in terminal

make all

sudo make install

copy rtl8188eufw.bin to /lib/firmware/rtlwifi/

intersected, 2017-10-24:
@nikhilskynet thanks for the comment, but will that actually work ? will I be able to put adapter in monitor mode? 
Thanks

nikhilskynet, 2017-10-26:
another driver by mfruba supports monitor mode...
https://github.com/mfruba/kernel

alexey2baranov, 2021-06-27:
> enter these in terminal make all sudo make install copy rtl8188eufw.bin to /lib/firmware/rtlwifi/

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.10.0-kali7-amd64/build M=/home/kali/rtl8188eu  modules
make[1]: *** /lib/modules/5.10.0-kali7-amd64/build: No such file or directory.  Stop.
make: *** [Makefile:155: modules] Error 2

lwfinger, 2021-06-28:
As has been stated many times, you need to install your kernel headers.