lwfinger/rtl8188eu #218: TP-LINK_TL-WN722N_v2 on raspberry pi 3

vallettea, 2017-07-10:
I tried the answer found here:
https://askubuntu.com/questions/912498/tl-wn722n-is-not-recognized

but it doesn't work:

```
root@raspberrypi:/usr/src# dkms add ./rtl8188eu

Creating symlink /var/lib/dkms/8188eu/1.0/source ->
                 /usr/src/8188eu-1.0

DKMS: add completed.
root@raspberrypi:/usr/src# dkms build 8188eu/1.0

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area....
'make' all.................................................................................................
cleaning build area....

DKMS: build completed.
root@raspberrypi:/usr/src# dkms install 8188eu/1.0

8188eu:
Running module version sanity check.

Good news! Module version v4.1.4_6773.20130222 for 8188eu.ko
exactly matches what is already found in kernel 4.9.28-v7+.
DKMS will not replace this module.
You may override by specifying --force.

depmod.....

: Unable to find an initial ram disk that I know how to handle.
Will not try to make an initrd.

DKMS: install completed.
root@raspberrypi:/usr/src# modprobe 8188eu
root@raspberrypi:/usr/src#
root@raspberrypi:/usr/src#
root@raspberrypi:/usr/src# lsusb
Bus 001 Device 004: ID 2357:010c
```

any help appreciated

lwfinger, 2017-07-10:
What part do you need help with. Your posted log shows that you do not need to use this module.

vallettea, 2017-07-11:
thanks for your answer. All the posts I've red about this dongle say that your driver can enable it. I thought that installing it would solve my problem but it doesn't. Any insight on this particular dongle (and its chipset) ?
best

AndyMender, 2017-07-11:
The chipset in this _dongle_ is definitely Realtek RTL8188EU as I use it with Fedora on my workstation. @vallettea could you also post back the output of `ifconfig` and/or `ip link` and the relevant information from dmesg about the driver (`dmesg | grep 8188eu`) to see whether the appropriate interface gets created at all and what exactly happens at the driver level?

vallettea, 2017-07-11:
thanks for your help @AndyMender ! here is the dmesg output:
```
[    3.391251] usb 1-1.5: New USB device found, idVendor=2357, idProduct=010c
[    3.401591] usb 1-1.5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    3.412287] usb 1-1.5: Product: 802.11n NIC
[    3.419762] usb 1-1.5: Manufacturer: Realtek
[    3.427285] usb 1-1.5: SerialNumber: 00E04C0001
[    3.497734] NET: Registered protocol family 10
[    3.507392] systemd[1]: Inserted module 'ipv6'
[    3.519770] systemd[1]: Set hostname to <raspberrypi>.
[    3.528313] systemd[1]: Initializing machine ID from random generator.
[    3.538184] systemd[1]: Installed transient /etc/machine-id file.
[    3.607909] uart-pl011 3f201000.serial: no DMA platform data
[    3.978942] systemd[1]: Cannot add dependency job for unit regenerate_ssh_host_keys.service, ignoring: Unit regenerate_ssh_host_keys.service failed to load: No such file or directory.
[    4.004705] systemd[1]: Cannot add dependency job for unit display-manager.service, ignoring: Unit display-manager.service failed to load: No such file or directory.
[    4.032458] systemd[1]: Expecting device dev-ttyAMA0.device...
[    4.045304] systemd[1]: Starting Forward Password Requests to Wall Directory Watch.
[    4.704226] systemd-udevd[143]: starting version 215
[    4.962039] EXT4-fs (mmcblk0p2): re-mounted. Opts: (null)
[    5.053593] gpiomem-bcm2835 3f200000.gpiomem: Initialised: Registers at 0x3f200000
[    5.768091] systemd-journald[138]: Received request to flush runtime journal from PID 1
[    6.386376] 8188eu: loading out-of-tree module taints kernel.
[    6.403773] Chip Version Info: CHIP_8188E_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
[    6.431397] EEPROM ID = 0x8129
[    6.434608] usbcore: registered new interface driver r8188eu
[    6.868880] R8188EU: Firmware Version 11, SubVersion 1, Signature 0x88e1
[    7.028570] random: crng init done
[    7.140332] usb 1-1.3: new high-speed USB device number 5 using dwc_otg
[    7.240718] MAC Address = 18:d6:c7:1b:ef:6c
[    7.243435] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    7.271908] usb 1-1.3: New USB device found, idVendor=12d1, idProduct=1c05
[    7.271918] usb 1-1.3: New USB device strings: Mfr=2, Product=1, SerialNumber=0
[    7.271924] usb 1-1.3: Product: HUAWEI HiLink
[    7.271930] usb 1-1.3: Manufacturer: HUAWEI
[    7.461132] usbcore: registered new interface driver cdc_ether
[    7.483968] usb 1-1.3: bad CDC descriptors
[    7.484228] usbcore: registered new interface driver rndis_host
[    7.502713] usb 1-1.3: bad CDC descriptors
[    7.505169] usbcore: registered new interface driver rndis_wlan
[    7.548777] usbcore: registered new interface driver usbserial
[    7.548866] usbcore: registered new interface driver usbserial_generic
[    7.548955] usbserial: USB Serial support registered for generic
[    7.593672] usbcore: registered new interface driver option
[    7.593766] usbserial: USB Serial support registered for GSM modem (1-port)
[    7.594152] option 1-1.3:1.0: GSM modem (1-port) converter detected
[    7.595872] usb 1-1.3: GSM modem (1-port) converter now attached to ttyUSB0
[    7.596019] option 1-1.3:1.1: GSM modem (1-port) converter detected
[    7.596473] usb 1-1.3: GSM modem (1-port) converter now attached to ttyUSB1
[    7.596560] option 1-1.3:1.2: GSM modem (1-port) converter detected
[    7.596941] usb 1-1.3: GSM modem (1-port) converter now attached to ttyUSB2
[    7.772821] Adding 102396k swap on /var/swap.  Priority:-1 extents:5 across:200700k SSFS
[    8.706557] PPP generic driver version 2.4.2
[    8.714794] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[    8.717672] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    8.764410] PPP BSD Compression module registered
[    8.785592] PPP Deflate Compression module registered
[    9.515789] R8188EU: INFO indicate disassoc
```

and ifconfig:

```
wlan0     Link encap:Ethernet  HWaddr 18:d6:c7:1b:ef:6c
          inet6 addr: fe80::ff4c:38ed:a62a:a374/64 Scope:Link
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

vallettea, 2017-07-12:
also:
```
root@raspberrypi:~# lsmod
Module                  Size  Used by
bluetooth             365511  2
ppp_deflate             4609  0
zlib_deflate           21885  1 ppp_deflate
bsd_comp                5992  0
ppp_async               8486  1
crc_ccitt               1771  1 ppp_async
ppp_generic            29466  7 ppp_async,bsd_comp,ppp_deflate
slhc                    5959  1 ppp_generic
option                 32679  1
usb_wwan                9483  1 option
usbserial              29943  5 option,usb_wwan
rndis_wlan             27555  0
rndis_host              7081  1 rndis_wlan
cdc_ether               6354  1 rndis_host
cfg80211              543027  1 rndis_wlan
rfkill                 20851  3 bluetooth,cfg80211
8188eu                809064  0
snd_bcm2835            24427  0
snd_pcm                98501  1 snd_bcm2835
snd_timer              23904  1 snd_pcm
snd                    70032  3 snd_timer,snd_bcm2835,snd_pcm
bcm2835_gpiomem         3940  0
uio_pdrv_genirq         3923  0
uio                    10204  1 uio_pdrv_genirq
fixed                   3285  0
ipv6                  406751  28
root@raspberrypi:~# ls /etc/modprobe.d/
50-8188eu.conf            blacklist-rtl8xxxu.conf   fbdev-blacklist.conf
blacklist-rtl8192cu.conf  dkms.conf
```

michaelgugino, 2017-07-15:
I'm not using a raspberry pi, but hopefully this will work for you and others:

https://github.com/michaelgugino/rtl8188EUS

I took the source from tp-link's site.  These are some mods to get the code to compile.  Next on my list to try is modifying the driver:  http://elixir.free-electrons.com/linux/v4.4/source/drivers/net/wireless/realtek/rtl8xxxu

That's in tree experimental source and should cover this chipset, but I haven't tried it yet.  It's also possible 8187 drivers might work.

vallettea, 2017-07-16:
I tried your solution @michaelgugino on raspberry, here is what i did and here is my problem:
Perhaps you have an idea (i'm not a linux expert at all)

```
apt-get update
apt-get install raspberrypi-kernel-headers
apt-get install binutils build-essential
git clone -b rpi-4.9.y --single-branch git://github.com/raspberrypi/linux.git
cd linux
cp /usr/src/linux-headers-4.9.35-v7+/Module.symvers .
cp /boot/config.txt .config
KSRC=$PWD
cd ..
git clone https://github.com/michaelgugino/rtl8188EUS.git
cd rtl8188EUS/
make
"******************************************"
"NO SKRC,we will use default KSRC"
"******************************************"
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.9.28-v7+/build M=/root/rtl8188EUS  modules
make[1]: *** /lib/modules/4.9.28-v7+/build: No such file or directory.  Stop.
Makefile:1367: recipe for target 'modules' failed
make: *** [modules] Error 2

mkdir /lib/modules/4.9.28-v7+/build
make
"******************************************"
"NO SKRC,we will use default KSRC"
"******************************************"
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.9.28-v7+/build M=/root/rtl8188EUS  modules
make[1]: Entering directory '/lib/modules/4.9.28-v7+/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/4.9.28-v7+/build'
Makefile:1367: recipe for target 'modules' failed
make: *** [modules] Error 2

```

michaelgugino, 2017-07-16:
Sorry about that, I missed didn't add 'export' the the line 'export KSRC=$PWD'.  That will add KSRC to your environment variables.  Just tells the vendor's make script where to find the linux source.

vallettea, 2017-07-16:
thanks for your help again.
I still had a problem:
```
make ARCH=armv7l CROSS_COMPILE= -C /root/linux M=/root/rtl8188EUS  modules
make[1]: Entering directory '/root/linux'
Makefile:628: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/root/linux'
Makefile:1367: recipe for target 'modules' failed
make: *** [modules] Error 2
```

but never mind I'll just buy another dongle.
best

closed 2017-07-16 by vallettea