lwfinger/rtl8188eu #219: possible buffer overflow warning compiling under Fedora 26

ian1632, 2017-07-14:
I'm guessing this is benign but it looks a bit scary.  GCC is 7.1.1 20170622.

[snip]
  CC [M]  /home/ian/src/rtl8188eu/os_dep/ioctl_linux.o
/home/ian/src/rtl8188eu/os_dep/ioctl_linux.c: In function ‘rtw_p2p_get_go_device_address’:
/home/ian/src/rtl8188eu/os_dep/ioctl_linux.c:3222:65: warning: ‘%.2X’ directive writing between 2 and 8 bytes into a region of size between 0 and 1 [-Wformat-overflow=]
   sprintf(go_devadd_str, "\n\ndev_add =%.2X:%.2X:%.2X:%.2X:%.2X:%.2X",
                                                                 ^~~~
/home/ian/src/rtl8188eu/os_dep/ioctl_linux.c:3222:26: note: using the range [0, 4294967295] for directive argument
   sprintf(go_devadd_str, "\n\ndev_add =%.2X:%.2X:%.2X:%.2X:%.2X:%.2X",
                          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/ian/src/rtl8188eu/os_dep/ioctl_linux.c:3222:3: note: ‘sprintf’ output between 29 and 65 bytes into a destination of size 27
   sprintf(go_devadd_str, "\n\ndev_add =%.2X:%.2X:%.2X:%.2X:%.2X:%.2X",
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    attr_content[0], attr_content[1], attr_content[2], attr_content[3], attr_content[4], attr_content[5]);
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[snip]

ian1632, 2017-07-14:
On closer inspection this looks like a problem, go_devadd_str looks to be 2 bytes too small.

It's declared as:

u8 go_devadd_str[17 + 10] = {0x00};

lwfinger, 2017-07-14:
Yes it is a bit small. I increased it to be large enough that the compiler will shut up.

closed 2017-07-14 by lwfinger