lwfinger/rtl8188eu #223: Can't get Access Point running

Alex66955, 2017-08-15:
I try to get the usb wireless adapter running as an access point.

I'am a bit lost and confused about the different tool and the rtl818eu driver.

**Embedded ARM System:**
Kernel: 4.1.10
Wireless Adapter: TL-WN722N

Supports this driver running for example an compatible wireless usb stick as AP ? 

As client everything works perfect:
- iwlist scan wlan0 works!
- Connection to a remote AP works! (with iwconfig)

But it seems that wpa_supplicant or hostapd can't work with that driver.

wpa_supplicant:

```
wpa_supplicant -Dwext -iwlan0 -c/etc/wpa_supplicant.conf
initialized wpa_supplicant
[ 2052.033960] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
[ 2053.351530] R8188EU: INFO indicate disassoc
wlan0: Driver does not support AP mode
wlan0: Driver does not support AP mode
```
hostapd (driver: nl80211):
```
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED
hostapd_free_hapd_data: Interface wlan0 wasn't started
```

Do someone has any suggestions how I can proceed?

Alex66955, 2017-08-16:
I proceed with hostpad from this repository because my current hostpad version seems to be not compatible with this driver.

I used the default configuration and give a try.
When I start hostapd I got following error:
```
Configuration file: /etc/hostapd.conf
ioctl[PRISM2_IOCTL_PRISM2_PARAM]: Bad address
Could not enable hostapd mode for interface wlan0
hostap driver initialization failed.
```

closed 2017-09-12 by Alex66955

evgeniysgs3, 2017-11-06:
It works with this params:
nterface=wlan0
driver=rtl871xdrv
ssid=testAP
hw_mode=g
channel=7
macaddr_acl=0
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=123123123

Thank you very much!!!