lwfinger/rtl8188eu #224: 8188eu: module verification failed: signature and/or required key missing - tainting kernel

Aque1228556367, 2017-08-28:
Thank you for your wireless driver. I compiled and installed it successfully, but my usb-wireless can't loaded when I reboot my system. My operating steps is:

· My OS: CentOS7 x86-64（Kernel Version: 3.10）
· My USB-Wireless: TP-LINK TL-WN725N 3.0

$ make all
......successfully
$ sudo make install
......successfully, generated file "8188eu.ko"
$ insmod 8188eu.ko
......successfully
$ ifconfig -a
......not find wireless, only "enp2s0", "lo", "virbr0" and "virbr0-nic"
I expect to appear "wlan0" or other wireless sign, but it isn't show.

$ lsmod | grep 8188
8188eu        768700    0

It has been installed successfully, isn't it?

Then I try to reboot system, execute "ifconfig -a", I don't find wireless.

When I showed my usb device with after inserting into my computer:
$ lsusb
Bus 003 Device 004: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter

Obviously, system had been recognized my usb-wireless. 

When I use "dmesg" command, I get it:
$ dmesg
8188eu: loading out-of-tree module taints kernel.
8188eu: module verification failed: signature and/or required key missing - tainting kernel

Is this error the cause of the problem? What should I do?

lwfinger, 2017-08-28:
You need to turn off secure booting. The out-of-kernel drivers cannot be signed in the manner required by secure booting.

Aque1228556367, 2017-08-29:
Thank you, I have turned off secure booting on my BIOS and restart my system, I find the driver 8188eu is disappeared. I don't find any info about "8188eu" in the log file which built cmd 'dmesg'. Then I retry to install 8188eu's driver, but I still can't find wireless in output of 'ifconfig'. What should I do ?

lucumt, 2018-03-02:
@Aque1228556367 Have you solved your problem? I met the same issue and want to know how you solve it.

maxpavlov, 2020-11-19:
Getting the same `8188eu: module verification failed: signature and/or required key missing - tainting kernel` error, no secure boot, Ubuntu 20.04, VMWare Workstation. Still digging...

lwfinger, 2020-11-19:
You need to turn off secure boot, or find some way to sign the module.

closed 2020-11-19 by lwfinger