lwfinger/rtl8188eu #225: Compilation problem on kernel 4.4.50-v7

Vishal1707, 2017-08-30:
Hello,

I have been trying to compile on linux Kali 4.4.50-v7 running on rasberry pi, but without any success. I am very new to linux and this cross compiling stuff.

I have issues in make, one probable reason may be not having right headers but I am not able to find what is the exact issue. Please find the error below:
 
```
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.4.50-v7/build M=/root/Documents/rtl8188eu  modules
make[1]: Entering directory '/usr/src/kernel'

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /root/Documents/rtl8188eu/core/rtw_ap.o
/bin/bash: ./scripts/recordmcount: cannot execute binary file: Exec format error
scripts/Makefile.build:258: recipe for target '/root/Documents/rtl8188eu/core/rtw_ap.o' failed
make[2]: *** [/root/Documents/rtl8188eu/core/rtw_ap.o] Error 126
Makefile:1402: recipe for target '_module_/root/Documents/rtl8188eu' failed
make[1]: *** [_module_/root/Documents/rtl8188eu] Error 2
make[1]: Leaving directory '/usr/src/kernel'
Makefile:151: recipe for target 'modules' failed
make: *** [modules] Error 2
```

Looking forward to any help. 
Thanks in advance.

shamsbppg, 2017-09-10:
same problem for me as well.May be the **linux-headers** to be installed first.But i cannot get any way to install the linux-header for **rpi arm v7l**.
I tried `apt-cache search linux-headers`
but didn't understand what should install for arm v7l.
**is there anyone who figure out this problem?**

ft-1408, 2017-09-29:
Hi there - totally same problem. I think there no 4.4.50-v7 linux headers for ARM achitecture. i found only for armhf. https://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/. But using them pointless

lwfinger, 2017-09-30:
If someone has built a kernel, there ARE headers available. You just need to find them. I have no boxes with ARM, thus I cannot help. Where are you getting your distribution?

ft-1408, 2017-09-30:
> Where are you getting your distribution?
At https://www.offensive-security.com/kali-linux-arm-images/ - at my raspberry pi 2 working only this one kali-2017.2-rpi0w-nexmon.img.xz