lwfinger/rtl8188eu #228: System freezing when connecting a tablet

colibris79, 2017-10-19:
Hi, 
I've been using this driver for more than 3 years and I love it. Usually it didn't give any problems but today I got a new tablet (Android 6.1) and after a few seconds connected to the hotspot the whole system freezes. I've tried updating to the last version of the driver (master branch) but I still have the same problem. What could it be? Would a change in the kernel fix it? Right now I am using last version of the driver (master) in my Linux Mint 17.3 (Ubuntu 14.04) and 3.19.0-32 kernel. 
Any suggestions?
Thanks!

lwfinger, 2017-10-19:
Is the tablet the hotspot? If so, it is doing something that the driver is not expecting. As you are using the latest 8188eu driver, changing the kernel is unlikely to help, but it might be worth the effort. Version 3.19 is pretty old.

When the PC freezes, there is some kind of oops message that is logged. Unless you can provide that, there is no way I can diagnose and fix the problem.

colibris79, 2017-10-19:
Hi lwfinger, thanks for your fast response. The hotspot is created in the PC and the tablet is connected to it. I have made some progress. Just for trying I compiled rtl8188eu-4.1.8_9499 and couldn't connect any devices to the hotspot (authentication problem, even though I was using the same hostapd.conf files from my previous driver) and when I was about to give up I just created an open hotspot (no password) and the tablet connects just fine. I still have no idea why my devices can't connect to the hotspot under that device. 
Any suggestions about which kernel should I install? 
I've looked into /var/log/syslog this is from one of the last crashes:

[log.txt](https://github.com/lwfinger/rtl8188eu/files/1399280/log.txt)

lwfinger, 2017-10-19:
There is no crash information in that log.

colibris79, 2017-10-19:
Sorry about that. Any other place I could look for the log then?
Or any other suggestion about which kernel could I try?

colibris79, 2017-10-20:
I think it has something to do with hostapd as well. I'll try messing around with it let's see where I can get.
Here someone had a similar problem:
https://github.com/oblique/create_ap/issues/23

colibris79, 2017-10-20:
I have installed kernel 4.4.0-93 and recompiled the last master driver and tried again but it still freezes (both open and secure hotspots). Even tried the most basic open hostapd.conf file but it still happens. So I am thinking about going back to rtl8188eu-4.1.8_9499 and try to get the secure hotspot working. 
Would this be an issue of the hostapd config file?

colibris79, 2017-10-20:
Well, after a long time trying I haven't been able to create a secure hotspot with rtl8188eu-4.1.8_9499 (open hotspot is working fine). I have tried different configurations but I'm still getting the authentication problem with all devices.

colibris79, 2017-10-21:
Ok. Finally I have managed to create a secure hotspot but has to be WEP and the devices can connect to it. Haven't been able to create a WPA one though.

colibris79, 2017-10-22:
Just as a final comment and in case it can help you with anything, I have tried a different usb device with rtl8192eu driver and your hostapd and it is working fine with all my devices, both WPA and WEP. This is the driver I am using:
https://launchpad.net/~hanipouspilot/+archive/ubuntu/rtlwifi/+build/10629089

But I guess the other two githubs version should be working as well:
https://github.com/masterzorag/RTL8192EU-linux
https://github.com/Mange/rtl8192eu-linux-driver

Thanks again lwfinger for all your work. I'll try future versions of your drivers and if I can get WPA to work with that specific device. :)