lwfinger/rtl8188eu #234: Inconsistent response to SIOCGIWESSID ioctl

spl237, 2017-11-23:
When not associated with an AP, other wifi device drivers respond to the SIOCGIWESSID ioctl with a zero-length string for the SSID; this is the behaviour expected by dhcpcd.

This driver returns an error code (-1) from the ioctl call - this causes dhcpcd to assume that the device is not a wireless interface and therefore it fails to work correctly with it thereafter.

It would be good if the behaviour of this driver could be brought into line with that of other drivers for similar devices; that is for the response to SIOCGIWESSID to be a zero-length string and a success code rather than an error.

lwfinger, 2017-11-23:
I think this issue has been fixed. Please test.

spl237, 2017-11-23:
That's great! I'll try it tomorrow and report back. Many thanks for the prompt fix.

spl237, 2017-11-24:
Yes, that's fixed this issue - many thanks!

closed 2017-11-24 by spl237