lwfinger/rtl8188eu #235: Device does not appear in rfkill list

spl237, 2017-11-23:
When the userland rfkill application is run with the "list wifi" argument, this device is not found - I assume this is something to do with the rfkill code in the driver.

solsticedhiver, 2019-04-19:
Same issue here with **TL-WN725N** dongle.

So this is not useable because networkmanager can't rfkill unblock it

lwfinger, 2019-04-19:
What do 'rfkill list' and 'iw dev' show?

solsticedhiver, 2019-04-20:
`rfkill list` or `iw dev` do not show the interface at all.

The device is shown in 
    $ lsusb
    [...]
    Bus 003 Device 009: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
    [...]

or

    $ ip link
    1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
        link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    2: enp1s0: ...
        ...
    3: wlp2s0: ...
        ...
    9: wlxc46e1fxxxxxx: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN mode DEFAULT group default qlen 1000
        link/ether xx:xx:xx:xx:xx:xx brd ff:ff:ff:ff:ff:ff

lwfinger, 2019-04-20:
You must be using the "master" branch. That one does not use mac80211, and is somewhat incompatible with some distros user space. Checkout branch v5.2.2.4 and try it.

solsticedhiver, 2019-04-20:
Yes. Works better with v5.2.2.4.
So was it written somewhere and I have missed it or one have to guess that we need v5.2.2.4 ?

The issue was the same on archlinux.

The mainline kernel driver r8188eu (that I am guessing is the same driver ?) has also the same issue.

lwfinger, 2019-04-20:
Yes, the driver in the kernel also does not use mac80211. It needs to be replaced, but I do not have any time.

lpapp, 2019-12-05:
Broken even from the v5.2.2.4 branch. Do you know what else to try? I am using Archlinux.

lwfinger, 2019-12-05:
Are you compiling it witl CFG80211 enabled? It may not pay any attention to rfkill. Why is that a problem?

lpapp, 2019-12-05:
I am not sure. Just make. I do not specify anything else.

lpapp ~ $ sudo wpa_supplicant -Dwext -B -i wlp0s29u1u2 -c /etc/wpa_supplicant/example.conf
Successfully initialized wpa_supplicant
rfkill: Cannot get wiphy information
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
lpapp ~ $ sudo wpa_supplicant -B -i wlp0s29u1u2 -c /etc/wpa_supplicant/example.conf
Successfully initialized wpa_supplicant
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlp0s29u1u2 disabled_11b_rates=0
wlp0s29u1u2: Failed to initialize driver interface
lpapp ~ $

lpapp, 2019-12-06:
Somehow, I get this. It does not make it into dkms status. Not sure why.

lpapp ~ $ lsmod | grep 8188
8188eu                798720  0
lpapp ~ $ dkms status
Error! Could not locate dkms.conf file.
File: /var/lib/dkms/8188eu/1.0.4930848/source/dkms.conf does not exist.
lpapp ~ $

lpapp, 2019-12-06:
OK, I rebuilt the driver. Now, when I use the v5.2.2.4 branch. And I plug in my device, the kernel crashes. Why?

5.4.2-arch1-1

Are you supposed to support 5.4.2 as that seems to be a bit ahead of 5.2.2.4 in your branch name?

SteveBetter, 2022-10-22:
@lpapp Have you solved the problem? Same here with linux kernel 6.0.2

lwfinger, 2022-10-22:
@lpapp: Your kernel crashes because it detected an error. What error? Sorry, that information is only available on your computer and no error logging was provided here.

The branch name is the version of the driver as specified by Realtek. The kernel number comes from Linus Torvalds. There is no reason that they should have any correspondence. In addition, I am a volunteer. It is not my responsibility to support anything! This kind of demand just makes me furious, which is why I took so long to answer!!!

closed 2022-10-22 by lwfinger

lpapp, 2022-10-22:
You must have misread something. I had no demand.