lwfinger/rtl8188eu #236: Monitor mode issue with TL-W722N V2

safal2000, 2017-11-25:
Hello, 
I have brought Tp-link 722n and installed all the suitable drivers it's working in my Kali 4.13.0 but when I tried to change it to monitor mode it showed up an error invalid argument. Can any one help me out.
Anticipating an immediate response.
Thank you.

kimocoder, 2017-11-25:
This driver doesn't support monitor mode..

safal2000, 2017-11-25:
Yesterday when I tried after installing lwfinger driver it support and I
was able to sniff package  but after when I rebooted my Kali and booted
again it showed up ab error

lwfinger, 2017-11-25:
That comment makes no sense; however as you have detected the error, you get to debug it. Sensible patches welcomed.

safal2000, 2017-11-26:
Alright can u tell me what's the meaning of SET 8B06 error invalid argument.
And I am able to switch it to auto mode , master mode but not monitor mode
why?

kimocoder, 2017-11-26:
@safal2000 The driver does NOT support monitor mode at this point.
with time, maybe it will be added in the future.

safal2000, 2017-11-26:
Thank you Christian

bab-777, 2017-12-19:
hello, Please some one help me! I have tp-link tl-722n v2 and I want to install the driver for kali Linux. I'm new in Linux and I can't find the driver. when  I try to use airmon-ng command the result is : 
PHY  	Interface	        Driver		Chipset

null	        wlan0		r8188eu

PrivetVsemRu, 2018-05-21:
when driver will support monitor mode?

lwfinger, 2018-05-21:
When you modify the driver to support that feature. I have no interest in it, and apparently Realtek does not either.

closed 2018-05-21 by lwfinger