lwfinger/rtl8188eu #237: MAC address change failure

andrewshadura, 2017-12-04:
Hi,

I'm not sure how relevant this is at this moment, but at least some of the previous versions used to have issues with changing the MAC address:
```
Jan 14 10:56:18 debian NetworkManager[3929]: <info>  [1484405778.9762] device (wlxe8de271f313a): state change: unavailable -> disconnected (reason 'supplicant-available') [20 30 42]
Jan 14 10:56:18 debian kernel: [ 1232.457289] R8188EU: ERROR indicate disassoc
Jan 14 10:56:19 debian NetworkManager[3929]: <warn>  [1484405779.0836] device (wlxe8de271f313a): set-hw-addr: new MAC address XX:XX:XX:XX:XX:XX not successfully set (scanning)
```
For more details, see https://bugs.debian.org/849122#26.

I’m not sure you the actual upstream, but if you are, it would be great to get that fixed.

uti5, 2023-05-10:
Changing the MAC address is still not possible. aircrack-ng/rtl8188eus does this, so it isn't impossible for this adapter, as the readme claims.

lwfinger, 2023-05-10:
That one you will need to debug yourself.

closed 2023-05-10 by lwfinger