lwfinger/rtl8188eu #238: Compile error for 4.15.0-rc3

mi12078, 2017-12-21:
When compiling for said kernel I get the following error:
```
$ make all
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.15.0-rc3/build M=/home/stefan/rtl8188eu  modules
make[1]: Entering directory '/home/stefan/linux_kernel/build/4.15.0-rc3'
  CC [M]  /home/stefan/rtl8188eu/core/rtw_ap.o
In file included from /home/stefan/linux_kernel/linux-stable/include/linux/list.h:9:0,
                 from /home/stefan/linux_kernel/linux-stable/include/linux/rculist.h:10,
                 from /home/stefan/linux_kernel/linux-stable/include/linux/sched/signal.h:5,
                 from /home/stefan/rtl8188eu/include/osdep_service.h:25,
                 from /home/stefan/rtl8188eu/core/rtw_ap.c:22:
/home/stefan/linux_kernel/linux-stable/include/linux/kernel.h:6:10: fatal error: stdarg.h: No such file or directory
 #include <stdarg.h>
          ^~~~~~~~~~
compilation terminated.
make[4]: *** [/home/stefan/linux_kernel/linux-stable/scripts/Makefile.build:311: /home/stefan/rtl8188eu/core/rtw_ap.o] Error 1
make[3]: *** [/home/stefan/linux_kernel/linux-stable/Makefile:1502: _module_/home/stefan/rtl8188eu] Error 2
make[2]: *** [Makefile:146: sub-make] Error 2
make[1]: *** [Makefile:24: __sub-make] Error 2
make[1]: Leaving directory '/home/stefan/linux_kernel/build/4.15.0-rc3'
make: *** [Makefile:152: modules] Error 2
```

The kernel which was booted at the time of compiling was 4.14.6-1-ARCH. There are no problems when building this module for aforementioned kernel.
4.15.0-rc3 was built out of source if it is of any importance.
Tried building with 4.15.0-rc3 booted, error still persists. 
Could it be that I somehow miscompiled the kernel?

riddicc, 2018-01-14:
same here... but with 4.15.r180107.gb2cd1df-1 (Manjaro Linux)...
```
make -C /lib/modules/4.15.0-1-MANJARO/build SUBDIRS=/usr/home/aw/devel/mydb/myswapb modules
make[1]: Entering directory '/usr/lib/modules/4.15.0-1-MANJARO/build'
  CC [M]  /usr/home/aw/devel/mydb/myswapb/myswap.o
In file included from /usr/home/aw/devel/mydb/myswapb/myswap.c:90:0:
./include/linux/kernel.h:6:10: fatal error: stdarg.h: No such file or directory
 #include <stdarg.h>
```
-arne

riddicc, 2018-01-14:
adding -I/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/include to ccflags-y in Makefile helped... :)
> ccflags-y = -Wno-declaration-after-statement -mpopcnt -I/usr/lib/gcc/x86_64-pc-linux-gnu/7.2.1/include
-arne

mi12078, 2018-01-28:
@riddicc Thanks, that helped solve the include error

franciscofreire, 2018-03-22:
I had the same problem on Arch Linux (4.15.11-1-ARCH), adding -I/usr/lib/gcc/x86_64-pc-linux-gnu/**7.3.1**/include to ccflags-y solved the issue

smonv, 2018-04-03:
@riddicc Thanks for the solution. This issue made me confuse a lots after long time not use my arch system.

jwoehrle, 2018-06-04:
helped me with fedora 28. The right directory is `-I/usr/lib/gcc/x86_64-redhat-linux/8/include`