lwfinger/rtl8188eu #239: make all command doesnt work.

sharkersharif, 2018-01-23:
hi, i am a newbie .. help me please.

"make all" command shows:

make ARCH=arm CROSS_COMPILE= -C /lib/modules/3.10.49-11903637/build M=/root/Desktop/rtl8188eu  modules
make[1]: *** /lib/modules/3.10.49-11903637/build: No such file or directory.  Stop.
Makefile:151: recipe for target 'modules' failed
make: *** [modules] Error 2

Additional info :
root@kali:~/Desktop/rtl8188eu# cat /proc/version            Linux version 3.10.49-11903637 (dpi@SWDG9704) (gcc version 4.8 (GCC) ) #1 SMP PREEMPT Fri Sep 15 19:32:22 KST 2017
root@kali:~/Desktop/rtl8188eu# uname -a                     Linux kali 3.10.49-11903637 #1 SMP PREEMPT Fri Sep 15 19:32:22 KST 2017 armv7l GNU/Linux

micenote, 2018-02-07:
check link： 
#ls -l /lib/modules/3.10.0-693.el7.x86_64

if have link：
#rm ./build
link:
#ln -s /usr/src/kernels/3.10.0-693.el7.x86_64/ /lib/modules/3.10.0-693.el7.x86_64/build

make again.

lwfinger, 2018-02-07:
The critical test is 'ls /lib/modules/$(uname -r)/build'. The resulting output MUST show a directory named "include".