lwfinger/rtl8188eu #241: Fails: Building kernel module for Nano 150M Banana Pi WiFi USB 2.0 Dongle (8188eu) for Banana Pi BPI-M3

arkhipenko, 2018-02-11:
I bought this adapter: Nano 150M Banana Pi WiFi USB 2.0 Dongle (here) because it says Banana Pi.
So one would assume it works with all Banana Pi’s. No such luck: 3.4.39-BPI-M3-Kernel does not have appropriate module included.

USB device ID: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter

I found the kernel source [here](https://github.com/BPI-SINOVOIP/BPI-M3-bsp).
I found the required module source here.

It compiles almost to the end, but then it breaks:

```
root@bpi-iot-ros-ai:~/workspace/RTL8188EU/rtl8188eu# make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/3.4.39-BPI-M3-Kernel/build M=/home/pi/workspace/RTL8188EU/rtl8188eu  modules
make[1]: Entering directory '/media/mikey/BPI-USER/github/build/release/BPI-M3-bsp/linux-sunxi'

  WARNING: Symbol version dump /media/mikey/BPI-USER/github/build/release/BPI-M3-bsp/linux-sunxi/Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_ap.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_br_ext.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_cmd.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_debug.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_efuse.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_ieee80211.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_io.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_ioctl_set.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_iol.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_led.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_mlme.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_mlme_ext.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_mp.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_mp_ioctl.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_pwrctrl.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_p2p.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_recv.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_rf.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_security.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_sreset.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_sta_mgt.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_wlan_util.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/core/rtw_xmit.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/HalHWImg8188E_MAC.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/HalHWImg8188E_BB.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/HalHWImg8188E_RF.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/HalPhyRf.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/HalPhyRf_8188e.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/HalPwrSeqCmd.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/Hal8188EPwrSeq.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/Hal8188ERateAdaptive.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/hal_intf.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/hal_com.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/odm.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/odm_debug.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/odm_interface.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/odm_HWConfig.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/odm_RegConfig8188E.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/odm_RTL8188E.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/rtl8188e_cmd.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/rtl8188e_dm.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/rtl8188e_hal_init.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/rtl8188e_mp.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/rtl8188e_phycfg.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/rtl8188e_rf6052.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/rtl8188e_rxdesc.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/rtl8188e_sreset.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/rtl8188e_xmit.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/rtl8188eu_led.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/rtl8188eu_recv.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/rtl8188eu_xmit.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/usb_halinit.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/hal/usb_ops_linux.o
  CC [M]  /home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.o
/home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.c:7814:2: error: unknown field ‘private’ specified in initializer
  .private = rtw_private_handler,
  ^
/home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.c:7814:13: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
  .private = rtw_private_handler,
             ^
/home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.c:7814:13: note: (near initialization for ‘rtw_handlers_def.get_wireless_stats’)
/home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.c:7815:2: error: unknown field ‘private_args’ specified in initializer
  .private_args = (struct iw_priv_args *)rtw_private_args,
  ^
/home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.c:7815:18: warning: excess elements in struct initializer
  .private_args = (struct iw_priv_args *)rtw_private_args,
                  ^
/home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.c:7815:18: note: (near initialization for ‘rtw_handlers_def’)
/home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.c:7816:2: error: unknown field ‘num_private’ specified in initializer
  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
  ^
/home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.c:7816:17: warning: excess elements in struct initializer
  .num_private = sizeof(rtw_private_handler) / sizeof(iw_handler),
                 ^
/home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.c:7816:17: note: (near initialization for ‘rtw_handlers_def’)
/home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.c:7817:2: error: unknown field ‘num_private_args’ specified in initializer
  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
  ^
/home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.c:7817:22: warning: excess elements in struct initializer
  .num_private_args = sizeof(rtw_private_args) / sizeof(struct iw_priv_args),
                      ^
/home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.c:7817:22: note: (near initialization for ‘rtw_handlers_def’)
scripts/Makefile.build:307: recipe for target '/home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.o' failed
make[2]: *** [/home/pi/workspace/RTL8188EU/rtl8188eu/os_dep/ioctl_linux.o] Error 1
Makefile:1370: recipe for target '_module_/home/pi/workspace/RTL8188EU/rtl8188eu' failed
make[1]: *** [_module_/home/pi/workspace/RTL8188EU/rtl8188eu] Error 2
make[1]: Leaving directory '/media/mikey/BPI-USER/github/build/release/BPI-M3-bsp/linux-sunxi'
Makefile:151: recipe for target 'modules' failed
make: *** [modules] Error 2
root@bpi-iot-ros-ai:~/workspace/RTL8188EU/rtl8188eu# 
```

Any help in finishing this compile process is greatly appreciated!

Thanks Anatoli

lwfinger, 2018-02-11:
This is very gutsy, and a little bit stupid. You get the source from some other repo, and then expect me to fix it. No thank you!!

closed 2018-02-11 by lwfinger

arkhipenko, 2018-02-11:
No need to be rude. I asked for help, that's it. If you can't help just say so. I was not expecting a miracle but was also not expecting such a rude rebuke... Good luck with your module with such attitude.