lwfinger/rtl8188eu #242: Can this driver support monitor mode?

102464, 2018-02-21:
i want to run aircrack-ng to crack wifi password, but it seemed that i can't set it to monitor mode.
root@ubuntu-OptiPlex-380:~# iwconfig
lo        no wireless extensions.

wlan0     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality=0/100  Signal level=0 dBm  Noise level=0 dBm
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

docker0   no wireless extensions.

eth0      no wireless extensions.

root@ubuntu-OptiPlex-380:~# ifconfig wlan0 down
root@ubuntu-OptiPlex-380:~# iwconfig wlan0 mode monitor 
Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan0 ; Invalid argument.
root@ubuntu-OptiPlex-380:~# ifconfig wlan0 up
root@ubuntu-OptiPlex-380:~#

lwfinger, 2018-02-21:
No, it cannot support monitor mode. If you want that feature, you will need to write it yourself.

closed 2018-02-21 by lwfinger

safal2000, 2018-02-21:
Hello there,
can u give me the written script of the code?? to write the driver

sinlice, 2022-02-25:
您好 ，请问这个功能有加上吗

lwfinger, 2022-02-25:
什么特点？