lwfinger/rtl8188eu #246: Driver causes BUG in CentOS kernel

zcooler, 2018-03-14:
OS: CentOS 7
Kernel: 3.10.0-693.21.1.el7.x86_64
Driver compiles from this github repo with two warnings:

> /root/wifi/rtl8188eu/os_dep/os_intfs.c:701:2: предупреждение: несовместимый тип указателя в инициализации [по умолчанию включена]
>   .ndo_select_queue = rtw_select_queue,
>   ^
> /root/wifi/rtl8188eu/os_dep/os_intfs.c:701:2: предупреждение: (где-то рядом с инициализацией для «rtw_netdev_ops.<anonymous>.ndo_select_queue») [по умолчанию включена]

hostapd compiles from this github repo without errors.
Starts without errors.
Wifi interface added to bridge

```
# brctl show bridge0
bridge name	bridge id		STP enabled	interfaces
bridge0		8000.74dada052ab9	no		eth0
 							wlan0
```
Ios devices can't connect. hostapd shows in log:
```
Mar 14 09:59:15 ofc-gw hostapd: wlan0: STA f4:f1:5a:42:77:65 IEEE 802.11: associated
Mar 14 09:59:15 ofc-gw hostapd: wlan0: STA f4:f1:5a:42:77:65 IEEE 802.11: associated
Mar 14 09:59:15 ofc-gw hostapd: wlan0: STA f4:f1:5a:42:77:65 WPA: received EAPOL-Key 2/4 Pairwise with unexpected replay counter
Mar 14 09:59:16 ofc-gw hostapd: wlan0: STA f4:f1:5a:42:77:65 WPA: received EAPOL-Key 2/4 Pairwise with unexpected replay counter
Mar 14 09:59:20 ofc-gw hostapd: wlan0: STA f4:f1:5a:42:77:65 IEEE 802.11: deauthenticated due to local deauth request
Mar 14 09:59:20 ofc-gw hostapd: wlan0: STA f4:f1:5a:42:77:65 IEEE 802.11: disassociated
```
Android devices connected successfully. 
```
Mar 14 10:00:43 ofc-gw hostapd: wlan0: STA 14:6b:72:7b:7c:bc IEEE 802.11: associated
Mar 14 10:00:43 ofc-gw hostapd: wlan0: STA 14:6b:72:7b:7c:bc IEEE 802.11: associated
Mar 14 10:00:43 ofc-gw hostapd: wlan0: STA 14:6b:72:7b:7c:bc WPA: received EAPOL-Key 2/4 Pairwise with unexpected replay counter
Mar 14 10:00:44 ofc-gw hostapd: wlan0: STA 14:6b:72:7b:7c:bc WPA: received EAPOL-Key 2/4 Pairwise with unexpected replay counter
Mar 14 10:00:44 ofc-gw hostapd: wlan0: STA 14:6b:72:7b:7c:bc RADIUS: starting accounting session 5AA8D626-00000000
Mar 14 10:00:44 ofc-gw hostapd: wlan0: STA 14:6b:72:7b:7c:bc WPA: pairwise key handshake completed (RSN)

```
BUT!
After about minute, in system messages log pops an error:

```
Mar 14 10:01:44 ofc-gw kernel: BUG: scheduling while atomic: swapper/0/0/0x10000100
Mar 14 10:01:44 ofc-gw kernel: Modules linked in: 8188eu(OE) xt_nat veth ipt_MASQUERADE nf_nat_masquerade_ipv4 nf_log_ipv4 nf_log_common xt_LOG ip_set_hash_ip nf_conntrack_irc nf_nat_pptp nf_nat_proto_gre nf_conntrack_pptp nf_conntrack_proto_gre nf_nat_ftp iptable_nat nf_conntrack_ipv4 nf_conntrack_ftp nf_defrag_ipv4 nf_nat_ipv4 xt_addrtype xt_conntrack nf_nat nf_conntrack br_netfilter 8021q garp mrp bridge stp llc xt_set ip_set nfnetlink iptable_filter dm_thin_pool dm_persistent_data dm_bio_prison dm_bufio iTCO_wdt iTCO_vendor_support snd_hda_codec_realtek snd_hda_codec_generic hp_wmi sparse_keymap gpio_ich rfkill snd_hda_intel ppdev snd_hda_codec mei_wdt coretemp snd_hda_core kvm snd_hwdep snd_seq snd_seq_device snd_pcm irqbypass snd_timer pcspkr snd sg lpc_ich soundcore parport_pc parport shpchp tpm_infineon wmi
Mar 14 10:01:44 ofc-gw kernel: mei_me mei acpi_cpufreq ip_tables xfs libcrc32c sd_mod crc_t10dif crct10dif_generic sr_mod cdrom crct10dif_common ata_generic i915 ahci libahci pata_acpi i2c_algo_bit drm_kms_helper libata serio_raw syscopyarea sysfillrect sysimgblt fb_sys_fops e1000e drm r8169 mii ptp pps_core video i2c_core dm_mirror dm_region_hash dm_log dm_mod
Mar 14 10:01:44 ofc-gw kernel: CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W  OE  ------------   3.10.0-693.21.1.el7.x86_64 #1
Mar 14 10:01:44 ofc-gw kernel: Hardware name: Hewlett-Packard HP Compaq 8000 Elite SFF PC/3646h, BIOS 786G7 v01.02 10/22/2009
Mar 14 10:01:44 ofc-gw kernel: Call Trace:
Mar 14 10:01:44 ofc-gw kernel: <IRQ>  [<ffffffff816ae7c8>] dump_stack+0x19/0x1b
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff816a8bf7>] __schedule_bug+0x64/0x72
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff816b4034>] __schedule+0x9a4/0xa30
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff810c79ec>] ? try_to_wake_up+0x18c/0x350
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff810c4bd6>] __cond_resched+0x26/0x30
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff816b438a>] _cond_resched+0x3a/0x50
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff811e40fc>] kmem_cache_alloc_trace+0x3c/0x200
Mar 14 10:01:44 ofc-gw kernel: [<ffffffffc07c969b>] ? rtw_chk_hi_queue_cmd+0x25/0xb2 [8188eu]
Mar 14 10:01:44 ofc-gw kernel: [<ffffffffc07c969b>] rtw_chk_hi_queue_cmd+0x25/0xb2 [8188eu]
Mar 14 10:01:44 ofc-gw kernel: [<ffffffffc083ac6f>] usb_write_port_complete+0x6e/0x3a8 [8188eu]
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff81496580>] __usb_hcd_giveback_urb+0x80/0x110
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff81497453>] usb_giveback_urb_bh+0xb3/0x120
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff81093e83>] tasklet_action+0x83/0x140
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff81094035>] __do_softirq+0xf5/0x280
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff816c3afc>] call_softirq+0x1c/0x30
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff8102d435>] do_softirq+0x65/0xa0
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff810943b5>] irq_exit+0x105/0x110
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff816c4d96>] do_IRQ+0x56/0xf0
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff816b7362>] common_interrupt+0x162/0x162
Mar 14 10:01:44 ofc-gw kernel: <EOI>  [<ffffffff81530b04>] ? cpuidle_enter_state+0x54/0xd0
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff81530afd>] ? cpuidle_enter_state+0x4d/0xd0
Mar 14 10:01:44 ofc-gw kernel: [<ffffffff81530c5e>] cpuidle_idle_call+0xde/0x230
[... 158 lines cut ...]
uuid=12345678-9abc-def0-1234-56789abcdef0
device_name=RTL8188EU
manufacturer=Realtek
model_name=RTW_SOFTAP
model_number=WLAN_CU
serial_number=12345
device_type=6-0050F204-1
os_version=01020300
config_methods=label display push_button keypad
driver=rtl871xdrv
beacon_int=100
hw_mode=g
ieee80211n=1
wme_enabled=1
ht_capab=[SHORT-GI-20][SHORT-GI-40]
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
max_num_sta=8
wpa_group_rekey=86400

```

Device: TL-WN722N

If needed any additional info - please post, I`ll add it.

lwfinger, 2018-03-14:
I do not have any systems with kernels as old as 3.10, thus I cannot see the warnings you get while compiling. I will try to find that problem and fix it, but that will take too much time for now.

I have fixed the scheduling while atomic bug. In addition, the "Operation not supported" situation will log the actual code that is failing.

Please pull and run again.

zcooler, 2018-03-14:
pulled, compiled under 

```
# uname -r
4.15.9-1.el7.elrepo.x86_64

```

Output of hostapd not changed. 

```
random: Trying to read entropy from /dev/random
Configuration file: ../../rtl_hostapd.conf
drv->ifindex=6
Configure bridge bridge0 for EAPOL traffic.
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
Allowed channel: mode=1 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=12 freq=2467 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=13 freq=2472 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=36 freq=5180 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=40 freq=5200 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=44 freq=5220 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=48 freq=5240 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=149 freq=5745 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=153 freq=5765 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=157 freq=5785 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=161 freq=5805 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=165 freq=5825 MHz max_tx_power=0 dBm
Completing interface initialization
Mode: IEEE 802.11g  Channel: 1  Frequency: 2412 MHz
RATE[0] rate=10 flags=0x1
RATE[1] rate=20 flags=0x1
RATE[2] rate=55 flags=0x1
RATE[3] rate=110 flags=0x1
RATE[4] rate=60 flags=0x0
RATE[5] rate=90 flags=0x0
RATE[6] rate=120 flags=0x0
RATE[7] rate=180 flags=0x0
RATE[8] rate=240 flags=0x0
RATE[9] rate=360 flags=0x0
RATE[10] rate=480 flags=0x0
RATE[11] rate=540 flags=0x0
Flushing old station entries
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Could not connect to kernel driver.
Deauthenticate all stations
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=2
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
[... 420 lines cut ...]
ioctl(4, SIOCSIWMODE, 0x7ffc9d20f730)   = 0
close(4)                                = 0
close(6)                                = 0
close(5)                                = 0
close(3)                                = 0
gettimeofday({1521045714, 157407}, NULL) = 0
write(1, "ELOOP: remaining socket: sock=8 eloop_data=0x1be1ec0 user_data=(nil) handler=0x424e60\n", 86ELOOP: remaining socket: sock=8 eloop_data=0x1be1ec0 user_data=(nil) handler=0x424e60
) = 86
exit_group(1)                           = ?
+++ exited with 1 +++

```

offtopic
Also, seems hostapd do nothing on line

> Configure bridge bridge0 for EAPOL traffic.

If wlan0 was not in bridge, it does not add wlan0 to bridge0, even if 

> bridge=bridge0

in config.
Or I am misunderstood this option.
/offtopic