lwfinger/rtl8188eu #250: This driver does not support fedora

Flowertome, 2018-04-05:
[image]

Flowertome, 2018-04-05:
[image]

lwfinger, 2018-04-05:
Sorry, I do not read Chinese. At least I think that is the language you have in those pictures.

The driver does support all versions of Fedora. Please write out the problem you are having, run the text through Google Translate, and post the resulting English. I suspect that you are missing the kernel headers package, or whatever Fedora calls it.

amibal, 2018-05-11:
I'm running this on Fedora 27, works pretty good for me.

closed 2018-05-11 by lwfinger