lwfinger/rtl8188eu #251: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter     __read_overflow2(); error when building v4.1.8_9499 branch

moham96, 2018-04-10:
Hi, 
I read in #112  that the  v4.1.8_9499 branch support `cfg80211` :

> 
> The master branch is based on Realtek's version "v4.1.4_6773.20130222".
> The v.1.8 version is based on version "v4.1.8_9499.20131104".
> 
> I do not know all the differences, but the main one is that 4.1.4 does not implement the cfg80211 calls, which is why cannot work with a recent hostapd. Version 4.1.8 does work with cfg80211

So now I'm trying to build the v4.1.8_9499 branch but on a 4.14 kernel. and when I run make I get the following error 
```

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.14.31-1-MANJARO/build M=/home/mohammad/rtl8188eu  modules
make[1]: Entering directory '/usr/lib/modules/4.14.31-1-MANJARO/build'
  CC [M]  /home/mohammad/rtl8188eu/core/rtw_cmd.o
  CC [M]  /home/mohammad/rtl8188eu/core/rtw_security.o
  CC [M]  /home/mohammad/rtl8188eu/core/rtw_debug.o
  CC [M]  /home/mohammad/rtl8188eu/core/rtw_io.o
  CC [M]  /home/mohammad/rtl8188eu/core/rtw_ioctl_query.o
  CC [M]  /home/mohammad/rtl8188eu/core/rtw_ioctl_set.o
  CC [M]  /home/mohammad/rtl8188eu/core/rtw_ieee80211.o
  CC [M]  /home/mohammad/rtl8188eu/core/rtw_mlme.o
  CC [M]  /home/mohammad/rtl8188eu/core/rtw_mlme_ext.o
  CC [M]  /home/mohammad/rtl8188eu/core/rtw_wlan_util.o
In file included from ./include/linux/bitmap.h:9:0,
                 from ./include/linux/cpumask.h:12,
                 from ./arch/x86/include/asm/paravirt.h:17,
                 from ./arch/x86/include/asm/irqflags.h:65,
                 from ./include/linux/irqflags.h:16,
                 from ./include/linux/rcupdate.h:39,
                 from ./include/linux/rculist.h:11,
                 from ./include/linux/sched/signal.h:5,
                 from /home/mohammad/rtl8188eu/include/osdep_service.h:25,
                 from /home/mohammad/rtl8188eu/core/rtw_wlan_util.c:23:
In function ‘memcpy’,
    inlined from ‘update_bmc_sta_support_rate’ at /home/mohammad/rtl8188eu/core/rtw_wlan_util.c:2006:3:
./include/linux/string.h:340:4: error: call to ‘__read_overflow2’ declared with attribute error: detected read beyond size of object passed as 2nd parameter
    __read_overflow2();
    ^~~~~~~~~~~~~~~~~~
make[2]: *** [scripts/Makefile.build:329: /home/mohammad/rtl8188eu/core/rtw_wlan_util.o] Error 1
make[1]: *** [Makefile:1528: _module_/home/mohammad/rtl8188eu] Error 2
make[1]: Leaving directory '/usr/lib/modules/4.14.31-1-MANJARO/build'
make: *** [Makefile:153: modules] Error 2

```
Another use faced the same error in the rtl8723au driver lwfinger/rtl8723au#66

lwfinger, 2018-04-11:
That problem is fixed along with various changes needed to build with kernels 4.14.0+. Sorry it took a while to get to it.

closed 2018-04-11 by lwfinger

moham96, 2018-04-11:
@lwfinger Thank you very much, it builds now but Network Manager can't connect with it.
What puzzles me is that network manager states in the systemd journal that it recognize the device once I load the module
```
Apr 11 20:37:46 mohammad-pc NetworkManager[24337]: <info>  [1523468266.4590] wifi-nl80211: (wlp0s26u1u1): using nl80211 for WiFi device control
```
but when I try to connect it states that there's no suitable device.

Using wpa_supplicant I can connect successfully only network manager has the problem.

also, I have one more question, is the master branch going to get updated to nl80211?

Thank you very much for your amazing work @lwfinger  :1st_place_medal: