lwfinger/rtl8188eu #254: TL-WN725N V3

HenryYoung42, 2018-06-05:
Does this driver work with V3 of the TP-Link TL-WN725N. I have the driver loaded under Ubuntu 18, I can see the adapter in lsusb, I can seethere's a Wireless interface in lshw, but getting nothing in ifconfig.

lwfinger, 2018-06-06:
Note that this is a multi-branch repo, thus your asking such a question without specifying the branch you use makes it impossible to answer the question. It works for me.

HenryYoung42, 2018-06-06:
I just closed the repo and built without specifying a branch which I guess means I used master. Is there a V3 specific branch I should use ? Thanks for your help.

lwfinger, 2018-06-06:
I use v4.1.8_9499 as the branch. It builds with cfg80211 and nl80211 enabled, and works best with a modern NetworkManager.

HenryYoung42, 2018-06-06:
Checked that out and it builds (one warning). After make install what dkms commends are required to uninstall the old version and add the new build, or did make install do everything needed ? I'm running 100% command like on Ubuntu Server 18.

HenryYoung42, 2018-06-06:
Tried dkms remove, build, install, then another reboot - same outcome. See device in lsusb and lshw but nothing in ifconfig. What's the next step in diagnosis, or did I do something obviously wrong ?

Kogam22, 2018-07-15:
Hi there @HenryYoung42 !
Would you mind sharing the v3 driver if you compiled it already ? Thanks.

lwfinger, 2018-07-15:
Are you aware that a compiled driver will work ONLY if you have exactly the same kernel, and are using the exact same compiler? That is what that Linux requires.

Kogam22, 2018-07-16:
@lwfinger : 

> Are you aware that a compiled driver will work ONLY if you have exactly the same kernel, and are using the exact same compiler? That is what that Linux requires.

Yup.
The problem is, the original drivers (Old linux kernel support) & the beta drivers for TL-WN725N (Latest linux kernel support) show some errors (2~3) in the .c files. That's why i wanted a reference.

@HenryYoung42 Thanks !

lwfinger, 2018-07-16:
"Some errors" is not specific enough for anyone to fix them.

Po-Hsuan-Huang, 2018-09-17:
@lwfinger  I had the same problem, can you confirm the driver on  v4.1.8_9499 branch works for V3 ?

lwfinger, 2018-09-17:
No, I cannot confirm that. If someone wants me to do that, you will need to send me a V3 device for my testing. I am retired, and do this on a volunteer basis. I do not mind doing that, but I am certainly not going to spend to buy an arbitrary device.

What really makes me angry is that people state that they are getting errors in .c files, and then NEVER specify what those errors are.

sseerrggii, 2018-09-18:
Many thanks @lwfinger for spend your time to keep this repo up to date.
When someone like me, with no experience in Linux drivers, buys a TL-WN725N_V3 and starts having a lot of problems with this device, it always helps to find some information about it.

I have warning on >Make all:
```
/home/slimbook/Descargas/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.c: In function ‘rtw_cfg80211_inform_bss’:
/home/slimbook/Descargas/rtl8188eu-4.1.8_9499/os_dep/ioctl_cfg80211.c:502:1: warning: the frame size of 1048 bytes is larger than 1024 bytes [-Wframe-larger-than=]
 }
^
```
After that i did:
```
sudo make install
sudo insmod 8188eu.ko 
```
And the device works pretty well for me.

dcohenb, 2019-03-02:
I can confirm it worked for me without any problems using a `tp-link TL-WN725N V3`
i wrote up on my use-case here:
https://medium.com/@daniel.cohen/setting-up-usb-wifi-dongle-on-pine64-dc41def83c75

thismarcoantonio, 2019-11-01:
After installing, nothing has changed, until I removed it from 3.0 and put it in 2.0 port. Worked in 18.04 following @dcohenb medium post :tada:

prashants, 2019-11-04:
Sorry this does not work in Ubuntu 19.10 kernel version 5.3.0-18-generic.

It compiles and installs correctly but after inserting the module nothing happens. It does not even show up in lshw.

prashants, 2019-11-04:
> I can confirm it worked for me without any problems using a `tp-link TL-WN725N V3`
> i wrote up on my use-case here:
> https://medium.com/@daniel.cohen/setting-up-usb-wifi-dongle-on-pine64-dc41def83c75

/home/user/temp/wifi/rtl8188eu/hal/usb_halinit.c: In function ‘GetHwReg8188EU’:
/home/user/temp/wifi/rtl8188eu/hal/usb_halinit.c:1925:19: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1925 |   *((u16 *)(val)) = haldata->BasicRateSet;
      |   ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
/home/user/temp/wifi/rtl8188eu/hal/usb_halinit.c:1926:2: note: here
 1926 |  case HW_VAR_TXPAUSE:
      |  ^~~~
  CC [M]  /home/user/temp/wifi/rtl8188eu/hal/usb_ops_linux.o
In file included from /home/user/temp/wifi/rtl8188eu/include/drv_types.h:49,
                 from /home/user/temp/wifi/rtl8188eu/hal/usb_ops_linux.c:23:
/home/user/temp/wifi/rtl8188eu/hal/usb_ops_linux.c: In function ‘usb_read_port_complete’:
/home/user/temp/wifi/rtl8188eu/include/rtw_debug.h:93:6: warning: this statement may fall through [-Wimplicit-fallthrough=]
   93 |   if (_level <= GlobalDebugLevel) {   \
      |      ^
/home/user/temp/wifi/rtl8188eu/hal/usb_ops_linux.c:555:4: note: in expansion of macro ‘RT_TRACE’
  555 |    RT_TRACE(_module_hci_ops_os_c_, _drv_err_, ("usb_read_port_complete:bSurpriseRemoved=true\n"));
      |    ^~~~~~~~
/home/user/temp/wifi/rtl8188eu/hal/usb_ops_linux.c:556:3: note: here
  556 |   case -ENOENT:
      |   ^~~~
  CC [M]  /home/user/temp/wifi/rtl8188eu/os_dep/ioctl_linux.o
  CC [M]  /home/user/temp/wifi/rtl8188eu/os_dep/mlme_linux.o
  CC [M]  /home/user/temp/wifi/rtl8188eu/os_dep/os_intfs.o
/home/user/temp/wifi/rtl8188eu/os_dep/os_intfs.c:704:22: error: initialization of ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *)’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *)’} from incompatible pointer type ‘u16 (*)(struct net_device *, struct sk_buff *, struct net_device *, u16 (*)(struct net_device *, struct sk_buff *, struct net_device *))’ {aka ‘short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *, short unsigned int (*)(struct net_device *, struct sk_buff *, struct net_device *))’} [-Werror=incompatible-pointer-types]
  704 |  .ndo_select_queue = rtw_select_queue,
      |                      ^~~~~~~~~~~~~~~~
/home/user/temp/wifi/rtl8188eu/os_dep/os_intfs.c:704:22: note: (near initialization for ‘rtw_netdev_ops.ndo_select_queue’)
cc1: some warnings being treated as errors
make[2]: *** [scripts/Makefile.build:290: /home/user/temp/wifi/rtl8188eu/os_dep/os_intfs.o] Error 1
make[1]: *** [Makefile:1653: _module_/home/user/temp/wifi/rtl8188eu] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.3.0-18-generic'
make: *** [Makefile:151: modules] Error 2

lwfinger, 2019-11-04:
Perhaps your device has a USB ID that is not in the device tables. Please post the appropriate line from the lsusb command.

I am confused. First you say that it builds OK, then you post a build listing with an error. I have enough to do without trying to understand this kind of stuff!

Are you using git to get your source? I just all 3 branches that you mention and all build just fine. v5.2.2.4 had an error, but not the one you mentioned. Just do a 'git pull' and rebuild. If you got a zip file, I have decided that I will no longer support that mode. There are too many  problems with stale source.

prashants, 2019-11-04:
Sorry for the confusion, the error is with respect to the guide here
https://medium.com/@daniel.cohen/setting-up-usb-wifi-dongle-on-pine64-dc41def83c75
when I compile the git checkout 9a56f32

prashants, 2019-11-04:
Yes I am using git to get the source. All the head of the branches master,
v4.1.8_9499 and v5.2.2.4 branch are compiling without any errors but after
inserting the module nothing happens. The dmesg also does not show any
error.

I repeat again that the error was respect to the git checkout 9a56f32.
Extremely sorry for the confusion.

Hope this clears it up.

lwfinger, 2019-11-04:
You still did not tell me what your USB ID is!

prashants, 2019-11-05:
$ sudo lsusb 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 006: ID 0b05:7775 ASUSTek Computer, Inc. Zenfone GO (ZB500KL) (Debug, RNDIS mode)
Bus 001 Device 004: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 003: ID 046d:c31c Logitech, Inc. Keyboard K120
Bus 001 Device 002: ID 093a:2510 Pixart Imaging, Inc. Optical Mouse
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

prashants, 2019-11-05:
Yes got it working just by adding the USB ID and recompiling the module !! Thanks a lot !

```
diff --git a/os_dep/usb_intf.c b/os_dep/usb_intf.c
index a4d0244..b40593c 100644
--- a/os_dep/usb_intf.c
+++ b/os_dep/usb_intf.c
@@ -60,6 +60,7 @@ static struct usb_device_id rtw_usb_id_tbl[] = {
        {USB_DEVICE(0x056E, 0x4008)}, /* Elecom WDC-150SU2M */
        {USB_DEVICE(0x2357, 0x010c)}, /* TP-Link TL-WN722N v2 */
        {USB_DEVICE(0x2357, 0x0111)}, /* TP-Link TL-WN727N v5.21 */
+       {USB_DEVICE(0x0bda, 0x8179)}, /* TP-Link TL-WN727N v3 */
        {}      /* Terminating entry */
 };

```

prashants, 2019-11-05:
Please find the attached device details as per the box.

[image]

lwfinger, 2019-11-05:
That device ID is in my source. If you had done a 'git pull', you could have saved a lot of trouble!!

closed 2019-11-05 by lwfinger