lwfinger/rtl8188eu #257: Problems with hostapd

rafacastalla, 2018-06-27:
He seguido estos pasos y no me funciona, sabeis donde puede estar el error?

https://github.com/lwfinger/rtl8188eu/issues/3

/rtl8188eu-master/hostapd-0.8/hostapd# sudo ./hostapd ../../rtl_hostapd.conf 

Configuration file: ../../rtl_hostapd.conf
drv->ifindex=6
l2_sock_recv==l2_sock_xmit=0x0xabaa40
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Could not connect to kernel driver.
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=2
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
rtl871x_set_key_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set encryption.
Using interface wlx7c8bca0ae363 with hwaddr 7c:8b:ca:0a:e3:63 and ssid 'rtwap'
WPS: Converting display to virtual_display for WPS 2.0 compliance
WPS: Converting push_button to virtual_push_button for WPS 2.0 compliance
rtl871x_set_beacon_ops
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to set beacon head/tail or DTIM period
rtl871x_set_wps_assoc_resp_ie
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Failed to initialize WPS Registrar
wlx7c8bca0ae363: Unable to setup interface.
ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Could not connect to kernel driver.
+rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=2
rmdir[ctrl_interface]: No such file or directory
ELOOP: remaining socket: sock=6 eloop_data=0xabaab0 user_data=(nil) handler=0x4381a5

lwfinger, 2018-06-27:
The error is
"ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Could not connect to kernel driver"

What kernel driver is hostapd looking for? I think you are using the master branch of this driver, which does not implement nl80211 and cfg80211. Instead, you should use the v4.1.8_9499 branch.

marcobazzani, 2018-10-24:
same problem here 
some issues with newer kernel 
was working on jessie raspbian not working anymore on stretch 4.14.71+

lwfinger, 2018-10-24:
Could you at least be a bit more specific? "Not working" is hardly definitive.

marcobazzani, 2018-10-25:
ok I tried both with master and v5.2.2.4 branch

config file:
```
interface=wlan0
ctrl_interface=/var/run/hostapd
ssid=rtwap
channel=6
wpa=2
wpa_passphrase=87654321
eap_server=1
wps_state=1
uuid=12345678-9abc-def0-1234-56789abcdef0
device_name=RTL8188EU
manufacturer=Realtek
model_name=RTW_SOFTAP
model_number=WLAN_CU
serial_number=12345
device_type=6-0050F204-1
os_version=01020300
config_methods=label display push_button keypad
driver=rtl871xdrv
beacon_int=100
hw_mode=g
ieee80211n=1
wme_enabled=1
ht_capab=[SHORT-GI-20][SHORT-GI-40]
wpa_key_mgmt=WPA-PSK
wpa_pairwise=CCMP
max_num_sta=8
wpa_group_rekey=86400
```
this is the output:

```
/usr/local/bin/hostapd -dd hostapd.conf.lwfinger
random: Trying to read entropy from /dev/random
Configuration file: hostapd.conf.lwfinger
drv->ifindex=4
l2_sock_recv==l2_sock_xmit=0x0x1a86a40
BSS count 1, BSSID mask 00:00:00:00:00:00 (0 bits)
Allowed channel: mode=1 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=12 freq=2467 MHz max_tx_power=0 dBm
Allowed channel: mode=1 chan=13 freq=2472 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=1 freq=2412 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=2 freq=2417 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=3 freq=2422 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=4 freq=2427 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=5 freq=2432 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=6 freq=2437 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=7 freq=2442 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=8 freq=2447 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=9 freq=2452 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=10 freq=2457 MHz max_tx_power=0 dBm
Allowed channel: mode=0 chan=11 freq=2462 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=36 freq=5180 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=40 freq=5200 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=44 freq=5220 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=48 freq=5240 MHz max_tx_power=0 dBm
Allowed channel: mode=2 chan=149 freq=5745 MHz max_tx_power=0 dBm
[... 97 lines cut ...]
+++ b/Makefile
@@ -91,8 +91,8 @@ CONFIG_RTW_SDIO_PM_KEEP_POWER = y
 ###################### MP HW TX MODE FOR VHT #######################
 CONFIG_MP_VHT_HW_TX_MODE = n
 ###################### Platform Related #######################
-CONFIG_PLATFORM_I386_PC = y
-CONFIG_PLATFORM_BCM2709 = n
+CONFIG_PLATFORM_I386_PC = n
+CONFIG_PLATFORM_BCM2709 = y
 CONFIG_PLATFORM_ANDROID_X86 = n
 CONFIG_PLATFORM_ANDROID_X86_N70 = n
 CONFIG_PLATFORM_ANDROID_INTEL_X86 = n
diff --git a/rtl_hostapd.conf b/rtl_hostapd.conf
index 172dda7..1fa631e 100644
--- a/rtl_hostapd.conf
+++ b/rtl_hostapd.conf
@@ -1,6 +1,6 @@
 ##### hostapd configuration file ##############################################

-interface=wlp0s2f1u2
+interface=wlan0
 ctrl_interface=/var/run/hostapd
 ssid=rtwap
 channel=6
```

marcobazzani, 2018-10-25:
Ok I'll give it a try.
Please count on me if you need some more tests.

marcobazzani, 2018-10-26:
It works!! awesome 
Thanks dude 👍

Ablinne, 2018-12-06:
I am still having problems with this, so maybe you can help me get things clarified. I am a bit confused about which version of the driver is supposed to work with which hostapd-driver (nl80211 vs rtl871xdrv). I am working on Arch linux, using a TP-Link TL-WN725N (v2?) wifi dongle with RTL8188EUS chip, using default Arch linux kernel, currently v4.19.4.

My understanding so far is:
* master branch supported the rtl871xdrv at some point, but does not seem to any more, as I always get `ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported` when trying to use rtl871xdrv with any of the driver versions (master, v4.1.8_9498, v5.2.2.4, staging version present in the kernel). (using the hostapd patch available at https://github.com/pritambaral/hostapd-rtl871xdrv)

* the v4.1.8_9498 and v5.2.2.4 branches should support the nl80211 driver, however I have gotten none of them to work with hostapd.

I have tried hostapd 2.7 from the regular arch linux packages as well as the current hostapd git master version for nl80211. Which of these two hostapd's I used had no effect on my results.

Using the v4.1.8_9498 module the access point sets up fine and I can connect (using an android phone), but the connection is "lost" after a short period of time. The network is bridged to my ethernet card and it obtains an IP address from the dhcp server on the network behind the bridge. After a few seconds packages just stop getting through. Here is my config and log for this case: https://pastebin.com/0dvvnsz8

On v5.2.2.4, the client sees the AP, but can't even connect. The phone tries three times, then gives up. Config and log for this case: https://pastebin.com/ZRyadbsR

Any idea? I have seen #16 , but my dongle works completely fine when used to connect to the buildings wifi...

lwfinger, 2018-12-06:
The critical configuration parameter is CONFIG_IOCTL_CFG80211. If that is defined, then hostapd should use the nl80211 interface. If not, then the rtl871xdrv. As most people want to use cfg80211 to configure their interfaces, the repos are generally set with CONFIG_IOCTL_CFG80211 defined. If you wish, you can turn it off in Makefile.

Repeating what I have said before. I know nothing of the internals of the chips. My only involvement is to make certain the drivers will compile when the kernel APIs change, and to fix any kernel warnings or crashes. If the device fails to work without a diagnostic, then there is little I can do. In general, the later the version, the more likely the device will work.

Ablinne, 2018-12-07:
Me again, with good news:

I managed to reinstate the legacy wireless ioctls by patching the kernel, see the patch here: https://pastebin.com/2nmLS4d6. This reverts the kernel commits 8bfb3676606454ffec836f56c5dc3e69dfc0956a and (with adaptations) 4f39a1f5870104b1670df2c09c831ac281896545.

After that, hostadp with the patch from https://github.com/pritambaral/hostapd-rtl871xdrv successfully uses the rtl871xdrv. Then I tried the three different branches of this driver:

* v5.2.2.4: hostapd starts, clients can not successfully associate
* v4.1.8_9498: hostapd starts, but bridging the interfaces does not work. WPA2 does not work.
* master: everything works (including wpa2), but when using bridging, my computer freezes after a few megabytes of transferred data. However using ip-forwarding (NAT) I got my AP to work like a charm.

Just to let everyone know how i did it, in case they also have the same chip...

dexinnet, 2019-11-19:
Can confirm this repository's driver branch v4.1.8-9499 works for TL-WN722N v2 2357:010c using unmodified hostapd 2.9 with driver nl80211 on Raspbian Buster, Kernel 4.19.75+. Make sure to select the correct branch when cloning.

closed 2019-11-20 by rafacastalla

hamishmb, 2020-02-12:
Hi, I was wondering if you'd had any luck getting this to work with Raspbian Buster (10)? I've partially completed a tutorial (still have a kernel panic problem) on this using resources fro various places, including here, and would like there to be an up to date resource on this some. My tutorial is at <a href="https://www.hamishmb.com/blog/how-to-make-a-raspi-ap-with-a-rtl8xxx-wifi-adaptor/">https://www.hamishmb.com/blog/how-to-make-a-raspi-ap-with-a-rtl8xxx-wifi-adaptor/</a> if you're interested.

Does anyone else still get a kernel panic even with the patch? It said it succeeded, but maybe I applied it wrong?

lwfinger, 2020-02-12:
Your tutorial discusses the RTL8192CU, but you post on an RTL8188EU issue.

hamishmb, 2020-02-12:
Yeah, I've had issues with both the RTL8192CU and 8188CUS, but they all use the same driver and both have the same crash, so I assume any advice for the 8188EU will probably be good (or at least informative) for the 8192CU/8188CUS too :)

lwfinger, 2020-02-12:
Porsche and Ford both make automobiles. Would a single instruction manual work for both?

The RTL8188EU is totally different from the RTL8192CU and RTL8188CU!!!!

You did not state what branch of this repo you are using. For the 8188eu, v5.2.2.4 is much better.

dexinnet, 2020-02-12:
Regarding **RTL8192EU**: Can confirm repository [https://github.com/Mange/rtl8192eu-linux-driver](url) branch realtek-4.4.x works for TL-WN822N v4, 2357:0108 using unmodified hostapd 2.9 with driver nl80211, WPA2 and NAT on Raspbian Buster, Kernel 4.19.97+. Make sure to select the correct branch when cloning.

muelli01, 2020-12-24:
The 5.2.2.4 driver from this repo is working with an unmodified hostapd 2.6 and the RTL8188EUS adapter for me:

Bus 001 Device 006: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter

Thanks!

whitslack, 2022-07-15:
I concur with @muelli01: the driver from the v5.2.2.4 branch of this repository works fine with an unmodified hostapd 2.10 and an RTL8188EUS adapter on a Raspberry Pi 1.

```
# uname -a
Linux raspberrypi 5.15.53+ #1572 Wed Jul 13 17:11:29 BST 2022 armv6l ARMv6-compatible processor rev 7 (v6l) BCM2835 GNU/Linux
```
```
# lsusb -d 0bda:
Bus 001 Device 004: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
```
```
# modinfo 8188eu | head -n6
filename:       /lib/modules/5.15.53+/kernel/drivers/net/wireless/8188eu.ko
version:        v5.2.2.4_25483.20171222
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     09579C867E0EE8769DED732
```
```
# hostapd -v 2>&1 | head -n1
hostapd v2.10
```
```
# iw wlan0 info
Interface wlan0
	ifindex 3
	wdev 0x1
	addr 00:7f:05:##:##:##
	ssid ##########
	type AP
	wiphy 0
	txpower 12.00 dBm
```

Unfortunately, the transmit power is too low for my application, and `iw wlan0 set txpower fixed 1800` apparently succeeds but has no effect.

lwfinger, 2022-07-15:
Yes, the routine called for set txpower just returns a zero to indicate success, and the get routine returns a constant 12.

I am trying to find where the power variable is stored.

lwfinger, 2022-07-16:
I just pushed some code that sets and gets the changes using iw wlan0 set txpower fixed 1800 and iw dev.

I do not know whether that will actually change the transmitted power. I also had to guess at the maximum value of 1800 mbm.

whitslack, 2022-07-16:
@lwfinger: Thanks for your efforts, but I think this isn't going to help me. After I reloaded the new module, now `iw wlan0 info` tells me the initial txpower is 42.00 dBm, which is physically impossible, as the adapter would be bursting into flames. I believe the power setting in these chips is an index into an internal table, not a dBm value, and you can't assume that power index 42 translates to 42 dBm.

lwfinger, 2022-07-17:
Mine said that the tx power was 42 dBm before I changed anything, but another time it said it was  0 dBm.

When you used iw to change the power, did it do anything?

whitslack, 2022-07-17:
@lwfinger: I didn't change the txpower because I knew it'd be silently capped at 18, whatever that means, and I didn't want to take the chance that I wouldn't be able to get it back to 42 without power cycling it. Just guessing that 18 is less power than 42.

jonathan84clark, 2023-01-24:
Hello Everyone, 
What an interesting and relevant conversation! I was wondering if I could get some help. 

Recently the company I work for had to upgrade all of their devices from Jessie to Buster. When the devices were on Jessie I was able to get the RTL8188EUS to work both as a WiFi client and as a Wifi access point.
On Buster I can get the RTL8188EUS to work as a Wifi client but NOT as a Wifi access point. When configured as an access point the hostapd service reports an error and I see the following in the syslog:
`Jan 24 00:03:35 ssgmvc hostapd[2676]: wlan0: Could not connect to kernel driver
Jan 24 00:03:35 ssgmvc hostapd[2676]: +rtl871x_sta_deauth_ops, ff:ff:ff:ff:ff:ff is deauth, reason=3
Jan 24 00:03:35 ssgmvc hostapd[2676]: rtl871x_set_key_ops
Jan 24 00:03:35 ssgmvc hostapd[2676]: ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Jan 24 00:03:35 ssgmvc hostapd[2676]: Failed to set encryption.
Jan 24 00:03:35 ssgmvc hostapd[2676]: rtl871x_set_key_ops
Jan 24 00:03:35 ssgmvc hostapd[2676]: ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Jan 24 00:03:35 ssgmvc hostapd[2676]: Failed to set encryption.
Jan 24 00:03:35 ssgmvc hostapd[2676]: rtl871x_set_key_ops
Jan 24 00:03:35 ssgmvc hostapd[2676]: ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Jan 24 00:03:35 ssgmvc hostapd[2676]: Failed to set encryption.
Jan 24 00:03:35 ssgmvc hostapd[2676]: rtl871x_set_key_ops
Jan 24 00:03:35 ssgmvc hostapd[2676]: ioctl[RTL_IOCTL_HOSTAPD]: Operation not supported
Jan 24 00:03:35 ssgmvc hostapd[2676]: Failed to set encryption.
Jan 24 00:03:35 ssgmvc hostapd[2676]: wlan0: AP-DISABLED
Jan 24 00:03:35 ssgmvc hostapd[2676]: hostapd_free_hapd_data: Interface wlan0 wasn't started
Jan 24 00:03:35 ssgmvc hostapd[2676]: ELOOP: remaining socket: sock=6 eloop_data=0xa2b600 user_data=(nil) handler=0x41cd0
Jan 24 00:03:35 ssgmvc systemd[1]: hostapd.service: Control process exited, code=exited, status=1/FAILURE
Jan 24 00:03:35 ssgmvc systemd[1]: hostapd.service: Failed with result 'exit-code'.
Jan 24 00:03:35 ssgmvc systemd[1]: Failed to start Advanced IEEE 802.11 AP and IEEE 802.1X/WPA/WPA2/EAP Authenticator.
`
Here is my output from iwconfig:
`pi@ssgmvc:~/BuildDir/rtl8188eu $ iwconfig
lo        no wireless extensions.

eth0      no wireless extensions.

eth1      no wireless extensions.

wlan0     unassociated  ESSID:""  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

wwan0     no wireless extensions.
`

I have made many attempts to resolve the issue. Here is what I tried so far:
1. Building and installing this driver on the system (branch v5.2.2.4)
2. Upgrading to hostapd 2.9
3. Installing the version of hostapd that is included in this driver
4. Using hostapd 2.4
5. Using hostapd 2.3

I was wondering if there is something I'm missing. I'm not super familiar with hostapd interacts with this driver, but it looks like something related to access points is broken.

Any help on this would be great. I'm also fine creating a new defect report if you need me to.

lwfinger, 2023-01-24:
I have no idea what Jessie and Buster mean, and have no interest in finding out. What kernel versions are involved?

Your messages all seem to be problems with setting/using encryption keys. None of that changed in the driver. Program hostapd uses the driver to transmit and receive messages - nothing else.

jonathan84clark, 2023-01-24:
The kernel version is 5.10.103-v7+
The reason I posted here is because the first comment in this thread has the same errors that I am seeing in my log file. Isn't the command 'ioctl' referring to an I/O control message? I thought that I/O controls would be something the driver would use to interact with a piece of hardware. I don't know much about Linux drivers so I could be wrong, that's what I thought though. 
If this is an encryption issue, do you have any recommendations on how I could work toward solving it?

lwfinger, 2023-01-24:
Please post your hostapd.conf.

jonathan84clark, 2023-01-24:
Thanks for the help. Here is my hostapd file:
`interface=wlan0
ssid=TestSSID
channel=7
hw_mode=g
wmm_enabled=0
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=TestPassword
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP`

lwfinger, 2023-01-24:
You are not specifying a driver for hostapd. You need to have a line that says
driver=nl80211

The update must have discarded the old-style IOCTL commands.

jonathan84clark, 2023-01-24:
Thanks for the help. I am no longer seeing the error that I was seeing. Now I'm seeing some new errors:

nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
hostapd_free_hapd_data: Interface wlan0 wasn't started

I followed this guide:
https://wireless.wiki.kernel.org/en/users/Documentation/hostapd because the nl80211 driver wasn't installed on my system. This installed the driver but the driver with hostapd but it still isn't happy for some reason.

lwfinger, 2023-01-24:
My hostapd.conf is attached:
[hostapd.conf.txt](https://github.com/lwfinger/rtl8188eu/files/10493021/hostapd.conf.txt)
I had to add the .txt for this website to see it.

jonathan84clark, 2023-01-24:
Thanks, I'll give it a try!

jonathan84clark, 2023-01-24:
I'm getting the same errors using your config file:
pi@ssgmvc:~ $ sudo hostapd /etc/hostapd/hostapd.conf
nl80211: Driver does not support authentication/association or connect commands
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED
wlan0: CTRL-EVENT-TERMINATING
hostapd_free_hapd_data: Interface wlan0 wasn't started

I'm going to see if I can reproduce this on another system with a fresh image.

jonathan84clark, 2023-01-25:
I was able to resolve the problem. Thanks for the help!

lwfinger, 2023-01-25:
What was the solution?

jonathan84clark, 2023-01-25:
It was super weird I'm still trying to understand why it works. I had to do the following.
1. Build and install the rtl8188eu driver from the master branch of https://github.com/lwfinger/rtl8188eu.git
2. Checkout the v5.2.2.4 branch
3. Build and install the rt8188eu driver from the v5.2.2.4 branch of https://github.com/lwfinger/rtl8188eu.git
4. Run the driver with hostapd version 2.3

I'm guessing that something in the makefile for the master branch is messing up the rtl8188eu driver because when I just install it from there my TP-Link (0bda:8179) adapter doesn't even show up, but it must be installing some needed dependencies.
Then when I install from the v5.2.2.4 branch it fixes the driver. 
Anyway, now I need to figure out how to get an update out to all of the customers, that resolves this issue.