lwfinger/rtl8188eu #259: TP-LINK TL-WN725N v2 driver support?

dipate, 2018-08-24:
I am following the link : [](https://wikidevi.com/wiki/TP-LINK_TL-WN725N_v1) to identify which version my device has and it looks like I have TL-WN725N v2.

Is the master branch of the repo suffice my version? I want to use it for the Ubuntu 16.04 x86_64 with the kernel version "4.4.0-122-generic" and want to use it as an AP on the machine. Is it supported with this driver?

Appreciate if anyone can reply.

Thanks,
Digesh

Ablinne, 2018-12-11:
Hi @dipate, according to https://wikidevi.com/wiki/TP-LINK_TL-WN725N_v2 I have the same device. I got it to work as an AP. See my comments on #257 about how I did it. With a 4.4 kernel you won't even have to build your own kernel.

joneslee85, 2020-02-02:
Yes it does. 

I am running Linux 5.4.15 and can confirm it works:

```
$ lsusb
Bus 001 Device 003: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter

$ sudo dnf groupinstall "Development Tools" "Development Libraries"
$ cd /tmp
$ sudo dnf install kernel-devel
$ git clone https://github.com/lwfinger/rtl8188eu.git

$ cd rtl8188eu
$ make
$ sudo make install
$ sudo depmod -a
$ sudo dracut -c -f -force
$ sudo modprobe 8188eu
$ modinfo 8188eu

filename:       /lib/modules/5.4.15-200.fc31.x86_64/kernel/drivers/net/wireless/8188eu.ko
version:        v4.1.4_6773.20130222
author:         Realtek Semiconductor Corp.
description:    Realtek Wireless Lan Driver
license:        GPL
srcversion:     762D07FBEAF35B1366430A8
alias:          usb:v2357p0111d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2357p010Cd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v056Ep4008d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p331Bd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3311d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p3310d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v2001p330Fd*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0DF6p0076d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v07B8p8179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp0179d*dc*dsc*dp*ic*isc*ip*in*
alias:          usb:v0BDAp8179d*dc*dsc*dp*ic*isc*ip*in*
depends:
name:           8188eu
vermagic:       5.4.15-200.fc31.x86_64 SMP mod_unload mprofile-kernel relocatable
parm:           rtw_ips_mode:The default IPS mode (int)
parm:           ifname:The default name to allocate for first interface (charp)
parm:           if2name:The default name to allocate for second interface (charp)
parm:           rtw_initmac:charp
parm:           rtw_channel_plan:int
parm:           rtw_chip_version:int
parm:           rtw_rfintfs:int
parm:           rtw_lbkmode:int
parm:           rtw_network_mode:int
parm:           rtw_channel:int
parm:           rtw_mp_mode:int
parm:           rtw_wmm_enable:int
parm:           rtw_vrtl_carrier_sense:int
parm:           rtw_vcs_type:int
parm:           rtw_busy_thresh:int
parm:           rtw_led_enable:int
parm:           rtw_ht_enable:int
parm:           rtw_cbw40_enable:int
parm:           rtw_ampdu_enable:int
parm:           rtw_rx_stbc:int
parm:           rtw_ampdu_amsdu:int
parm:           rtw_lowrate_two_xmit:int
parm:           rtw_rf_config:int
parm:           rtw_power_mgnt:int
parm:           rtw_smart_ps:int
parm:           rtw_low_power:int
parm:           rtw_wifi_spec:int
parm:           rtw_antdiv_cfg:int
parm:           rtw_antdiv_type:int
parm:           rtw_enusbss:int
parm:           rtw_hwpdn_mode:int
parm:           rtw_hwpwrp_detect:int
parm:           rtw_hw_wps_pbc:int
parm:           rtw_max_roaming_times:The max roaming times to try (uint)
parm:           rtw_fw_iol:FW IOL (int)
parm:           rtw_mc2u_disable:int
parm:           rtw_80211d:Enable 802.11d mechanism (int)
parm:           rtw_notch_filter:0:Disable, 1:Enable, 2:Enable only for P2P (uint)
parm:           debug:Set debug level (1-9) (default 1) (int)

$ nmcli device
DEVICE      TYPE      STATE         CONNECTION
wlP3p1s0u2  wifi      disconnected  --
```

ahillelt, 2021-02-16:
I have been having issues getting mine to work, I have a RockPi 4A with Ubuntu 18.04 on it, kernel 4.4.154-110. The RockPi 4A has an RK3399 chip and no native wifi, just ethernet. My TP-Link is brand new and I believe Tp-Link is calling it 'v3' according to their site: https://www.tp-link.com/us/support/download/tl-wn725n/#Driver

The drives from TP-LINK don't work for me either. I am curious for any insights here, as always appreciate any and all help

I have tried both https://github.com/lwfinger/rtl8188eu and the modified version from https://github.com/quickreflex/rtl8188eus

When I try to make the file I get the following error:

```
make[1]: Entering directory '/usr/src/linux-headers-4.4.154-110-rockchip-gcef30e88a9f5'
Makefile:645: arch/aarch64/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/aarch64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.4.154-110-rockchip-gcef30e88a9f5'
Makefile:1911: recipe for target 'modules' failed
make: *** [modules] Error 2
```

when trying with DKMS I get

```
DKMS make.log for 8188eu-1.0 for kernel 4.4.154-110-rockchip-gcef30e88a9f5 (aarch64)
Tue Feb 16 07:23:48 UTC 2021
make ARCH=aarch64 CROSS_COMPILE= -C /lib/modules/4.4.154-110-rockchip-gcef30e88a9f5/build M=/var/lib/dkms/8188eu/1.0/b$
make[1]: Entering directory '/usr/src/linux-headers-4.4.154-110-rockchip-gcef30e88a9f5'
Makefile:645: arch/aarch64/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/aarch64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.4.154-110-rockchip-gcef30e88a9f5'
Makefile:1911: recipe for target 'modules' failed
make: *** [modules] Error 2
```

AmburgD, 2025-02-08:
Hey everyone, I know this is an old thread but I need some help. I have this exact model of Wifi Adapter and I am trying to follow the steps to install the driver on openWrt 24.10.0 for Raspberry Pi. When I run make it tells me that the build doesn't exist, which I am assuming means that I have not installed the kernel headers. I do not know where to get or put these files in order to build this driver correctly and if anyone would be able to help I would be extremely grateful.

runlevel5, 2025-02-10:
> Hey everyone, I know this is an old thread but I need some help. I have this exact model of Wifi Adapter and I am trying to follow the steps to install the driver on openWrt 24.10.0 for Raspberry [...]

if you want people to aid you, please provide much more details, for example distribution, kernel version, compiler version and full steps by steps (plus outputs)

AmburgD, 2025-02-10:
Apologies, I am running the latest version of openWRT 24.10.0 Kernel Version 6.6. As for compiler I am just using the latest version of Make that comes with opkg on openWRT. I'm unsure of how to use dkms since it can't be installed with opkg.

runlevel5, 2025-02-10:
Please provide the full output of:

```
make all
```