lwfinger/rtl8188eu #263: Problem on 18.04

hectormartin42, 2018-10-05:
sudo modprobe rtl8188eu
modprobe: FATAL: Module rtl8188eu not found in directory /lib/modules/4.15.0-34-generic

lwfinger, 2018-10-05:
There is no driver named rtl8188eu in Linux. What are you trying to do?

moash, 2018-10-12:
Lwfinger, Please contact me so we can work on configuring the driver for 18.04 TP link released a package to make the driver

lwfinger, 2018-10-12:
My regular E-mail address is Larry.Finger@lwfinger.net. PLease contact me there, and be mujch clearer on what you want to do.

michael-upton97, 2019-02-03:
Hey LwFinger,

Its a shame this went offline as i have the same problem. to be clear on whats happening, there are many online tutorials on installing the correct drivers for various tp-link devices, most of them contain the repository. here is an example: 
`https://askubuntu.com/questions/678134/how-to-install-tp-link-wn725n-wifi-usb-adapter-on-ubuntu-ubuntu-14-04-3-lts` 
there is a step after the make process that requires the mod probe that me and hectormartin42 have been struggling with. is this step necessary and if so how to we get around it?

lwfinger, 2019-02-03:
The standard kernel has two drivers called r8188eu and rtl8xxxu that will conflict with the driver named 8188eu from my repo. You need to blacklist the standard ones. As root, create a file named /etc/modprobe.d/50-rtl8188eu.conf and add two lines

blacklist r8188eu
blacklist rtl8xxxu

Once you have done "make" and "sudo make install" and then rebooted, then 
lsmod | egrep "8188|8xxxu"
should only show 8188eu and the device should work.

infinitdimension, 2019-02-05:
> > The standard kernel has two drivers called r8188eu and rtl8xxxu that will conflict with the driver named 8188eu from my repo. You need to blacklist the standard ones. As root, create a file named /etc/modprobe.d/50-rtl8188eu.conf and add two lines
> > blacklist r8188eu
> > blacklist rtl8xxxu
> > Once you have done "make" and "sudo make install" and then rebooted, then
> > lsmod | egrep "8188|8xxxu"
> > should only show 8188eu and the device should work.
> 
> Hi @lwfinger all right?
> 
> I have been trying make a device that came with your work available in a CD and I have been receiving help (but no luck so far): [Mega](https://forums.linuxmint.com/viewtopic.php?f=53&t=287314)
> 
> The driver files are available [here](https://mega.nz/#!fzJlSIiR!LKfvAcCtyu_EcXI3cXrNTByIDAjeU1_swo5jV3PN-qo) (so you can check if you want and have the time haha)
> 
> I believe the dongle came from China. Although I bought it in a local store. On Windows, I tried to gather infor about the device and no maker is found to be shown for example. So it's really generic...
> 
> But on Cinnamon, I can't even get it to be recognized on the USB port level.
> 
> I tried your solution, but because of the following problems (when lsmod | egrep "8188|8xxxu" there's no details. The install fails, after all...):
> 
> `$ sudo bash install.sh ################################################## Realtek Wi-Fi driver Auto installation script Novembor, 21 2011 v1.1.0 ################################################## Decompress the driver source tar ball: rtl8188EUS_linux_v4.1.4_6773.20130222.tar.gz rtl8188EUS_linux_v4.1.4_6773.20130222/ rtl8188EUS_linux_v4.1.4_6773.20130222/runwpa rtl8188EUS_linux_v4.1.4_6773.20130222/core/ rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_xmit.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_ioctl_query.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_bt_mp.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/efuse/ rtl8188EUS_linux_v4.1.4_6773.20130222/core/efuse/rtw_efuse.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_recv.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_br_ext.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_wapi.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_eeprom.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_debug.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_tdls.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_p2p.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_ieee80211.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_security.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_cmd.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_mlme.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_mp.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_wapi_sms4.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_sreset.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_sta_mgt.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_rf.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_pwrctrl.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_wlan_util.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_mlme_ext.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_io.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_ap.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_led.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_ioctl_rtl.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_mp_ioctl.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_ioctl_set.c rtl8188EUS_linux_v4.1.4_6773.20130222/core/rtw_iol.c rtl8188EUS_linux_v4.1.4_6773.20130222/wlan0dhcp rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/ rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/osdep_service.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/ rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/ioctl_linux.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/recv_linux.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/os_intfs.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/pci_ops_linux.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/custom_gpio_linux.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/usb_intf.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/mlme_linux.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/gspi_intf.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/pci_intf.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/sdio_intf.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/rtw_android.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/xmit_linux.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/gspi_ops_linux.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/usb_ops_linux.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/sdio_ops_linux.c rtl8188EUS_linux_v4.1.4_6773.20130222/os_dep/linux/ioctl_cfg80211.c rtl8188EUS_linux_v4.1.4_6773.20130222/include/ rtl8188EUS_linux_v4.1.4_6773.20130222/include/hal_com.h rtl8188EUS_linux_v4.1.4_6773.20130222/include/wlan_bssdef.h rtl8188EUS_linux_v4.1.4_6773.20130222/include/cmd_osdep.h rtl8188EUS_linux_v4.1.4_6773.20130222/include/rtw_recv.h rtl8188EUS_linux_v4.1.4_6773.20130222/include/rtw_mlme_ext.h rtl8188EUS_linux_v4.1.4_6773.20130222/include/wifi.h rtl8188EUS_linux_v4.1.4_6773.20130222/include/rtl8192c_led.h rtl8188EUS_linux_v4.1.4_6773.20130222/include/rtl8192d_recv.h rtl8188EUS_linux_v4.1.4_6773.20130222/include/custom_gpio.h rtl8188EUS_linux_v4.1.4_6773.20130222/include/HalPwrSeqCmd.h rtl8188EUS_linux_v4.1.4_6773.20130222/include/rtl8723a_dm.h rtl8188EUS_linux_v4.1.4_6773.20130222/include/rtl8188e_dm.h rtl8188EUS_linux_v4.1.4_6773.20130222/include/Hal8192CP [... cut ...] r 2 make[1]: Leaving directory '/usr/src/linux-headers-4.15.0-20-generic' Makefile:678: recipe for target 'modules' failed make: *** [modules] Error 2 ################################################## Compile make driver error: 2 Please check error Mesg ##################################################`
> 
> If you can help, I'll certainly be grateful!!! :D
> 
> Thanks :D

Hi again! All solved as per the Linux Mint Forum! User JeremyB alongside with others helped a lot!!! :D 

Thanks anyhow :D

facug91, 2020-08-11:
Hi, I'm trying to use a usb wifi, with this driver.  
I've tried the installation method, but didn't work. I also tried this blacklisting you said, and when I run `lsmod | egrep "8188|8xxxu"` it shows nothing.

lwfinger, 2020-08-11:
I need the whole 'lsusb' output, at least for the Realtek line. Those numbers are important.

facug91, 2020-08-11:
Bus 001 Device 009: ID 0bda:f179 Realtek Semiconductor Corp.

lwfinger, 2020-08-11:
You do have an RTL8188EU device, but it will not be recognized by either rtl8xxxu or r8188eu.

Do a 'make' and 'sudo make install' followed by a reboot. At that point, 'lsmod | grep 8188' should show a driver.

facug91, 2020-08-11:
Just in case, I did those 3 steps again, but I have the same result, no output for `lsmod | grep 8188`

lwfinger, 2020-08-11:
My mistake. That device was in one of my drivers, but not in the ones in this repo. Do a 'git pull', then repeat those 3 steps.

facug91, 2020-08-12:
Thank you @lwfinger , now when I execute `lsmod | grep 8188`, it outputs `8188eu                729088  0`, and my computer recognize the wifi usb.  
The problem now is that there's no network detected, I don't know if the dongle is not working, or some problem with my configuration/driver. Do you have any idea?

facug91, 2020-08-12:
A friend of mine has the same dongle and has the same problem, so it is not something particular to mine.

lwfinger, 2020-08-12:
Is there anything about this in the output of the dmesg command?

Your dongle is an RTL8188FU, not RTL8188EU. The master branch probably does not handle that one very well. Do the following:

git checkout origin/v5.2.2.5 -b v5.2.2.5
make
sudo make install
reboot

facug91, 2020-08-13:
I have the CD of the dongle, and the driver inside says "RTL8188EUS_linux_v4.1.4_6773.20130222". The problem is that when I try to compile it, it throws an error.  
I've tried the branch v5.2.2.4, but now the Wi-Fi doesn't detect any adapter connected. Anyway, the output of `lsmod | grep 8188` changed:  
```
8188eu               1495040  0
cfg80211              712704  1 8188eu
```

facug91, 2020-08-13:
As you said, I tried one I found for RTL8188FU (https://github.com/kelebek333/rtl8188fu), and it worked. Thank you so much lwfinger!