lwfinger/rtl8188eu #264: testing v5.2.2.4 on a Raspberry pi - LED not working

fgimenezm, 2018-10-26:
Hello.
First of all thanks for making this available. Awesome job!

About the LED, I compiled the driver "as-is", just git clone, make all and sudo make install. It worked without issues (iwconfig reports a lower signal level than other drivers but it still works ok).

I didn't change the Makefile at all so it's using the CONFIG_PLATFORM_I386_PC=y which apparently doesn't break anything. Not sure if I should change that for an RPI.

Anyway, the reason I'm opening this issue is to ask if there is anything else I need to do to enable the LED with the v5.2.2.4 driver.

Thanks.

joneslee85, 2020-02-02:
@fgimenezm could you please be more specific? What is LED? What hardware are you using? Any details on the expected behaviour would be needed if you want people to be able help you

fgimenezm, 2020-02-03:
Sure, It's a TP-Link WN725N (device ID 0bda:8179) connected to a Raspberry Pi.

With the kernel driver (staging: r8188eu v4.1.4_6773.20130222) this adapter has a green LED that blinks when there is traffic.

I think the older 4.x branch from this repo had it too but I'm not sure. I don't have that version compiled right now to test. I'll compile it and update this post in a few hours with the test result.

But with the v5.2.2.4 branch the LED remains always off.

paperoga-dev, 2020-03-17:
Hi, I confirm what @fgimenezm reports. With 5.x version, the LED on the TP-Link dongle is always off.

lwfinger, 2020-03-19:
My device has no LED, thus I cannot debug this issue. When I check the code, everything looks OK, If anyone finds a fix, I will merge it.

MrEngman, 2020-03-19:
You probably need to uncomment line 151 in file include/autoconf.h

/* #define CONFIG_LED */    <<< uncomment this line
#ifdef CONFIG_LED
	#define CONFIG_SW_LED
	#ifdef CONFIG_SW_LED
		/* #define CONFIG_LED_HANDLED_BY_CMD_THREAD */
	#endif
#endif /* CONFIG_LED */

MrEngman

paperoga-dev, 2020-03-22:
I confirm that the @MrEngman  proposal fixes the issue. Uncommenting line 151 restores the LED blink.

Thanks!

fgimenezm, 2020-03-22:
Indeed it works. Un-commented that line and now it blinks properly.

lwfinger, 2020-03-22:
I should have caught line, but I completely forgot about autoconf.h. I am glad it is working.

closed 2020-03-22 by lwfinger