lwfinger/rtl8188eu #266: Kernel warning with 4.14.75

Menion2k, 2018-11-07:
Trying to use 0bda:8179 dongle from TpLink in Openwrt running in Virtualbox with guest addosn, I get kernel warnings and device is not attached. Same happens with another fork of the driver
The driver is built on v5.2.2.4 branch
Current WEXT staging driver works on the same setup (of course it is currenlty removed)

```
[   87.596336] RTW: rtl8188eu v5.2.2.4_25483.20171222
[   87.597085] RTW: build time: Nov  6 2018 12:13:11
[   87.659036] RTW: hal_com_config_channel_plan chplan:0x20
[   87.659860] RTW: rtw_regsty_chk_target_tx_power_valid return _FALSE for band:0, path:0, rs:0, t:-1
[   87.660956] ------------[ cut here ]------------
[   87.661544] WARNING: CPU: 0 PID: 1923 at wiphy_register+0x25e/0x850 [cfg80211]
[   87.662601] Modules linked in: 8188eu(+) ath9k_htc ath9k_common rtl8821ae rtl8192se rtl8192de rtl8192cu rtl8192ce rtl8192c_common rtl_usb rtl_pci rt2800usb rt2800pci rt2800mmio rt2800lib rt2500usb rt2500pci rt2400pci pppoe ppp_async btcoexist ath9k_hw ath rtlwifi rtl8xxxu rtl818x_pci rtl8187 rt2x00usb rt2x00pci rt2x00mmio rt2x00lib rndis_host pppox ppp_generic nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 cdc_ether xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD usbnet slhc r8712u r8169 nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack
[   87.669478]  lib80211_crypt_wep lib80211_crypt_tkip lib80211_crypt_ccmp lib80211 iptable_mangle iptable_filter ip_tables e1000e crc_itu_t crc_ccitt compat i2c_dev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables igb i2c_algo_bit e1000 tun vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 eeprom_93cx6 xhci_plat_hcd ehci_platform xfs libcrc32c button_hotplug ptp pps_core mii [last unloaded: 8188eu]
[   87.672991] CPU: 0 PID: 1923 Comm: modprobe Tainted: G        W       4.14.72 #0
[   87.673852] Hardware name: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[   87.674777] task: ffff880006427300 task.stack: ffffc900004f8000
[   87.675462] RIP: 0010:wiphy_register+0x25e/0x850 [cfg80211]
[   87.676097] RSP: 0018:ffffc900004fb9a0 EFLAGS: 00010246
[   87.676751] RAX: 0000000000000000 RBX: ffff880006bc8300 RCX: ffffffffa084ae80
[   87.677616] RDX: 0000000000000000 RSI: ffffffff81c7aee0 RDI: ffff880006bc8300
[   87.678599] RBP: ffffc900004fba10 R08: 0000000000000000 R09: ffffc90000081869
[   87.679407] R10: ffffc900004fba28 R11: 0000000000000001 R12: ffffc90000080698
[   87.680137] R13: ffff880006bcc000 R14: 0000000000000000 R15: ffffffffa082a355
[   87.681248] FS:  00007ff17b05bb88(0000) GS:ffff880007c00000(0000) knlGS:0000000000000000
[   87.682163] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[   87.682915] CR2: 00007fe4162ceb40 CR3: 00000000067e2001 CR4: 00000000000606f0
[   87.683778] Call Trace:
[   87.684333]  ? complete_all+0x3d/0x50
[   87.684830]  ? __kmalloc+0x3d/0xf0
[   87.685383]  rtw_wiphy_register+0x47/0x4a [8188eu]
[   87.685970]  ? rtw_wiphy_register+0x47/0x4a [8188eu]
[   87.686603]  rtw_cfg80211_ndev_res_register+0x14/0x54 [8188eu]
[   87.687299]  rtw_init_netdev_name+0x18d/0x1bd [8188eu]
[   87.687907]  rtw_os_ndevs_init+0x140/0x1c6 [8188eu]
[   87.688917]  rtw_set_hal_ops+0x74d/0x840 [8188eu]
[   87.690530]  usb_probe_interface+0x188/0x220
[   87.691299]  driver_probe_device+0x133/0x2c0
[   87.692219]  __driver_attach+0x6f/0xa0
[   87.692774]  ? driver_probe_device+0x2c0/0x2c0
[   87.693537]  bus_for_each_dev+0x52/0x90
[   87.694160]  driver_attach+0x19/0x20
[   87.694672]  bus_add_driver+0xf0/0x1f0
[   87.695215]  driver_register+0x83/0xc0
[   87.695729]  usb_register_driver+0x7a/0x130
[   87.696296]  ? 0xffffffffa0889000
[   87.696799]  init_module+0x57/0x1000 [8188eu]
[   87.697473]  do_one_initcall+0x9a/0x170
[   87.698026]  ? __vunmap+0x88/0xb0
[   87.698497]  ? vfree+0x60/0x70
[   87.698950]  do_init_module+0x55/0x1d4
[   87.699499]  load_module+0x1b8d/0x2250
[   87.700075]  SyS_init_module+0x107/0x120
[   87.700689]  ? SyS_init_module+0x107/0x120
[   87.701406]  do_syscall_64+0x115/0x1b0
[   87.701908]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
[   87.702536] RIP: 0033:0x7ff17ae21fd8
[   87.703062] RSP: 002b:00007ffc50736b90 EFLAGS: 00000212 ORIG_RAX: 00000000000000af
[   87.703928] RAX: ffffffffffffffda RBX: 0000000000000000 RCX: 00007ff17ae21fd8
[   87.704676] RDX: 0000000000402b2f RSI: 0000000000274190 RDI: 00007ff17a75c020
[   87.705442] RBP: 00000000006040a0 R08: 0000000000000000 R09: 0000000000000802
[   87.706173] R10: 0000000000000000 R11: 0000000000000212 R12: 00007ff17a75c020
[   87.706982] R13: 0000000000000003 R14: 0000000000402b2f R15: 0000000000274190
[   87.707759] Code: 00 0f 84 2d 05 00 00 0f 0b bb ea ff ff ff e9 ee 05 00 00 8b 0a 89 8f 00 08 00 00 0f b7 52 04 66 89 97 04 08 00 00 66 85 c0 75 04 <0f> 0b eb 15 89 c2 66 81 e2 fe 1f 66 39 d0 74 09 0f 0b 66 89 97
[   87.710174] ---[ end trace 312f56694fd4fe2f ]---
[   87.710850] ------------[ cut here ]------------
[   87.711469] WARNING: CPU: 0 PID: 1923 at wiphy_register+0x4fe/0x850 [cfg80211]
[   87.712363] Modules linked in: 8188eu(+) ath9k_htc ath9k_common rtl8821ae rtl8192se rtl8192de rtl8192cu rtl8192ce rtl8192c_common rtl_usb rtl_pci rt2800usb rt2800pci rt2800mmio rt2800lib rt2500usb rt2500pci rt2400pci pppoe ppp_async btcoexist ath9k_hw ath rtlwifi rtl8xxxu rtl818x_pci rtl8187 rt2x00usb rt2x00pci rt2x00mmio rt2x00lib rndis_host pppox ppp_generic nf_conntrack_ipv6 mac80211 iptable_nat ipt_REJECT ipt_MASQUERADE cfg80211 cdc_ether xt_time xt_tcpudp xt_state xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_TCPMSS xt_REDIRECT xt_LOG xt_FLOWOFFLOAD usbnet slhc r8712u r8169 nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_flow_table_hw nf_flow_table nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack
[   87.718993]  lib80211_crypt_wep lib80211_crypt_tkip lib80211_crypt_ccmp lib80211 iptable_mangle iptable_filter ip_tables e1000e crc_itu_t crc_ccitt compat i2c_dev ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables igb i2c_algo_bit e1000 tun vfat fat ntfs nls_utf8 nls_iso8859_1 nls_cp437 eeprom_93cx6 xhci_plat_hcd ehci_platform xfs libcrc32c button_hotplug ptp pps_core mii [last unloaded: 8188eu]
[   87.722683] CPU: 0 PID: 1923 Comm: modprobe Tainted: G        W       4.14.72 #0
[   87.723629] Hardware name: innotek GmbH VirtualBox/VirtualBox, BIOS VirtualBox 12/01/2006
[... 125 lines cut ...]
[   87.834041]  usb_register_driver+0x7a/0x130
[   87.834711]  ? 0xffffffffa0889000
[   87.835240]  init_module+0x57/0x1000 [8188eu]
[   87.835839]  do_one_initcall+0x9a/0x170
[   87.836437]  ? __vunmap+0x88/0xb0
[   87.836908]  ? vfree+0x60/0x70
[   87.837404]  do_init_module+0x55/0x1d4
[   87.837938]  load_module+0x1b8d/0x2250
[   87.838473]  SyS_init_module+0x107/0x120
[   87.838980]  ? SyS_init_module+0x107/0x120
[   87.839848]  do_syscall_64+0x115/0x1b0
[   87.840396]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
[   87.841055] RIP: 0033:0x7ff17ae21fd8
[   87.841585] RSP: 002b:00007ffc50736b90 EFLAGS: 00000212 ORIG_RAX: 00000000000000af
[   87.842467] RAX: ffffffffffffffda RBX: 0000000000000000 RCX: 00007ff17ae21fd8
[   87.843285] RDX: 0000000000402b2f RSI: 0000000000274190 RDI: 00007ff17a75c020
[   87.844011] RBP: 00000000006040a0 R08: 0000000000000000 R09: 0000000000000802
[   87.844801] R10: 0000000000000000 R11: 0000000000000212 R12: 00007ff17a75c020
[   87.845585] R13: 0000000000000003 R14: 0000000000402b2f R15: 0000000000274190
[   87.846358] Code: 74 34 0f b6 87 c0 48 00 00 49 8d b6 93 34 00 00 84 c0 74 0d 49 8d b6 a3 34 00 00 3c 01 49 0f 45 f7 e8 fe f0 ff ff 83 f8 01 74 0b <0f> 0b 48 63 c3 4d 8d 2c c4 eb 3c 83 c3 01 eb 0c bb 00 00 00 00
[   87.848426] ---[ end trace 312f56694fd4fe32 ]---
[   88.110525] usb 1-2: reset high-speed USB device number 3 using xhci_hcd
[   88.404766] usbcore: registered new interface driver rtl8188eu
root@OpenWrt:~#
```

md3kung, 2018-11-23:
Tried the driver in v5.2.2.4 on a MIPS device running OpenWrt 18.06.1. I'm getting a shutdown when I connect my Wireless USB adapter.
How were you able to get the WEXT driver in staging to work on OpenWrt. I can get it to scan but can't get it to connect to my access point.

lwfinger, 2018-11-23:
Is your MIPS device big endian? If so, I have fixes all the Sparse endian warnings, but the driver has not been tested with BE hardware. The test is on my TODO list.

I do not do much with openWRT, and I have never used r8188eu (the one in staging) on any such device. I do have an openWRT-x86 virtual machine that I can use for testing in LE mode.

md3kung, 2018-11-23:
My system is Little Endian. I Just checked using `echo -n I | hexdump -o | awk '{ print substr($2,6,1); exit}'` [(source)](https://serverfault.com/questions/163487/how-to-tell-if-a-linux-system-is-big-endian-or-little-endian) on my MIPS device.

This is the Makefile I used :
```
include $(TOPDIR)/rules.mk

PKG_NAME:=rtl8188eu
PKG_RELEASE=1

PKG_LICENSE:=GPLv2
PKG_LICENSE_FILES:=

PKG_SOURCE_URL:=https://github.com/lwfinger/rtl8188eu.git
PKG_SOURCE_PROTO:=git
PKG_SOURCE_DATE:=2018-11-06
PKG_SOURCE_VERSION:=0c57c68d3eab5ece30d6287e08f2e533bf0e2d3b

PKG_MAINTAINER:=md3kung
PKG_BUILD_PARALLEL:=1

STAMP_CONFIGURED_DEPENDS := $(STAGING_DIR)/usr/include/mac80211-backport/backport/autoconf.h
 
include $(INCLUDE_DIR)/kernel.mk
include $(INCLUDE_DIR)/package.mk

define KernelPackage/rtl8188eu
  SUBMENU:=Wireless Drivers
  TITLE:=Driver for RTL8188EU Wireless Chipsets
  DEPENDS:= @USB_SUPPORT +kmod-mac80211 +kmod-lib80211 +rtl8188eu-firmware +kmod-usb-core +@DRIVER_11N_SUPPORT
  FILES:=$(PKG_BUILD_DIR)/8188eu.ko
  AUTOLOAD:=$(call AutoProbe,8188eu)
  PROVIDES:=kmod-rtl8188eu
endef

USER_EXTRA_CFLAGS  = \
	-I$(PKG_BUILD_DIR)/include \
	-I$(STAGING_DIR)/usr/include/mac80211-backport/uapi \
	-I$(STAGING_DIR)/usr/include/mac80211-backport \
	-I$(STAGING_DIR)/usr/include/mac80211/uapi \
	-I$(STAGING_DIR)/usr/include/mac80211 \
	-include backport/autoconf.h \
	-include backport/backport.h

CT_MAKEDEFS += CONFIG_RTL8188EU=m

USER_EXTRA_CFLAGS += -DCONFIG_IOCTL_CFG80211
USER_EXTRA_CFLAGS += -DRTW_USE_CFG80211_STA_EVENT

ifneq ($(findstring c,$(OPENWRT_VERBOSE)),)
  CT_MAKEDEFS += V=1
endif

define Build/Compile
    +$(MAKE) $(KERNEL_MAKEOPTS) $(PKG_JOBS) M=$(PKG_BUILD_DIR) \
    USER_EXTRA_CFLAGS="$(USER_EXTRA_CFLAGS)" \
    $(CT_MAKEDEFS) modules
endef

$(eval $(call KernelPackage,rtl8188eu))
```
And the patch to disable I386 build :
```
diff --git a/Makefile b/Makefile
index 9ac57ab..6f840fb 100755
--- a/Makefile
+++ b/Makefile
@@ -88,7 +88,7 @@ CONFIG_RTW_SDIO_PM_KEEP_POWER = y
 ###################### MP HW TX MODE FOR VHT #######################
 CONFIG_MP_VHT_HW_TX_MODE = n
 ###################### Platform Related #######################
-CONFIG_PLATFORM_I386_PC = y
+CONFIG_PLATFORM_I386_PC = n
 CONFIG_PLATFORM_BCM2709 = n
 CONFIG_PLATFORM_ANDROID_X86 = n
 CONFIG_PLATFORM_ANDROID_X86_N70 = n
```

Menion2k, 2018-11-28:
> Tried the driver in v5.2.2.4 on a MIPS device running OpenWrt 18.06.1. I'm getting a shutdown when I connect my Wireless USB adapter.
> How were you able to get the WEXT driver in staging to work on OpenWrt. I can get it to scan but can't get it to connect to my access point.

The WEXT one works, but you need to compile full hostap with WEXT extension enabled.
I made all the modification in branch rtl888eu of https://github.com/Menion2k/openwrt.git 

 

> I do have an openWRT-x86 virtual machine that I can use for testing in LE mode.

It is my setup, with kernel 4.14 where I get the kernel warning above. Have you been able to replicate the issue?

lwfinger, 2018-11-30:
I have not had a chance to look at this issue yet.

lwfinger, 2018-11-30:
I do have a question. Is it possible to get the kernel dump when v5.2.2.4 shuts down? I am presuming that to be a kernel oops.

md3kung, 2018-11-30:
So I made a bit of progress on this issue in order to get it working on OpenWrt I had to :
- Rename include/linux/wireless.h to include/linux/old_wireless.h
- Rename include/autoconfig.h to include/old_autoconfig.h
- Change `#include "autoconf.h"` in line 22 of include/drv_conf.h to 
```
#include <generated/autoconf.h>
#include "old_autoconf.h"
```
- Use a new Makefile :
```
include $(TOPDIR)/rules.mk

PKG_NAME:=rtl8188eu
PKG_RELEASE=1

PKG_LICENSE:=GPLv2
PKG_LICENSE_FILES:=

PKG_MAINTAINER:=Md3Kung
PKG_BUILD_PARALLEL:=1

STAMP_CONFIGURED_DEPENDS := $(STAGING_DIR)/usr/include/mac80211-backport/backport/autoconf.h
 
include $(INCLUDE_DIR)/kernel.mk
include $(INCLUDE_DIR)/package.mk

define KernelPackage/rtl8188eu
  SUBMENU:=Wireless Drivers
  TITLE:=Driver for RTL8188EU Wireless Chipsets
  DEPENDS:= @USB_SUPPORT +@DRIVER_WEXT_SUPPORT +kmod-mac80211 +kmod-lib80211 +rtl8188eu-firmware +kmod-usb-core +@DRIVER_11N_SUPPORT
  FILES:=$(PKG_BUILD_DIR)/8188eu.ko
  AUTOLOAD:=$(call AutoProbe,8188eu)
  PROVIDES:=kmod-rtl8188eu
endef

USER_MAKEDEFS += CONFIG_RTL8188EU=m

ifneq ($(findstring c,$(OPENWRT_VERBOSE)),)
  USER_MAKEDEFS += V=1
endif

NOSTDINC_FLAGS += \
	-I$(STAGING_DIR)/usr/include/mac80211-backport/uapi \
	-I$(STAGING_DIR)/usr/include/mac80211-backport \
	-I$(STAGING_DIR)/usr/include/mac80211/uapi \
	-I$(STAGING_DIR)/usr/include/mac80211 \
	-I$(PKG_BUILD_DIR)/include \
	-include backport/backport.h

NOSTDINC_FLAGS += -DCONFIG_LITTLE_ENDIAN
NOSTDINC_FLAGS += -DCONFIG_IOCTL_CFG80211
NOSTDINC_FLAGS += -DRTW_USE_CFG80211_STA_EVENT

define Build/Prepare
	rm -rf $(PKG_BUILD_DIR)
	mkdir -p $(PKG_BUILD_DIR)
	$(CP) ./source/* $(PKG_BUILD_DIR)/
	$(Build/Patch)
endef

define Build/Compile
	+$(MAKE) $(PKG_JOBS) $(USER_MAKEDEFS) -C "$(LINUX_DIR)" \
	$(KERNEL_MAKE_FLAGS) \
	KSRC=$(LINUX_DIR) SUBDIRS="$(PKG_BUILD_DIR)" \
    	NOSTDINC_FLAGS="$(NOSTDINC_FLAGS)" \
    	modules
endef

$(eval $(call KernelPackage,rtl8188eu))
```
Result : Driver works on OpenWrt 18.06.1, detects my USB dongle with cfg80211 (No warnings with dmesg), able to scan (clients appear), but **can't connect it to my WPA/WPA2 home router** when i try either it does nothing or just hangs and i have to manually reboot the device.

lwfinger, 2018-11-30:
I understand the addition of <genrated/autoconf.h>. That gets the configuration parameters for the kernel. Did you really need to rename autoconf.h in the driver? The preprocessor should have been able to distinguish between the <> and "" notation.

To try to debug the WPA/WPA2 connection, start wpa_supplicant with switches -dd and -k. That should capture the dialog between the NIC and the AP.

xifi-kif, 2018-12-01:
👍 I'm also following the topic with much interest 👍 
Will try to provide feedback for what I can

Menion2k, 2018-12-02:
> So I made a bit of progress on this issue in order to get it working on OpenWrt I had to :
> 
> * Rename include/linux/wireless.h to include/linux/old_wireless.h
> * Rename include/autoconfig.h to include/old_autoconfig.h
> * Change `#include "autoconf.h"` in line 22 of include/drv_conf.h to
> ......

Ok so you have created a custom kernel mod target and build via top-level openwrt makefile, right?
I think I start to understand the problem now.
@lwfinger Openwrt uses backport wireless driver package instead of the in kernel one, and use it to build the entire mac80211 subsystem and its related drivers.
So if someone, like me, build the kernel against the kernel build, he would endup in a driver built against kernel mac80211&cfg80211 subsystem and not the actually used one that is backport.
So the kernel warning I get I think depends on this.
Using the openwrt package makefile plugins should circumvent this issue, provided that the configuration and the includes are ok with this particular kernel modules configuration

Menion2k, 2018-12-06:
Hi
I succeed in compiling the kernel module on openwrt snapshots.
I needed to move -I$(PKG_BUILD_DIR)/include  on top of the Makefile. But I can reproduce the problem of @md3kung : linux crash or hang as soon as you try to inspect network after attempting to join an AP.
I think the problem is with the KERNEL_VERSION identification: it pickup the kernel version of the Openwrt linux kernel, while the entire mac80211 subsystem is actually a different one (normally more recent, like 4.19.x). I don't know if it is ok with just forcing the KERNEL_VERSION to the backport one, somehow, or the driver needs refactoring in order to support this kind of configuration

lwfinger, 2018-12-06:
The conditional code changes based on the LINUX_VERSION_CODE and KERNEL_VERSION macros do assume that the entire rest of the kernel conforms to the APIs of that version. Mixing the backports version of mac80211 with an unmodified 8188eu driver could lead to some API ambiguity. As not all fixes for API changes involve the MAC layer, a blanket forcing of LINUX_VERSION_CODE is not likely to work. If it builds, then the kinds of crashes that you see are likely as the driver's idea of memory are likely different from that of mac80211. Can you point me to the URL for the openWRT source you are using, and any changes you making to the source? Perhaps I can find some configuration variable that defines the API version of the mac80211 subsystem.

Menion2k, 2018-12-06:
The openwrt project is hosted here: https://openwrt.org/
In order to develop applications or kernel module on top of openwrt you have two choice:

1) Grab the source and build everything from scratch, including toolchain [see here](https://openwrt.org/docs/guide-developer/quickstart-build-images)
2) Download SDK with prebuilt toolchain, build an image (based on SDK target, so x86 for x86 SDK, mips for mips SDK, etc...) in order to have a kernel and libraries reference [here](https://openwrt.org/downloads) and [here](https://oldwiki.archive.openwrt.org/doc/howto/obtain.firmware.sdk)

Both way are valid, the second is way faster. Once you have toolchain available and the image built, you can easly cross compile just setting **STAGING_DIR** variable pointing to the **staging_dir** of your openwrt root, because the toolchain have embedded sysroot, prefix and everything.
The makefile of @md3kung is an openwrt package makefile. For the development it is better and faster to just go for a command line cross compilation, but you can use the Makefile for reference in order to build the correct make command, if you are in trouble I can help, just consider that the correct NOSTDINC_FLAGS sequence is:

```
NOSTDINC_FLAGS += \
	-I$(PKG_BUILD_DIR)/include \
	-I$(STAGING_DIR)/usr/include/mac80211-backport/uapi \
	-I$(STAGING_DIR)/usr/include/mac80211-backport \
	-I$(STAGING_DIR)/usr/include/mac80211/uapi \
	-I$(STAGING_DIR)/usr/include/mac80211 \
	-include backport/backport.h

```
Now, coming back to the problem. Yes it is caused by the mix of kernel and backport driver.
Openwrt uses backport to compile all the mac80211 and cfg80211 wireless drivers (WEXT in staging, very few indeed, are built from kernel directly).
The life of a backport driver is easy, in the end they have #ifdef on the kernel version for only the uapi outside the mac80211 and cfg80211 boundary, since within the backport itself, the version of everything is identical.
The out of tree driver life instead is complicated. You should detect, somehow, that you are compiling in mixed kernel&backport configuration and check the kernel version for everything than cfg80211&nl80211, like USB, netdev, etc..., while you need to check the backport version for the cfg80211 part.
I really don't know how it can be done, maybe some other out of tree drivers do it? There are a number of thread in openwrt forum of people trying to get some recent or out of tree drivers, mostly USB AC wireless drivers, to work in openwrt, with mixed result, and now I think I know what is the problem
Today I checked this topic for a while, there is the backport "version.h" that is a wrapper on top of kernel version.h. It includes it, plus it adds some extra macro and define. Maybe that is the way to handle such configuration.
For sure the driver needs some adaptation if you are willing to do it. In the end, once hopefully, it will get way into mainline, we will come back to a "regular" kernel module or backport module compilation.

lwfinger, 2018-12-07:
I got the make started, but it would be a great help if someone could post a copy of their ".config".

lwfinger, 2018-12-07:
I booted the x86 firmware that I generated. The kernel version is 4.14.82. There should not be too many differences between it and the backport version should not be too much.

Menion2k, 2018-12-07:
There are. Things happened in 4.19.x which is the backport version they use.

aep, 2021-01-08:
got the v5.2.2.4 branch working thanks to @md3kung  comment, but indeed it just deadlocks the kernel on initialization.

looks like exactly the same issue as aircrack-ng/rtl8812au

lwfinger, 2021-01-08:
You jump on a 2-year old threaed with a changed subject. I am ignoring all of this stuff.

closed 2021-01-08 by lwfinger