lwfinger/rtl8188eu #268: problem with wpa_supplicant

hyuma78, 2018-11-13:
I configure my wifi dongle DWA-121 ID 2001:331b but i've the problem that wpa_supplicant doesn't start at boot and i've to type everytime the commands: sudo wpa_supplicant -Dnl80211 -iwlan0 -c/etc/wpa_supplicant/wpa_supplicant.conf otherwise my interface doesn't get any ip from AP. thanks in advance.

lwfinger, 2018-11-13:
That is not a driver problem. Something in your user-space network control is screwed up. What network control method are you using? I recommend NetworkManager.

closed 2018-11-13 by lwfinger

hyuma78, 2018-11-13:
I don't have any network control method I suppose... i only sent command by terminal

lwfinger, 2018-11-13:
Then it is expected that you will need to start wpa_supplicant manually for a secured connection.

hyuma78, 2018-11-13:
I'm using this dongle with my retropie and after the boot i can't see any ip on wlan0 interface.

lwfinger, 2018-11-13:
As long as you are not using a network control program such as NetworkManager, wicd, etc., then you will need to manually start the supplemental programs such as wpa_supplicant. Your system is behaving exactly as I would expect. The driver is NOT the problem!

hyuma78, 2018-11-13:
Ok, so it's because retropie doesn't use any network-manager just you said, am i correct? thanks