lwfinger/rtl8188eu #270: Problems to install the module on 4.19-clear kernel

ltsdw, 2018-11-21:
Getting this:

**# **modprobe 8188eu****
**modprobe: ERROR: could not insert '8188eu': Exec format error**

Output from dmesg:

**[15916.199721] 8188eu: version magic '4.19.2-2-clear SMP mod_unload ' should be '4.19.2-2-clear SMP preempt mod_unload '**

lwfinger, 2018-11-21:
Your kernel headers and the compiler version MUST match exactly. "SMP mod_unload" is not the same as "SMP preempt mod_unload". I have never seen this problem as most of my kernels are self-generated, which sidesteps this problem. I'm not sure how to fix it.

ltsdw, 2018-11-21:
I'll let this issue open and close when I be able to found a solution.

mmetak, 2018-11-21:
Hi maintainer of said clear kernel in aur here. I just tried to build this module and it builds and installs fine, no errors. What I suspect is that there's a mismatch on @ltsdw system. Based on his error it shows `4.19.2-2-clear SMP mod_unload` which is correct for linux-clear kernel, but the second part looks like it's from the official arch kernel (has preempt enabled) with `SMP preempt`.

lwfinger, 2018-11-21:
Thanks for your contribution. It was obvious that there was a mismatch between the kernel and the headers. Now we know that the wrong kernel is the problem.

I probably need to see if I can modify the Makefile to detect the kind of header/kernel mismatch and fail with a more explanatory message.

ltsdw, 2018-11-21:
Recompiling the kernel again due to latest update and I deliberated used:

`CONFIG_PREEMPT=y
CONFIG_PREEMPT_RCU=y
CONFIG_PREEMPT_COUNT=y
CONFIG_DEBUG_PREEMPT=y
`

and the module was builded and installed fine. I'll close this issue. Thanks to @mmetak and @lwfinger .

closed 2018-11-21 by ltsdw