lwfinger/rtl8188eu #271: Is there hope for wicd?

plasticassius, 2018-11-21:
Thanks for the great driver, I imagine I wouldn't be using my TP-Link TL-WN722N V3 otherwise.

After some struggling, I realized that the userspace utility wicd doesn't work for me #209. Although I can use the card without wicd, is there a way to make it work with it? I'm using the master fork, would a different one be better? Out of curiosity, what are the major differences between the forks?

lwfinger, 2018-11-21:
I have no idea as to the answer to your questions. I had not used wicd for many years, but a recent installation of Q4OS tried to have me use it. It still did not work. I uninstalled it and installed NetworkManager, and my network was fine. That is what I recommend.

plasticassius, 2018-11-22:
I can try NetworkManager, but I suspect that any userspace utility won't work since iw dev doesn't show the TP-Link TL-WN722N V3 device. Although, it does show the built in (broken) wifi device when it's not blacklisted. The behavior I see is largely the same as in #209. Commands I put into /etc/network/interfaces do actually work.

I thought that the driver might be affecting how the device can be used from userspace, and that there may be differences between the forks.

lwfinger, 2018-11-22:
The only differences in the forks are the base version from Realtek and the configuration. The most important of the latter is whether cg80211 is enabled. It is not in the kernel driver, but is enabled in most of the GitHub devices. If your driver does not show in iw, then you have one with cfg80211 turned off, and your user-space network control must use the wireless extensions methods.

All of my recent work is with the v5.2.2.4 branch of this repo. That driver works very well although you do need to blacklist r8188eu and rtl8xxxu.

plasticassius, 2018-11-23:
So, branch v5.2.2.4 does make the user-space network control work, meaning wicd and likely NetworkManager. One noticeable difference is that the device led doesn't light, but it does connect properly.

I take it that what you mean by "kernel driver" are the kernel_code and master forks, and that the other two forks have cg80211 enabled.