lwfinger/rtl8188eu #272: Installed fine and I can connect but doesn't get internet

fpilee, 2018-12-20:
I have Ubuntu 16.04 
I installed the linux headers and the build essentials tools
I followed the instructions and it compiled and then I installed 
I can see the wifi networks and connect to it, It takes like 1 minute to connect 
But when I try to open a webpage doesn't load the site, it shows the icon loading in the browser but never start

lwfinger, 2018-12-20:
Are you sure you are using this driver? If so, which branch? Please post the output of the lsmod command.

fpilee, 2018-12-21:
Yes, I compiled the master branch. 

lsmod
```
Module                  Size  Used by
snd_hda_codec_hdmi     49152  1
cfg80211              622592  0
snd_hda_codec_realtek   106496  1
snd_hda_codec_generic    73728  1 snd_hda_codec_realtek
8188eu                724992  0
snd_hda_intel          40960  3
snd_hda_codec         126976  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
snd_hda_core           81920  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
snd_hwdep              20480  1 snd_hda_codec
snd_pcm                98304  4 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_core
snd_seq_midi           16384  0
input_leds             16384  0
joydev                 24576  0
snd_seq_midi_event     16384  1 snd_seq_midi
snd_rawmidi            32768  1 snd_seq_midi
snd_seq                65536  2 snd_seq_midi,snd_seq_midi_event
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
snd_timer              32768  2 snd_seq,snd_pcm
edac_mce_amd           28672  0
kvm_amd                86016  0
kvm                   598016  1 kvm_amd
irqbypass              16384  1 kvm
serio_raw              16384  0
wmi_bmof               16384  0
k10temp                16384  0
snd                    81920  17 snd_hda_codec_generic,snd_seq,snd_seq_device,snd_hda_codec_hdmi,snd_hwdep,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek,snd_timer,snd_pcm,snd_rawmidi
soundcore              16384  1 snd
i2c_nforce2            16384  0
shpchp                 36864  0
mac_hid                16384  0
parport_pc             36864  0
ppdev                  20480  0
lp                     20480  0
parport                49152  3 parport_pc,lp,ppdev
autofs4                40960  2
hid_generic            16384  0
usbhid                 49152  0
hid                   118784  2 usbhid,hid_generic
nouveau              1724416  3
mxm_wmi                16384  1 nouveau
i2c_algo_bit           16384  1 nouveau
ttm                   106496  1 nouveau
pata_acpi              16384  0
drm_kms_helper        172032  1 nouveau
syscopyarea            16384  1 drm_kms_helper
sysfillrect            16384  1 drm_kms_helper
sysimgblt              16384  1 drm_kms_helper
fb_sys_fops            16384  1 drm_kms_helper
drm                   401408  6 drm_kms_helper,ttm,nouveau
ahci                   36864  2
forcedeth              69632  0
libahci                32768  1 ahci
pata_amd               20480  0
psmouse               151552  0
video                  45056  1 nouveau
wmi                    24576  3 wmi_bmof,mxm_wmi,nouveau
```

lsusb
```
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 003: ID 04d9:a06b Holtek Semiconductor, Inc. 
Bus 004 Device 002: ID 046d:c083 Logitech, Inc. 
Bus 004 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 002: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 003 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
```

4.15.0-43-generic #46~16.04.1

fpilee, 2018-12-30:
use branch 4

closed 2018-12-30 by fpilee